dresden-elektronik/deconz-rest-plugin #3787: complete set of release notes / log of changes to the ConBee II firmware?

chas-iot, 2020-12-01:
## Describe the question or issue you are having
Is there a complete set of release notes / log of changes to the ConBee II firmware - if yes, where?
Is there a correct or better forum to ask this question?

## Screenshots
N/A

## Environment
 - Host system: Raspberry Pi 4
 - Running method: WebThings IoT Gateway (was Mozilla IoT Gateway)
 - Firmware version: (266680700)
 - deCONZ version: 2.06
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
N/A

## Additional context
* I had a working installation of a WebThings IoT Gateway, based on a ConBee II dongle
* I installed deCONZ and used the instructions to update the firmware of the dongle with GCFFlasher_internal
* After the update, the Gateway no longer connects to the the dongle. No response is seen to the probe on /dev/ttyACM0 requesting the version.
* * `deCONZ` does work
* * `lsusb` shows details of the device (edited extract below)
> Bus 001 Device 033: ID 1cf1:0030 Dresden Elektronik                                                                                                                                     
> Device Descriptor:                                                                                                                                      
>   bDeviceClass            2 Communications                                                                                                                                              
>   bMaxPacketSize0        64                                                                                                                                                             
>   idVendor           0x1cf1 Dresden Elektronik                                                                                                                                          
>   idProduct          0x0030                                                                                                                                                             
>   bcdDevice            1.00                                                                                                                                                             
>   iManufacturer           1 dresden elektronik ingenieurtechnik GmbH                                                                                                                    
>   iProduct                2 ConBee II                                                                                                                                                   
>   iSerial                 3 DE2217317                                                                                                                                                   
* I tried installing older versions of the firmware, but still no response.
* I'd like to review the change log to see if there is anything obvious, such as changes to the way the serial port is access (baud rate, etc).

Mimiix, 2020-12-01:
Hi!

Depending on the version you come from, there were changes to the udev. However, there were some versions ago in which udev was a thing that changed. From what version did you come?

Cheers,

chas-iot, 2020-12-01:
I've received a patch that means that I'm operational again. The device probing uses 5.1.1 Read Firmware Version Request from https://deconz.dresden-elektronik.de/raspbian/deCONZ-Serial-Protocol-en_1.14.pdf

With later versions of the firmware, this probe command no longer works - the it appears that length changes from 5 to 9 and additional zero padding is required. I'm searching for updated documentation so that we can support both the original shipped firmware and updated firmware.

chas-iot, 2020-12-01:
I received a pointer to this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3260#issuecomment-692771585

Mimiix, 2020-12-01:
@chas-iot I'll ask Manup to reply here. This goes above my knowledge 😅.

chas-iot, 2020-12-03:
@Mimiix I'm not sure of the protocols - roughly how long to get a response?

chas-iot, 2020-12-12:
@Mimiix any update?

Mimiix, 2020-12-12:
Haven't heard from him.  Issue is still open on my end. I'll ask him again.

MattL0, 2020-12-13:
I was also wondering if there was release note for firmware . I was sure I saw one before, I should have dreamt of it lol

paulkohm, 2020-12-17:
I'm a brand-new webthings user (running on an Ubuntu 20.04 workstation on an old Mac Mini) and I can't get webthings to recognize my new Conbee II. This open issue makes me wonder if this is maybe just something about the latest firmware? deCONZ and the webapp work fine!

chas-iot, 2020-12-18:
> I'm a brand-new webthings user (running on an Ubuntu 20.04 workstation on an old Mac Mini) and I can't get webthings to recognize my new Conbee II. This open issue makes me wonder if this is maybe just something about the latest firmware? deCONZ and the webapp work fine!

@paulkohm there's patches for the WebThings ZigBee adapter that make it work with ConBee II latest frmware. These patches are awaiting testing to ensure they don't break for ConBee II with older firmware and, ideally, documentation of the changes to the serial protocol that I requested here.

See (part 1) https://github.com/WebThingsIO/deconz-api/pull/17 and (part 2) https://github.com/WebThingsIO/zigbee-adapter/pull/265

closed 2021-03-05 by github-actions[bot]