dresden-elektronik/deconz-rest-plugin #3788: Device name

canedje, 2020-12-01:
## Device

- Product name: The device name as shown on the product or package. BlitzWolf BW-SPH13
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.BW-SPH13
- Device type : Please remove all unrelated device types. 
  - Light
 

## Screenshots

I did order in the past from Banggood a Heiman Blitzwolf BW-SPH13
This worked good.
I now did order again two devices.
These ones are not showing power messaurement.
If I look in the Deconz GUI I do see there is not Heiman as Vendor but: 
https://media.discordapp.net/attachments/688407408485204035/783409280660537364/unknown.png
If I look at the cluster infor there is power Measurement (See pictures above)
In HA there is no power measurement
My early one (BlitzWolf 1) has:
https://media.discordapp.net/attachments/688407408485204035/783409401435914290/unknown.png?width=1442&height=373

Mimiix, 2020-12-02:
@canedje In ha, it creates a Consumption and Power entity. Mostly named sensor.power_22 or something like that.

canedje, 2020-12-02:
No it isn't

Diondk, 2020-12-10:
I have the same issue as described above. i received 2 blitzwolfs but 1 of them isnt providing the power meter sensor in home assistant. 

When i check the log files i also have this error: 

09:31:08:227 Tuya debug 7 : Missing manufacture name for 0x588e81fffeff5d5f
09:31:08:227 DB SELECT manufacturername FROM nodes WHERE mac LIKE '58:8e:81:ff:fe:ff:5d:5f%' COLLATE NOCASE: _TZ3000_g5xawfcq

Already tried repairing the device but no luck.

[image]

caro7372, 2020-12-17:
> No it isn't
> EDIT:
> I did remove and repair the devices.
> Now it is showing indeed the power entity.
> I 'm only not possible to repair both of the devices again
> One is working oke now
> The other I can not pair anymore
> It is showing up in VNC but not with a name.
> I tried several times to repair, but without any luck.
> Any Advice?
> [image]
> It only creates this node as above in VNC.
> No device in the Phoscon app
> 
> If I look into the Docker Log I do see this message:
> [image]
> 
> Then after restarting Deconz Docker, there are no light visible anymore in the app.
> I do need to restore a backup to get my lights back???

How did you manage to repair one of the plugs?

J05HI, 2020-12-25:
I have the same problem. Ordered one again with the name "_TZ3000_g5xawfcq" now. Has the manufacturer changed something?

jimz011, 2020-12-29:
Same problem here, received 2 today and only on/off is showing. I see people talking about repairing, though how can we do that?

J05HI, 2020-12-30:
> Same problem here, received 2 today and only on/off is showing. I see people talking about repairing, though how can we do that?

I already removed it and repaired several times. This doesn't solve the problem.
I repaired an old one and have the same problem for it too. Looks like there is a bug in the latest version.

minke890, 2021-01-10:
I have the same issue with 3 newly ordered BW-SPH13's. Three 'older' ones work including power and consumption. Repairing does not work. No idea how to repair them, but apparently this does not work.

Smanar, 2021-01-11:
Someone that have a working and a not working one can compare the cluster list to check if they have not a difference somewhere ?

J05HI, 2021-01-15:
@Smanar 
The only difference I can see is the Application Version and Manufacturer Name.

Old working:
[image]

New not working:
[image]

Smanar, 2021-01-15:
It s not a small change ^^.
Both create the sensor device ? (with consumption) ?
If not can you give me the manufacture number  (in node information) and the 6 first character in MAC adress.

For information, the working one natively use
```
    { VENDOR_NONE, "TS0121", silabs3MacPrefix }, // Tuya/Blitzwolf smart plug
    { VENDOR_EMBER, "TS0121", silabs3MacPrefix }, // Tuya/Blitzwolf smart plug
```
VENDOR_NONE = 0x0000
VENDOR_EMBER = 0x1002
silabs3MacPrefix  = 0xec1bbd......

In reality the white list is more permissive, but the one you have can have an impact on detection.

If both create the sensor entry, forget my previous line, and pls check for cluster list.

luchtenburger, 2021-01-15:
I have similar issues with my two BW-SHP13 plugs, purchased from Banggood.
First I've used the deConz addon for the connection in HA. For one plug (Plug 2) I had both power and engergy data, for the other plug (Plug 3), I had no sensor data (only On and Off switch).

I've removed the deConz addon and installed the ZHA integration. 
Now I have the power reading for both plugs and no energy sensor (same for both plugs)

This is the data I have in the ZHA integration:
[image]

The energy sensor is created, but the data is unknown:
[image]

Mimiix, 2021-01-15:
@luchtenburger As this is about zha, it doesn't add much here. Only the ID I guess. The rest is irrelevant. @Smanar, is this a new version?

LodewijkSioen, 2021-01-18:
This is what happened to me with the new version (TZ3000_g5xawfcq).
I paired the plugs in Phoscon and gave them a new name. They appeared in Home Assistant with the new name, but without power sensors. After a while the power sensors appeared in Home Assistant, but with random names. The names of the devices then also reverted back to a random name in Home Assistant.
[image]
[image]
What other information can I give?

Mimiix, 2021-01-18:
I've asked Manuel to reply here. See what goes. I understand that there might be something wrong here.

userMak, 2021-01-18:
> This is what happened to me with the new version (TZ3000_g5xawfcq).
> I paired the plugs in Phoscon and gave them a new name. They appeared in Home Assistant with the new name, but without power [...]

I have exactly the same problem.

jimz011, 2021-01-18:
Funny how so many of you have power readings at all. I have repaired my devices a bunch of times and even let it run for a few days. No power sensor at all in Home Assistant. 

I see people talking about the name not corresponding  though I believe that is the normal behavior since my other blitwolfs (the ones that do show power consumption) also have random names. Though they are shown in the device list (when you click on devices in HA you can see which sensors are associated with that device).

The ones that don't work (the ones with the same serial as the ones posted here) do not show any power consumption at all.

I really wonder how you guys were able to get power consumption readings at all.

jimz011, 2021-01-18:
Not working:

[image]

Working:
[image]

userMak, 2021-01-18:
> Funny how so many of you have power readings at all. I have repaired my devices a bunch of times and even let it run for a few days. No power sensor at all in Home Assistant.
>
> I see people talking [...]

It is very strange. I bought 4 plugs from the same shop. 3 of them are working (with power consumption etc) and one of them it is not.

LodewijkSioen, 2021-01-19:
One extra thing: I can turn the plug on/off from the Phoscon UI, but using the Switch Entity from Home Assistant doesn't work. The toggle reflects the current state of the plug, but using it as a switch does nothing.

J05HI, 2021-01-19:
@Smanar
I've reset the plug to the factory defaults through deCONZ. Afterwards I've repaired the device. After reloading the integration in Home Assistant it shows the power and consumption and everything works as it should.
Where's the key problem here?

userMak, 2021-01-19:
> @Smanar
> I've reset the plug to the factory defaults through deCONZ. Afterwards I've repaired the device. After reloading the integration in Home Assistant it shows the power and consumption and everything works as it should.
> Where's the key problem here?

How did you reset the plug through Deconz please?

Mimiix, 2021-01-19:
@userMak Please read the manual for that.

**On a moderation note: Please don't clutter the issue with #metoo or #howcanipls questions. **

@Smanar @manup @SwoopX Is this a issue of the device (firmware) or something we can fix. Either way, we need a way forward.

manup, 2021-01-19:
We put the devices and a bunch of other Tuya stuff on the order list, hard to tell what's actually going on otherwise.

https://de.banggood.com/BlitzWolf-BW-SHP13-Zigbee-3_0-Smart-WIFI-Socket-16A-EU-Plug-Electricity-Metering-APP-Remote-Controller-Timer-Work-with-Amazon-Alexa-Google-Home-p-1679992.html?cur_warehouse=CN&rmmds=search

They seem not to be available on Amazon :/

up2datecyborg, 2021-01-23:
Hi,
I have exactly the same problem, but I'll explain what happened in my case:
- I had a running Home Assistant with a Conbee II USB stick.
- I first paired a SHP13 with success
- I bought another Conbee II, to migrate to a new raspberry with a new case, etc
- I can pair the same plug with the new Conbee II, and see it as a light in Phoscon (until this step, everything ok)
- When I sync in Home Assistant, I only see the plug, no power and no power consumption
- Deconz seems to see the info of power and consumption, but, calling the API I only see the plug, no power and no power consumption, for other SHP13s I can see plug+power+consumption

Another plug I wanted to migrate worked succesfully.

I bought 18 more SHP13 plugs, and when pairing them for the first time, I don't get any issue with them. So I don't know if it has something to do with the fact that it was already paired once...

I tried to repair it more than 20 times, I tried what @J05HI said about factory reset through deconz, but no luck with this plug.
I also tried to backup and restore conbee config via phoscon web page, but no luck neither.

If you need any useful info calling the API, or screenshot of deconz, I'd be pleased to help to resolve this problem.

TheBisk, 2021-02-05:
[image]
[image]
i see that in deconz app the state is syncronized while in home assistant the state is off but the switch it is on. Why?
I cannot control the switch via home assistant only via deconz...any fix? thanks

Mimiix, 2021-02-05:
@TheBisk  The same  goes for you https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3788#issuecomment-762782418

**I'll lock this issue until @manup confirms / fixes it.**

manup, 2021-02-22:
Related PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4448

Mimiix, 2021-03-23:
**As there  was a fix included in 2.10.1, this is reopened for any feedback on it. **
**Please note that you need to remove the device (using the restapi/Phoscon) and re-add it after updating to the new version**

marithpl, 2021-03-23:
I connected 2 smart plugs Blitzwolf BW-SHP13 to Deconz. Unfortunately without power consumption in Home Assistant.

**Device:**
Blitzwolf BW-SHP13
**Phoscon App:**
Vendor_TZ3000_g5xawfcq

Host system: NAS
Running method: Marthoc Docker container
Firmware version: 26660700
deCONZ version: 2.10.02 / 23.12.2020
Device: ConBee II

[image]

marithpl, 2021-03-24:
@manup Changes will be released in 2.10.05?

Mimiix, 2021-03-24:
No clue what he did there. I told him while we were on the phone 😂. He didn't know either.

It has his attention and he is picking the device up today at work. It won't come in 2.10.5 (as that would never come). Next beta will be 2.11.0. I make sure he does a fix if needed :)

marithpl, 2021-03-24:
I will appreciate that :)

Pedder007, 2021-03-25:
Hi all, me too 👍 
I was struggeling now over days with that issue. Also having the device firstly paired to my dev environment successfully, incl. all measurements, and then, when switching it to my prod enviroment, no more measurement any longer at all. It really runs me crazy until I found the related threads here. 
Just updated to 2.10.04 but it also didn't bring the solution, but then I read to the end here ;-)
... I will wait then :-)

manup, 2021-03-27:
Hey all, of course I forgot to take the device with me when I was in the company :facepalm: so will test during the week.

The above changes are where already part of [v2.10.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.10.1-beta) and readings should be done as long as the `ZHAConsumption` and `ZHAPower` sensors are created and returned in the `/sensors` REST-API.

From my database I can see that there were values queried from the device here is the last state:

ModelId: _TZ3000_g5xawfcq

```
ZHAConsumption : 60:a4:23:ff:fe:5e:2a:2d-01-0702
{
    "consumption": 130,
    "lastupdated": "2021-02-20T12:49:39.524"
}

ZHAPower :  60:a4:23:ff:fe:5e:2a:2d-01-0b04
{
    "current": 0,
    "lastupdated": "2021-02-20T12:50:12.158",
    "power": 0,
    "voltage": 232
}
```

I've noticed that the MAC address prefix looks different than the above not sure if it's relevant :thinking:

SebLz, 2021-03-28:
> Same problem here, received 2 today and only on/off is showing. I see people talking about repairing, though how can we do that?

By "repair" I assume people mean "pair again", so by pressing the SHP13 on/off button for a few seconds to make it join the zigbee network again. This confused me too at first.

I also have the same issue, tried to delete (with reset param in deconz rest API) device, pair again, restart deconz / hassio and it still doesn't show power consumption (but I do get the readings in deconz). The funny thing is that the plug that is not working anymore, used to work in older deconz/HA versions. I have another SHP13 with exactly the same Manufactureur Name, ZCL & application versions that has no problem at all.

Mimiix, 2021-03-28:
Can you provide screenshots of both basic clusters?

SebLz, 2021-03-28:
Here they are ("prise deshumidificateur" has no metering in HA, "prise desktopseb" does):
[image]
[image]

Pedder007, 2021-03-28:
The problem at least only occures, when one is trying to pair the same plug the second time.  From then on, the sensors seems not any longer beeing discovered by Phoscon. It then only is discovered as a simple light further more, regardless you go to another environment (ConBee stick / Phoscon / deCONZ).
Have this since trying to migrate a plug from my dev environment to my prod environment. Moving back the plug to dev also does not work.
Took now a new one, directly connected it to my prod environment and it works there - for sure will not move it or disconnect unless this issue is not solved 😅

SwoopX, 2021-03-28:
@SebLz Can you please also provide the screenshots for the node info panel for both devices?

SebLz, 2021-03-28:
[image]
[image]

Mimiix, 2021-03-28:
@SwoopX notice the Mac prefix?

marithpl, 2021-04-02:
> No clue what he did there. I told him while we were on the phone 😂. He didn't know either.
> 
> It has his attention and he is picking the device up today at work. It won't come in 2.10.5 (as that would never come). Next beta will be 2.11.0. I make sure he does a fix if needed :)

Do you know when 2.11.0 will release?

Finne75, 2021-04-05:
Hello, speaking about MAC prefixes - I have one plug where metering works (paired that with 2.9.3 already) , and one without (paired that today with 2.10.4). They seem to have the same prefix. I will tinker a bit more with the new one - until now, I did the usual delete/re-pair cycle, reading of descriptors, without success.

The one with metering:

[image]

The one without:

[image]

manup, 2021-04-05:
While currently testing the device I've noticed, that the sensor resources for power measurement are only created when the device is joined via sensor search. Therefore in the case these are missing, start search for sensors and power cycle the device (doesn't need to be reset).

Finne75, 2021-04-05:
@manup that worked, thank you!

marithpl, 2021-04-05:
> While currently testing the device I've noticed, that the sensor resources for power measurement are only created when the device is joined via sensor search. Therefore in the case these are missing, start search for sensors and power cycle the device (doesn't need to be reset).

It doesn’t work for me.  But maybe I made something wrong. I tried 2 ways:
1) Start looking for sensor and click turn on/off button on device a few times. 
2) Remove device from phoscon and made the same as above.

Finne75, 2021-04-05:
@marithpl try to unplug/plug the device from the socket.

marithpl, 2021-04-06:
> @marithpl try to unplug/plug the device from the socket.
Doesn't work for me.

SebLz, 2021-04-06:
This weekend I was working on my electrical installation and doing so I had to turn off / on power multiple times the circuit on which my SHP13 with no sensors was plugged in. Yesterday, I noticed the sensors had appeared in HA ! This happened even though I did not start a search for new sensors in deconz.

Mimiix, 2021-04-06:
@marithpl I am starting to think yours is defective.

marithpl, 2021-04-06:
I have 2 plugs, none of them doesn’t show power sensors.

SebLz, 2021-04-06:
> I have 2 plugs, none of them doesn’t show power sensors.
> But I found wired behavior, when I remove a device from phoscon/deconz it's not removed from HA.

After doing that, have you tried to:

refresh device list + remove orphaned entries
[image]

Reload integration:
[image]

?

marithpl, 2021-04-06:
I try this way

Mimiix, 2021-04-06:
> I have 2 plugs, none of them doesn’t show power sensors.

Do they show in the rest api?

marithpl, 2021-04-06:
```
"13": {
"etag": "86434a0dc2be1e8922e17ae5d38ad1c2",
"hascolor": false,
"lastannounced": null,
"lastseen": "2021-04-06T18:44Z",
"manufacturername": "_TZ3000_g5xawfcq",
"modelid": "TS0121",
"name": "Smart plug 13",
-"state": {
"alert": "none",
"on": true,
"reachable": true
},
"swversion": null,
"type": "Smart plug",
"uniqueid": "60:a4:23:ff:fe:61:d7:fd-01"
},
-"15": {
"etag": "ac7a27f3daaa4d4d870c360fdf4b4f3e",
"hascolor": false,
"lastannounced": null,
"lastseen": "2021-04-06T18:45Z",
"manufacturername": "_TZ3000_g5xawfcq",
"modelid": "TS0121",
"name": "Smart plug 15",
-"state": {
"alert": "none",
"on": true,
"reachable": true
},
"swversion": null,
"type": "Smart plug",
"uniqueid": "60:a4:23:ff:fe:5e:2a:db-01"
}
```

Only as lights. I try to remove them from deconz/phoscn, I've cleared orphaned devices in HA, plug it off from power socket and of course I try to pair it from sensor tab not from light tab.

Mimiix, 2021-04-06:
> ```
> "13": {
> "etag": "86434a0dc2be1e8922e17ae5d38ad1c2",
> "hascolor": false,
> "lastannounced": null,
> "lastseen": "2021-04-06T18:44Z",
> "manufacturername": "_TZ3000_g5xawfcq",
> "modelid": "TS0121", [...]

How about the sensor api call?

marithpl, 2021-04-06:
I can't find any thing in common with this plugs. Only motion/door/water sensors.

monotonytonytony, 2021-04-06:
> @marithpl try to unplug/plug the device from the socket.

That worked for me!!
I got 6 of those when I first tried to add them by searching for light for the first one and did not worked.

Then I added the 5 others with sensor search and it worked but I could not add the first one with power sensor even after deleting and reset the plug with sensor search.

Then I saw @manup post so I just made clicked on sensor search and then unpluged and plugged the item and that WORKED!!

No need to reset the plug.

Pedder007, 2021-04-06:
Hi all, I'm sorry, but I can do here whatever I want (re-discover/re-pair, plug/unplug, switch with or without load, reset, ...), this first of my plugs, which I already paired formerly to my dev environment, is not delivering any measures to my prod environment.
And I just verified again that this already happens while pairing it to my ConBee stick / deCONZ. I just took another new one of these plugs, firstly paired it as switch and secondly as sensor and this one again works great.
The problem plug is already not by Phoscon discoverd as sensor, as the respective search process always ends without a result.

[73 more replies]

closed 2022-04-22 by github-actions[bot]