dresden-elektronik/deconz-rest-plugin #3792: Paulmann Gent Remote Control 500.67

danieldemus, 2020-12-02:
## Device

- Product name: Gent Remote Control
- Manufacturer: Paulmann Licht GmbH
- Model identifier: RemoteControl
- Device type :
  - Remote
  - Switch
## Screenshots
Screenshots and further information already added to prematurly closed issue #1603

This issue is opened because #1603 was incorrectly set to stale, and has since been ignored despite multiple requests to re-open.

Mimiix, 2020-12-02:
Good catch! I'll close this one and re-open 1603. However: That issue has not all screenshots required.

Mimiix, 2020-12-02:
Ignore that: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3587

Please use that issue. That one is for this remote too.

closed 2020-12-02 by Mimiix

danieldemus, 2021-01-27:
Hi, thanks for checking but the stale bot closed #3587

ITSSeu, 2021-04-29:
Hi, all of the old trends from 2019 to 2020 and 2021 are closed. Is there a piece of hope to get the remote on the supported list? I am really interested in. Many thanks to you!

Mimiix, 2021-04-30:
@ITSSeu It would be best if a dev can have a handson. Perhaps anyone is willing to send one to a dev?

Artur-Fedjukevits, 2022-09-01:
I have the same remote control. I would be very grateful to see this device as supported.