dresden-elektronik/deconz-rest-plugin #3794: Entire network stopped responding

fiLLLip, 2020-12-02:
## Describe the question or issue you are having

My entire system stopped responding, suddenly. Did not do anything, just noticed that the lamps stopped responding and the sensors stopped triggering actions in HomeAssistant.

## Screenshots

[image]
(following covid rules, keeping its distance, removing near relations)

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Marthoc Docker container
 - Firmware version: 26390500
 - deCONZ version: 2.05.88 
 - Device: RaspBee I
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
```
21:29:47:215 Current channel 25
21:29:47:237 Device TTL 1926 s flags: 0x7
21:30:28:170 Device protocol version: 0x010C
21:30:28:201 aps request id: 224 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0xCCCCCCFFFE58FF11 (0xFFFD)
21:30:28:301 aps request id: 224 finished, erase from queue
21:30:28:531 Current channel 25
21:30:28:643 CTRL ANT_CTRL 0x03
21:30:31:209 Device protocol version: 0x010C
21:30:31:458 Current channel 25
21:30:31:595 CTRL ANT_CTRL 0x03
21:30:31:669 start reconnect to network
21:30:33:264 Device protocol version: 0x010C
21:30:33:458 Current channel 25
21:30:33:530 CTRL ANT_CTRL 0x03
21:30:34:525 Mgmt_Lqi_req zdpSeq: 19 to 0x00212EFFFF041124 start index 0
21:30:34:526 APS-DATA.request id: 139, addrmode: 0x03, addr: 0x00212effff041124, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
21:30:34:624 APS-DATA.confirm id: 139, status: 0x00 SUCCESS
21:30:34:625 APS-DATA.confirm request id: 139 -> confirmed, timeout 20009416
21:30:34:644 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
21:30:34:645 APS-DATA.indication request id: 139 -> finished
21:30:34:646 APS-DATA.request id: 139 erase from queue
21:30:34:647 ZDP status = 0x00 -> SUCCESS
21:30:34:647 ZDP Mgmt_Lqi_rsp zdpSeq: 19 from 0x00212EFFFF041124 total: 0, startIndex: 0, listCount: 0
21:30:35:006 APS-DATA.request id: 143, addrmode: 0x02, addr: 0xfffd, profile: 0x0000, cluster: 0x0000, ep: 0x00 -> 0x00 queue: 0 len: 11 tx.options 0x00
21:30:35:007 send NWK_addr_req to 0x086BD7FFFE2C5F2A, last seen 0 s, last seen by neighbors 0 s
21:30:35:261 APS-DATA.confirm id: 143, status: 0x00 SUCCESS
21:30:36:920 reconnect network done
21:30:47:218 Current channel 25
21:30:47:241 Device TTL 1866 s flags: 0x7
21:30:49:885 Mgmt_Lqi_req zdpSeq: 21 to 0x00212EFFFF041124 start index 0
21:30:49:887 APS-DATA.request id: 203, addrmode: 0x03, addr: 0x00212effff041124, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
21:30:49:935 APS-DATA.confirm id: 203, status: 0x00 SUCCESS
21:30:49:937 APS-DATA.confirm request id: 203 -> confirmed, timeout 16148704
21:30:50:002 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
21:30:50:005 APS-DATA.indication request id: 203 -> finished
21:30:50:006 APS-DATA.request id: 203 erase from queue
21:30:50:007 ZDP status = 0x00 -> SUCCESS
21:30:50:007 ZDP Mgmt_Lqi_rsp zdpSeq: 21 from 0x00212EFFFF041124 total: 0, startIndex: 0, listCount: 0
21:30:52:294 dev /dev/ttyAMA0
21:30:52:314 GW firmware version: 0x26390500
21:30:52:316 GW firmware version is up to date: 0x26390500
21:31:05:357 Mgmt_Lqi_req zdpSeq: 22 to 0x00212EFFFF041124 start index 0
21:31:05:358 APS-DATA.request id: 11, addrmode: 0x03, addr: 0x00212effff041124, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
21:31:05:407 APS-DATA.confirm id: 11, status: 0x00 SUCCESS
21:31:05:409 APS-DATA.confirm request id: 11 -> confirmed, timeout 16148704
21:31:05:524 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: 0
21:31:05:525 APS-DATA.indication request id: 11 -> finished
21:31:05:526 APS-DATA.request id: 11 erase from queue
21:31:05:527 ZDP status = 0x00 -> SUCCESS
21:31:05:527 ZDP Mgmt_Lqi_rsp zdpSeq: 22 from 0x00212EFFFF041124 total: 0, startIndex: 0, listCount: 0
21:31:14:203 Node_Descriptor_req zdpSeq: 23 to 0xCCCCCCFFFE58FF11
21:31:14:206 APS-DATA.request id: 49, addrmode: 0x02, addr: 0x9d6b, profile: 0x0000, cluster: 0x0002, ep: 0x00 -> 0x00 queue: 1 len: 3 tx.options 0x00
21:31:20:549 Mgmt_Lqi_req zdpSeq: 24 to 0x00212EFFFF041124 start index 0
21:31:20:551 APS-DATA.request id: 75, addrmode: 0x03, addr: 0x00212effff041124, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
21:31:20:600 APS-DATA.confirm id: 75, status: 0x00 SUCCESS
21:31:20:602 APS-DATA.confirm request id: 75 -> confirmed, timeout 20016360
21:31:20:676 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 0, rssi: -31
21:31:20:678 APS-DATA.indication request id: 75 -> finished
21:31:20:680 APS-DATA.request id: 75 erase from queue
21:31:20:681 ZDP status = 0x00 -> SUCCESS
21:31:20:683 ZDP Mgmt_Lqi_rsp zdpSeq: 24 from 0x00212EFFFF041124 total: 0, startIndex: 0, listCount: 0
21:31:24:406 APS-DATA.confirm id: 49, status: 0xD0 
21:31:24:869 APS-DATA.request id: 94, addrmode: 0x02, addr: 0xfffd, profile: 0x0000, cluster: 0x0000, ep: 0x00 -> 0x00 queue: 1 len: 11 tx.options 0x00
21:31:24:870 send NWK_addr_req to 0xCCCCCCFFFE58FF11, last seen 0 s, last seen by neighbors 0 s
21:31:25:126 APS-DATA.confirm id: 94, status: 0x00 SUCCESS
21:31:35:116 aps request id: 143 prf: 0x0000 cl: 0x0000 timeout (confirmed: 1) to 0x086BD7FFFE2C5F2A (0xFFFD)
21:31:35:193 aps request id: 143 finished, erase from queue
```

## Additional context
Mostly Xiaomi sensors, and Ikea bulbs, drivers and plugins. I tried to go to Phoscon and ALT-click settings and load previous settings, with no luck. Also tried the Leave -> Save -> Join path with no luck. I see something weird, that when I change settings, and click save, channel settings don't stick

fiLLLip, 2020-12-03:
A strange development! I have some Trådfri lights for the wardrobe, and they respond to group commands, but is seen as offline both in Phoscon and Home Assistant, and there is no "link" shown in Deconz. And also the IKEA RGB bulbs seem to respond to something. Common for those are they are Light Link.
https://www.ikea.com/no/no/p/tradfri-driver-for-tradlos-kontroll-gra-60342656/
[image]

During writing this, I also achieved brief contact with one of the other Ikea bulbs, just after "rebooting" it (turning the switch on and off), but then it fell offline again. I managed to turn it on via Phoscon before it went awall.
[image]

Any ideas?

Mimiix, 2020-12-03:
Just some general advice:
Powercycle the lights, see if they come back.
Restart the RPI

Also, to confirm: You have no other usb devices attached to the pi?

fiLLLip, 2020-12-03:
> Powercycle the lights, see if they come back.

Done, does not help

> Restart the RPI.

Done, does not help. Also removed power, and let it sit, and then repower, does also not help.

> Also, to confirm: You have no other usb devices attached to the pi?

I have two units connected, a Aeotec ZWave dongle, and a Lectrosonics DM1624. Would those interfere?
I'm only forwarding the `--device=/dev/ttyAMA0` to docker. Zwave is working fine though, no issues there (`/dev/ttyACM0`).

Smanar, 2020-12-03:
>(following covid rules, keeping its distance, removing near relations)

Lol, right, you network respect rules. 

But it seem all your devices have grayed name ?
You haven't a backup to restore ?

Have you tried to compare your setting with this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

fiLLLip, 2020-12-03:
> But it seem all your devices have grayed name ?
> You haven't a backup to restore ?

Unfortunately, no :(
 
> Have you tried to compare your setting with this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

I have flashed the latest firmware, using :latest docker instance, and tried the leave -> save -> joing approach, without any luck.

Noticing more weirdness as we go here. Now most groups respond to some commands, everything looks like it's offline. And just no, a device (not group) responded to off command, but not on...

dcosson, 2020-12-04:
This just happened to me too, same problem with all devices grayed out and not responding all of a sudden. I've restarted everything and I tried deleting and re-pairing some of the zigbee devices and now the phoscon app won't recognize anything in pairing mode either.

> Also, to confirm: You have no other usb devices attached to the pi?

I plugged in a new usb hard drive to the raspberry pi before the system broke, it sounds like that could be an issue? How can I pass the right usb interface when running the deconz app?

fiLLLip, 2020-12-04:
Well, the weirdness of this amazes me. My network has successfully vaccinated against covid...
[image]
If someone wants/can explain what happened to my network, that would be great! 
Short recap:

- Suddenly network died
- I clicked all the buttons without success
- Gave up
- Suddenly network started working! At least most of it

I'm going to take a backup now...

SwoopX, 2020-12-04:
Did you do a firmware upfate before that happened?

fiLLLip, 2020-12-04:
> Did you do a firmware upfate before that happened?

Nope, the installation has been left untouched for almost a month before the network suddenly stopped responding.

fiLLLip, 2020-12-07:
Well, it died again... Now everything is gone again. Could it be a malfunctional RaspBee?

Mimiix, 2020-12-07:
IS there a possibility to try without other USB devices? Maybe something up with that?

fiLLLip, 2020-12-07:
> IS there a possibility to try without other USB devices? Maybe something up with that?

Of course I could try, but is there a particular reason to why you suspect that interferes? The configuration has not changes in several months, working just fine. And the RaspBee is connected to the Raspberry Pi headers, not USB.

Mimiix, 2020-12-07:
@fiLLLip I've read earlier about USB devices switching locations. It's just trying to troubleshoot where the issue might be, if you have no issues when the other USB's aren't there: might as wel fix it. Also, i've seen a case in which a Logitech unifying USB caused interference on a Raspbee. 

Also could very well be a PowerSupply going bad.

M2k80, 2020-12-20:
i am having exactly the the same, i am using the marthoc/deconz docker on unraid. i am getting much errors like this error APSDE-DATA.confirm: 0xD0 on task. no chance to get the network correct working again. for me its working on unraid .35 beta but on RC1 and RC2 it does not. between .35 and rc1 and rc2, as i know there is a newer kernel. i am not a proffessional, but maybe kernel related changes could be the problem for all of this???

Mimiix, 2020-12-21:
@M2k80 That means theres interference issues. Are you using a extension cable? 

I'd suggest opening a own issue.

M2k80, 2020-12-21:
no i am not using an extension cable. but on unraid beta .35 there is no problem all is working fine and with rc1 and rc2 the stick is no more able to talk to all my devices, i tried many time to replug it, when i was lucky some devices come back in deconz interface, i can see the green lines, but also homeassistant cant communicate with them which are available....downgrade to beta35 instantly fix the problem. i reported my problem also in the unraid forum, but i think i have to wait to open an extra issue  until unraid 5.9 is final and more users are using it.
beta35 uses kernel 5.8.18 and docker version 19.03.13
rc2 uses kernel 5.10.1 and docker version 19.03.14

fiLLLip, 2020-12-22:
> @fiLLLip I've read earlier about USB devices switching locations. It's just trying to troubleshoot where the issue might be, if you have no issues when the other USB's aren't there: might as wel fix [...]

I tested without any other USB's connected, same issue. Also tested another power supply without any luck. Still facing the same problems now as I initially did. Could it be possible that the Raspbee is going faulty? I'm considering buying a Conbee2 to replace it with. Does deCONZ support in place replacement of controllers? Would it connect to my network if reusing all the settings from today?

Mimiix, 2020-12-22:
No, it should go _faulty_ like that.

I'd suggest updating to the latest version and then see again.  Also, channel can't be changed when the device is _in network_. Best way is to leave the network, then change channel, then click "join" again.

fiLLLip, 2021-02-01:
The network is still dead, with nothing suggesting what causes it. I'm transitioning to zigbee2mqtt once I get hold of a dongle that supports it.

Mimiix, 2021-02-01:
@fiLLLip Can you provide some more logs?

closed 2021-03-05 by github-actions[bot]