dresden-elektronik/deconz-rest-plugin #3795: Opple switch 2-gang and 3-gang can not be paired

stheinen, 2020-12-03:
## Describe the question or issue you are having

I tried to pair an OPPLE 2-gang as well as an 3-gang switch. 
Pairing failed! 
There are already OPPLE switches paired using an older firmware.
I tried to pair with the PWA as well as with the PWAbeta app.

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: (26680700)
 - deCONZ version: (2.07.00)
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

[pairing-deconz-log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5633762/pairing-deconz-log.txt)

## Additional context

stheinen, 2020-12-03:
I tried it again. Now the pairing worked.

stheinen, 2020-12-04:
The behavior of the switches is not ok. I have to investigate that further, but I do not get any respose in the deconz iobroker adapter. 
Does anybody else have that issue?
Can I downgrade to an older firmware version?

stheinen, 2020-12-06:
Ein Downgrade der Firmware auf Version 26670700 mit ´deCONZ 2.07.00 hat das Problem nicht gelöst.

Nach dem Downgrade auf deCONZ 2.05.86 / 15.10.2020 werden die Tastendrücke des Schalter richtig im deconz iobroker Adaper angezeigt.
Auch gibt es keine unmotivierten Schaltvorgänge mehr.

SwoopX, 2020-12-06:
You need to push the "C" button on the back.

stheinen, 2020-12-07:
This is not the probam as I have already paired successfully multiple Opple switches in the older Firmware!

SwoopX, 2020-12-07:
Have you done it?

stheinen, 2020-12-07:
Yes!
The switches have been paired with the old Software version.
The data points generated for the 2-gang switches are weird! It seems to me that there is no correct  initialization of the names within the deconz adapter.
Best Regards
Stefan Heinen

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>

ninja-warrior, 2020-12-07:
Hey guys,

I have also some trouble in connecting a couple of new Opple 3-gang switches to ConBee II. I've already connected some switches with an older version successfully to Deconz and use this switches as datapoints in IOBroker. 

Yesterday, I tried to connect three new Opple 3-gang switches to ConBee II (the same way I did with the older version). Unfourtantely I am not able to see the devices in my device list (Category: Schalter). But for whatever reason, when I hit a button at the newly "not successfully connected" switches all of my deconz devices turn off. When I hit the same button again all of my devices turn on again. Seems like something is wrong in the process of connecting the switch to deconz.

Any idea how to fix that behaviour?

Best regards,
David

vhrabe, 2021-01-02:
Hello,

I have the same problem with new opple 3-ways switches.
Can't be paired with new 2.09.00 version. When I try to pair, application is closed and can't be open again till reboot whole raspberry.
After reboot and running Deconz app, left button turn_off all lights and right button turn_on all lights.

Regards Vaclav

hsteinme, 2021-01-02:
I too have basic problems integrating an Opple 3-gang switch. Two or three weeks ago I integrated the first switch of this kind into deConz. It has been running without problems since then.

This week I tried to use a second Opple 3-gang switch. It could be paired with the gateway quite quickly. But every push of a button in the top row turns all the lights and actuators in the house controlled by deConz on or off. Quite strangely, this erroneous behaviour occurs even after the switch has been deleted in the Phoscon app. It also occurs regardless of whether the switch is already assigned to a light group or not.

I have been able to reproduce this behaviour with the following combinations of deConz versions and firmware states:

- 2.07.01 + 26580700
- 2.05.87 + 26580700
- 2.09.00 + 26580700
- 2.09.00 + 26660700
- 2.09.00 + 26680700

However, the constellation 2.05.84 + 26580700 is interesting. There the 3-gang Opple switch really switches only those lamps to which it is assigned. 

BUT: Under 2.05.84 I cannot assign the functions "on or dim up" and "off or dim down" to the switch. Only the functions "on", "off", "dim up x%" and "dim down x%" are available. Not only does this make the dimming process unwieldy (several individual pressings instead of one continuous pressing), it also requires twice as many buttons for on/off/dimming. So this cannot be a solution (for me).

Mimiix, 2021-01-03:
It's already told that you need to click the C button on the back once.

hsteinme, 2021-01-03:
Thank you for your friendly hint. Just some minutes ago I also found the solution [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3670#issuecomment-743731751).

hsteinme, 2021-01-03:
But one point is still unclear to me: Many users have reported that they were able to use the Opple 3-gang switch without any problems under version 2.05.84, but not under other versions. Can it really be that all these people have carried out the required post-pair click on the reset button accidently under 2.05.84 - and never otherwise? Or does the software still have something to do with the question of whether this switch works properly or not?

Mimiix, 2021-01-03:
> But one point is still unclear to me: Many users have reported that they were able to use the Opple 3-gang switch without any problems under version 2.05.84, but not under other versions. Can it [...]

It did that with mine when i got it too. It's just how the switch works. I just clicked it and it worked. I didn't bother to ask around and just tinkered with it.

vhrabe, 2021-01-03:
I click C button on back, Switch was working for few hours, now again turn on all zigbbe light or turn off.
now pressing the C button don't help in any way. I'm on version 2.09.00 + 26680700

rvsoftz, 2021-01-17:
I was able to downgrade to version 2.05.84 in my Ubuntu installation, and that fixed the problem. "Deconz_event" Events are sent to HA.

How can I downgrade to version 2.05.84 in my Hass.io installation? Can I do it thru SSH?

Thank you

hsteinme, 2021-01-21:
> It's already told that you need to click the C button on the back once.

After the friendly advice here, I now have six 3-gang Opple switches successfully in use. I have not yet been able to pair a 1-gang Opple switch (https://de.aliexpress.com/item/4001154276289.html). Even if I press the reset button on the back of the switch for several seconds, the light on the top left front does not start flashing. I have tried many times, all to no avail. Raspberry and Conbee restarts have brought no improvement.

Has anyone of you ever successfully got the 1-gang Opple switch to work?

hsteinme, 2021-01-22:
This is a real annoyance! Bought two brand new Opple 1-speed switches. Neither of them was willing to pair. I put new batteries in both switches and they were paired with the gateway in seconds. A look into the internet shows that I am not the only one who has experienced this annoyance. So: If the control light does not flicker after a few seconds when you press the reset button on Opple switches, insert a new battery straight away. Auntie YouTube shows us how to do this: https://www.youtube.com/watch?v=HC41WsIPzmQ

Mimiix, 2021-01-22:
So they basically shipped them with dead batteries? That's a shame.

Mimiix, 2021-01-22:
@stheinen Can we close this issue?

rvsoftz, 2021-01-22:
"Short press C after pairing did the Job! Thank you very much!"

This small step saved my day :) Working with latest version in Hassio

MrWGT, 2021-02-13:
I own a 3gang version running great but the 2 gang is driving me mad.

Even after delete, repair, short C press after pairing I cannot get key assignment to work.

Some keys work, sometimes other lights are controlled too. Switching one light off turns other on and so on.

Can't find a psttetn

MrWGT, 2021-02-13:
removed from deconz, restart, repaired, Short pressed C. Opple 2gang behaves weird.

```
20:03:41:119 Multi state present value: 0x0001 (1), lumi.remote.b486opcn01
**20:03:41:121 [INFO] - Button 1002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), action: Off press, payload: 5500210100, zclSeq: 87**
20:03:41:122 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
**20:03:41:128 [INFO] - Button 2002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x02, cluster: MULTISTATE_INPUT (0x0012), action: On press, payload: 5500210100, zclSeq: 88**
20:03:41:129 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
20:03:41:133 rule event /sensors/58/state/lastupdated: 0 -> 0
20:03:41:134 trigger rule 73 - Rule ON
20:03:41:134 trigger rule 17 - Rule PUSHDIMDOWN_OFF
20:03:41:135 trigger rule 23 - Rule PUSHDIMDOWN_OFF
20:03:41:135 trigger rule 41 - Rule PUSHDIMUP_ON
20:03:41:141 rule event /sensors/58/state/lastupdated: 0 -> 0
20:03:41:141 trigger rule 73 - Rule ON
20:03:41:142 delayed group sending
20:03:41:142 trigger rule 17 - Rule PUSHDIMDOWN_OFF
20:03:41:142 delayed group sending
20:03:41:142 delayed group sending
20:03:41:142 trigger rule 23 - Rule PUSHDIMDOWN_OFF
20:03:41:143 Replace task 2098 type 14 in queue cluster 0x0006 with newer task 2100 of same type. 4 runnig tasks
20:03:41:143 delayed group sending
20:03:41:143 delayed group sending
20:03:41:143 trigger rule 41 - Rule PUSHDIMUP_ON
20:03:41:143 Replace task 2100 type 14 in queue cluster 0x0006 with newer task 2102 of same type. 4 runnig tasks
20:03:41:144 delayed group sending
20:03:41:144 delayed group sending
20:03:41:157 delayed group sending
20:03:41:158 delayed group sending
20:03:41:161 delayed group sending
20:03:41:162 delayed group sending
20:03:41:169 delayed group sending
20:03:41:170 delayed group sending
20:03:41:224 ZCL attribute report 0x842E14FFFE37DEEB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:03:41:266 ZCL attribute report 0x842E14FFFE231B4D for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:03:41:509 [INFO] - Button 3002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x03, cluster: MULTISTATE_INPUT (0x0012), action: Dim down press, payload: 5500210100, zclSeq: 89
20:03:41:510 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
20:03:41:514 rule event /sensors/58/state/lastupdated: 0 -> 0
20:03:41:514 trigger rule 92 - Rule PUSHDIMUP_ON
20:03:41:515 trigger rule 95 - Rule PUSHDIMUP_ON
20:03:41:515 Replace task 2107 type 14 in queue cluster 0x0006 with newer task 2109 of same type. 1 runnig tasks
20:03:41:515 trigger rule 10 - Rule OFF
20:03:41:516 trigger rule 59 - Rule OFF
20:03:41:516 delayed group sending
20:03:41:516 trigger rule 76 - Rule PUSHDIMDOWN_OFF
20:03:41:516 delayed group sending
20:03:41:536 delayed group sending
20:03:41:540 delayed group sending
20:03:41:569 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:41:595 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:42:218 ZCL attribute report 0x842E14FFFE37DEEB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
**20:03:42:231 [INFO] - Button 4002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x04, cluster: MULTISTATE_INPUT (0x0012), action: Dim up press, payload: 5500210100, zclSeq: 90**
20:03:42:232 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x04, frame control: 0x18, mfcode: 0x0000 
20:03:42:241 ZCL attribute report 0x842E14FFFE231B4D for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:03:42:253 rule event /sensors/58/state/lastupdated: 0 -> 0
20:03:42:254 trigger rule 69 - Rule OFF
20:03:42:254 trigger rule 12 - Rule OFF
20:03:42:255 delayed group sending
20:03:42:255 trigger rule 44 - Rule PUSHDIMDOWN_OFF
20:03:42:256 delayed group sending
20:03:42:256 trigger rule 57 - Rule OFF
20:03:42:256 Replace task 2122 type 14 in queue cluster 0x0006 with newer task 2126 of same type. 2 runnig tasks
20:03:42:257 delayed group sending
20:03:42:257 trigger rule 64 - Rule PUSHDIMDOWN_OFF
20:03:42:258 delayed group sending
20:03:42:258 delayed group sending
20:03:42:258 trigger rule 96 - Rule PUSHDIMDOWN_OFF
20:03:42:259 Replace task 2128 type 14 in queue cluster 0x0006 with newer task 2130 of same type. 2 runnig tasks
20:03:42:259 delayed group sending
20:03:42:259 delayed group sending
20:03:42:270 delayed group sending
20:03:42:270 delayed group sending
20:03:42:273 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:42:273 delayed group sending
20:03:42:274 delayed group sending
20:03:42:277 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:42:277 delayed group sending
20:03:42:277 delayed group sending
20:03:42:386 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:42:395 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
20:03:43:070 sensor 40 (lumi.sensor_motion.aq2): disable presence
```

MrWGT, 2021-02-13:
Top-Left to bottom-right, more time between key presses:

```
20:14:05:939 Multi state present value: 0x0001 (1), lumi.remote.b486opcn01
20:14:05:940 [INFO] - Button 1002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), action: Off press, payload: 5500210100, zclSeq: 92
20:14:05:942 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:14:05:969 rule event /sensors/58/state/lastupdated: 0 -> 0
20:14:05:971 trigger rule 70 - Rule ON
20:14:05:972 trigger rule 26 - Rule ON
20:14:05:973 delayed group sending
20:14:05:974 trigger rule 82 - Rule PUSHDIMUP_ON
20:14:05:975 Replace task 5271 type 14 in queue cluster 0x0006 with newer task 5273 of same type. 1 runnig tasks
20:14:05:976 delayed group sending
20:14:05:992 delayed group sending
20:14:05:997 delayed group sending
20:14:09:741 [INFO] - Button 2002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x02, cluster: MULTISTATE_INPUT (0x0012), action: On press, payload: 5500210100, zclSeq: 93
20:14:09:742 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
20:14:09:756 rule event /sensors/58/state/lastupdated: 0 -> 0
20:14:09:758 trigger rule 73 - Rule ON
20:14:09:759 trigger rule 17 - Rule PUSHDIMDOWN_OFF
20:14:09:760 trigger rule 23 - Rule PUSHDIMDOWN_OFF
20:14:09:761 delayed group sending
20:14:09:762 trigger rule 41 - Rule PUSHDIMUP_ON
20:14:09:762 Replace task 5296 type 14 in queue cluster 0x0006 with newer task 5298 of same type. 2 runnig tasks
20:14:09:763 delayed group sending
20:14:09:783 delayed group sending
20:14:09:794 delayed group sending
20:14:09:800 delayed group sending
20:14:12:324 [INFO] - Button 3002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x03, cluster: MULTISTATE_INPUT (0x0012), action: Dim down press, payload: 5500210100, zclSeq: 94
20:14:12:325 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
20:14:12:334 rule event /sensors/58/state/lastupdated: 0 -> 0
20:14:12:335 trigger rule 92 - Rule PUSHDIMUP_ON
20:14:12:335 trigger rule 95 - Rule PUSHDIMUP_ON
20:14:12:335 trigger rule 10 - Rule OFF
20:14:12:335 trigger rule 59 - Rule OFF
20:14:12:336 delayed group sending
20:14:12:336 trigger rule 76 - Rule PUSHDIMDOWN_OFF
20:14:12:336 delayed group sending
20:14:12:336 delayed group sending
20:14:12:348 delayed group sending
20:14:12:349 delayed group sending
20:14:12:365 delayed group sending
20:14:12:365 delayed group sending
20:14:12:370 delayed group sending
20:14:12:370 delayed group sending
20:14:12:375 delayed group sending
20:14:12:376 delayed group sending
20:14:12:461 ZCL attribute report 0x842E14FFFE231B4D for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:14:12:466 ZCL attribute report 0x842E14FFFE37DEEB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:14:13:167 [INFO] - No button map for: TH01, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0402, command: 0x0A, payload: 000029AF07, zclSeq: 223
20:14:13:167 ZCL attribute report 0x00124B0022417D70 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:14:14:988 [INFO] - Button 4002 - lumi.remote.b486opcn01, unicast to: 0x0000, endpoint: 0x04, cluster: MULTISTATE_INPUT (0x0012), action: Dim up press, payload: 5500210100, zclSeq: 95
20:14:14:988 ZCL attribute report 0x04CF8CDF3C7915D3 for cluster: 0x0012, ep: 0x04, frame control: 0x18, mfcode: 0x0000 
20:14:14:995 rule event /sensors/58/state/lastupdated: 0 -> 0
20:14:14:996 trigger rule 69 - Rule OFF
20:14:14:996 trigger rule 12 - Rule OFF
20:14:14:997 delayed group sending
20:14:14:997 trigger rule 44 - Rule PUSHDIMDOWN_OFF
20:14:14:997 delayed group sending
20:14:14:998 trigger rule 57 - Rule OFF
20:14:14:998 Replace task 5339 type 14 in queue cluster 0x0006 with newer task 5343 of same type. 2 runnig tasks
20:14:14:999 delayed group sending
20:14:14:999 trigger rule 64 - Rule PUSHDIMDOWN_OFF
20:14:14:000 delayed group sending
20:14:14:000 delayed group sending
20:14:14:001 trigger rule 96 - Rule PUSHDIMDOWN_OFF
20:14:14:001 Replace task 5345 type 14 in queue cluster 0x0006 with newer task 5347 of same type. 2 runnig tasks
20:14:14:001 delayed group sending
20:14:14:002 delayed group sending
20:14:15:018 delayed group sending
20:14:15:019 delayed group sending
20:14:15:022 delayed group sending
20:14:15:023 delayed group sending
20:14:15:123 ZCL attribute report 0x842E14FFFE37DEEB for cluster: 0x0300, ep: 0x01, frame control: 0x18, mfcode: 0x0000
```

closed 2021-03-16 by github-actions[bot]