dresden-elektronik/deconz-rest-plugin #3796: Heiman gas sensor HS1CG-E with new model id GASSensor-EF-3.0

adolfomendes, 2020-12-03:
# Device

- Product name: Gas sensor HS1CG-E 
- Manufacturer: Heiman
- Model identifier: GASSensor-EF-3.0
- Device type : Sensor

## Screenshots

[image]
[image]

SwoopX, 2020-12-12:
Closing this one because a) supported and b) incomplete device request.

closed 2020-12-12 by SwoopX

centauri, 2020-12-22:
> Closing this one because a) supported and b) incomplete device request.

Does this mean it is supported see image below?
Version: 2.07.01 
Firmware: 26680700
[image]

SwoopX, 2020-12-23:
This looks like an incorrectly paired device. How have you paired it in Phoscon, via light or sensor search?

Anyway, I'd suggest to remove the device from power, start sensor search in Phoscon and plug it in again. If you're really unlucky, this device is one of the edge cases, which my IAS zone refactor should take care of in future during pairing...

centauri, 2020-12-23:
> This looks like an incorrectly paired device. How have you paired it in Phoscon, via light or sensor search?
>
> Anyway, I'd suggest to remove the device from power, start sensor search in Phoscon and [...]

Thank you for the reply. I did a "add new sensor" again but this time i kept on pressing the test button every second or so until the text "Gas sensor" appeared in deconz and then it also appeared in phoscon.