dresden-elektronik/deconz-rest-plugin #3797: Not able my 3th BlitzWolf BW-SHP13 causing not showing lights in GUI

canedje, 2020-12-03:
## Describe the bug

After trying 100 times, not able to pair. The 2 other ones are pairing oke.
If I try to pair at lights the BlitWolf it is not adding a device.
In VNC there is  a new device showing:
After restart the Deconz Docker after some tries to pair the device: at my Phoscon/ Deconz app is not showing the lights anymore in the LLight GUI chapter
Switches and Sensors are still there.
I do need to load a backup to let the app shwowing the lights back in the GUI again.
Log:
```
Still not able to pair.
If I try to pair at my Phoscon/ Deconz app is not showing the lights anymore.
Switches and Sensors are still there.
I do need to load a backup to let the app shwowing the lights back in the GUI again.
Log:
ErrorWarningSystemArrayLogin

15:21:07:521 ZCL configure reporting rsp seq: 210 0x000B57FFFED8BF8C for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
15:21:07:524 0x000B57FFFED48CF6 force poll (2)
15:21:07:525 delay sending request 237 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:07:525 delay sending request 243 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:07:525 delay sending request 244 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:07:542 0x000B57FFFEDC18B0 force poll (2)
15:21:07:542 delay sending request 237 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:07:545 delay sending request 237 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:07:545 delay sending request 244 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:07:567 0x000B57FFFED8BF8C force poll (2)
15:21:07:625 0x000B57FFFEDC18B0 force poll (2)
15:21:07:647 Bind response success for 0x000b57fffed48cf6 ep: 0x01 cluster: 0x0300
15:21:07:647 skip configure report for cluster: 0x0300 attr: 0x0007 of node 0x000B57FFFED48CF6 (seems to be active)
15:21:07:647 skip configure report for cluster: 0x0300 attr: 0x0003 of node 0x000B57FFFED48CF6 (wait reading or unsupported)
15:21:07:647 skip configure report for cluster: 0x0300 attr: 0x0004 of node 0x000B57FFFED48CF6 (wait reading or unsupported)
15:21:07:648 skip configure report for cluster: 0x0300 attr: 0x0008 of node 0x000B57FFFED48CF6 (seems to be active)
15:21:07:670 0x000B57FFFEDC18B0 force poll (2)
15:21:07:674 0x000B57FFFED8BF8C force poll (2)
15:21:08:679 ZCL attribute report 0x04CF8CDF3C78B547 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
15:21:08:697 send move to color temperature 450 to 0x14B457FFFE2D7975
15:21:08:732 0x14B457FFFE2D7975 force poll (2)
15:21:08:745 delay sending request 4 dt 0 ms to 0x14B457FFFE2D7975, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:08:769 0x14B457FFFE2D7975 force poll (2)
15:21:08:796 0x14B457FFFE2D7975 force poll (2)
15:21:08:801 send move to color temperature 450 to 0x000B57FFFEDBFBC6
15:21:08:842 0x000B57FFFEDBFBC6 force poll (2)
15:21:08:845 delay sending request 10 dt 0 ms to 0x000B57FFFEDBFBC6, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:08:882 0x000B57FFFEDBFBC6 force poll (2)
15:21:08:897 send move to color temperature 450 to 0x000B57FFFED48CF6
15:21:08:928 0x000B57FFFEDBFBC6 force poll (2)
15:21:08:928 delay sending request 13 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:08:928 delay sending request 14 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:08:945 delay sending request 13 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:08:946 delay sending request 14 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:08:978 0x000B57FFFED48CF6 force poll (2)
15:21:08:997 send move to color temperature 450 to 0x000B57FFFED8BF8C
15:21:08:998 delay sending request 14 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:045 delay sending request 14 dt 0 ms to 0x000B57FFFED48CF6, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:045 delay sending request 18 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:09:045 delay sending request 19 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:055 0x000B57FFFED48CF6 force poll (2)
15:21:09:082 0x000B57FFFED8BF8C force poll (2)
15:21:09:100 send move to color temperature 450 to 0x000B57FFFEDC18B0
15:21:09:100 delay sending request 19 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:133 0x000B57FFFED48CF6 force poll (2)
15:21:09:133 delay sending request 19 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:133 delay sending request 23 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:09:133 delay sending request 24 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:145 delay sending request 19 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:146 delay sending request 23 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0008 onAir: 1
15:21:09:146 delay sending request 24 dt 0 ms to 0x000B57FFFEDC18B0, ep: 0x01 cluster: 0x0300 onAir: 1
15:21:09:153 0x000B57FFFEDC18B0 force poll (2)
15:21:09:154 delay sending request 19 dt 0 ms to 0x000B57FFFED8BF8C, ep: 0x01 cluster: 0x0300 onAir: 1
[... 756 lines cut ...]

15:36:03:455 ZCL attribute report 0x04CF8CDF3C78B547 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
```
## Steps to reproduce the behavior

## Expected behavior

## Screenshots

added devices in VNC
[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

canedje, 2020-12-03:
Becasue of not showing in the "connected devices" is it not possible to remove the switch from the network.
I also don't know how to reset the BlitzWolf (Not mentioned somewhere). How can I get the BlitzWolf out of the Zigbee network?
It is still showing up in the VNC even after deleting the node

Mimiix, 2020-12-03:
Is this a duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3788 ?

canedje, 2020-12-03:
No there the problem was that the devices didn't show the power consumption. In the beginning I could pair both devices bud they didn't show power consumption.
After a day I removed the devices, and re-paired one device and it was showing power what it didn't before.
After that I  tried to re-pair the second device also, bud from then on it is not possible anymore to pair.

I did mention this situation at the end of that threath, but because I think it is a different problem I did make a new issue.
Is that right to do so?

SwoopX, 2020-12-03:
> It is showing up "half" in VNC with two(?) nodes, but not in the GUI "devices connected"

What does that mean? Not sure what you want to express here.

When the plug is not creating the power and consumption sensors, it typically helps to start sensor search in Phoscon and apply a load to the plug.

I got the feeling here that you put like 3 different topics into one, so please one issue per topic. Regardless, can you please share a screenshot of the node info panel for the troublesome devices?

[image]

canedje, 2020-12-04:
This topic is only about not able to pair the Blitzwolf anymore. 
It is showing up in VNC with 2 nodes but not in the Phoscon GUI
At trying to pair, it also create a kind of crash in the GUI not showing lights anymore in "light connected"
I will add node info in a moment ( I removed the device)

canedje, 2020-12-04:
Is it possible to add a button to the Phoscon app "Remove device" and then put the devcice in pairing mode, but instead of pairing removing the device from the network?
Because now there is no posibility to remove a device from the network if it is not vissible in the GUI?

canedje, 2020-12-05:
Supprise supprise.
This morning I did connect the Blitzwolf to a wallplug. Not touching it only connect to the power.
I opened the phoscon app and did do push the “add a new light” button at the lights chapter
I did not put the blitzwolf in pair mode.
Suddenly in about a minute the Blitzwolf popped up n the GUI.
It also appeared the right way in the VNC.
So now it is working. I really don’t how this is possible, but I’m happy

closed 2020-12-05 by canedje