dresden-elektronik/deconz-rest-plugin #3798: Plugs (develco 131) not always detected

Rania-M, 2020-12-03:
## Device

- Product name: Develco SP131

- Device type : Plug/light

Hi, i've been using deconz api with develco plugs, it seems that the devices detection is not always reliable, as for multiple plugs having the same model doesn't always get recognized.
I usually put the devices in the same configuration and some of them get detected but some doesn't,, given that they are from the same manufacturer, i wonder if this is a bug from deconz or some mistake i made.
If anyone had the same issue or has an idea of what is going on, please let me know.
Thank you

closed 2020-12-03 by close-issue-app[bot]

Rania-M, 2020-12-03:
Hi, thank you for your response, however, i can not provide any screenshot as the subject of my problem is that my devices are not recognized, i'm used to participate in this git, if you can give me a way to ask this question without screenshots please let me know

Mimiix, 2020-12-07:
_- all given header lines (starting with ###) must not be deleted_