dresden-elektronik/deconz-rest-plugin #3800: Devices not always recognized

Rania-M, 2020-12-03:
## Describe the bug
Develco plug 131
Product name: Develco SP131

Device type : Plug/light

Hi, i've been using deconz api with develco plugs, it seems that the devices detection is not always reliable, as for multiple plugs having the same model doesn't always get recognized.

## Steps to reproduce the behavior

I usually put the devices in the same configuration and some of them get detected but some doesn't,, given that they are from the same manufacturer, i wonder if this is a bug from deconz or some mistake i made.
If anyone had the same issue or has an idea of what is going on, please let me know.
Thank you

## Expected behavior

## Screenshots
none

## Environment
Host system: Raspberry Pi
Running method: Raspbian
Firmware version: 0x0000000
deCONZ version: 2.05.88
Device: ConBee II / RaspBee I
Do you use an USB extension cable: no

## deCONZ Logs
none

## Additional context
none

Smanar, 2020-12-03:
Hello, what is the manufacture ?
Problem are generally for sensors.

SwoopX, 2020-12-03:
@Smanar Develco SPLZB-131

So what's the issue now? Reset and re-pair the devices.

Rania-M, 2020-12-04:
> Hello, what is the manufacture ?
> Problem are generally for sensors.

Hi, the manufacturer name is Develco Products A/S

> @Smanar Develco SPLZB-131
> 
> So what's the issue now? Reset and re-pair the devices.

Hi, i already did that as mentionned before, but still it works sometimes but not always (with the same devices !)

Smanar, 2020-12-04:
Yep, sorry, bad reading ^^

closed 2021-02-08 by Mimiix