dresden-elektronik/deconz-rest-plugin #3804: Rest API should not produce illegal characters

eweinell, 2020-12-03:
## Describe the bug
At least for one manufacturer ID, deconz-rest-api produces invalid json responses containing illegal control characters. According to VNC Config, manufacturer ID in question is 0x1002. The Rest API yields the character sequence 0x10 0xef 0xbf 0xbd 0x2e 0x01, with the first and last character interfering with json parsing. The rest is glibberish.

As side effect, the Deconz PWA stops working (JS errors out)

I think deconz-rest-plugin should always mask invalid characters before writing responses.

## Steps to reproduce the behavior

Curl REST API for this specific device with man-code 0x1002 and parse the response
curl http://{{host}}/api/{{key}}/lights/{{device}} | jq .

## Expected behavior

## Screenshots

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container 
 - Firmware version: 26680700
 - deCONZ version: 2.07.00
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? no

## deCONZ Logs

ebaauw, 2020-12-03:
Can you post the raw (not json-parsed) response to the `curl`?  And a GUI screenshot of the _Basic_ cluster attributes (after reading them).

SwoopX, 2020-12-03:
So far, I've seen something comparable only with iHorn temp./hum. sensors and that is more or less caught in the code

[image]

eweinell, 2020-12-03:
Here's the REST response (hexdumped)
```
00000000  7b 22 65 74 61 67 22 3a  22 66 62 34 35 36 39 32  |{"etag":"fb45692|
00000010  61 30 31 35 33 63 30 30  38 33 62 38 65 32 37 38  |a0153c0083b8e278|
00000020  35 31 64 33 65 34 63 36  66 22 2c 22 68 61 73 63  |51d3e4c6f","hasc|
00000030  6f 6c 6f 72 22 3a 66 61  6c 73 65 2c 22 6c 61 73  |olor":false,"las|
00000040  74 61 6e 6e 6f 75 6e 63  65 64 22 3a 22 32 30 32  |tannounced":"202|
00000050  30 2d 31 32 2d 30 33 54  31 35 3a 30 34 3a 32 34  |0-12-03T15:04:24|
00000060  5a 22 2c 22 6c 61 73 74  73 65 65 6e 22 3a 22 32  |Z","lastseen":"2|
00000070  30 32 30 2d 31 32 2d 30  33 54 32 31 3a 34 32 5a  |020-12-03T21:42Z|
00000080  22 2c 22 6d 61 6e 75 66  61 63 74 75 72 65 72 6e  |","manufacturern|
00000090  61 6d 65 22 3a 22 10 ef  bf bd 2e 01 22 2c 22 6d  |ame":"......","m|
000000a0  6f 64 65 6c 69 64 22 3a  22 54 53 30 31 31 46 22  |odelid":"TS011F"|
000000b0  2c 22 6e 61 6d 65 22 3a  22 5a 42 5f 53 77 69 74  |,"name":"ZB_Swit|
000000c0  63 68 5f 30 31 22 2c 22  73 74 61 74 65 22 3a 7b  |ch_01","state":{|
000000d0  22 61 6c 65 72 74 22 3a  22 6e 6f 6e 65 22 2c 22  |"alert":"none","|
000000e0  6f 6e 22 3a 66 61 6c 73  65 2c 22 72 65 61 63 68  |on":false,"reach|
000000f0  61 62 6c 65 22 3a 74 72  75 65 7d 2c 22 73 77 76  |able":true},"swv|
00000100  65 72 73 69 6f 6e 22 3a  6e 75 6c 6c 2c 22 74 79  |ersion":null,"ty|
00000110  70 65 22 3a 22 4f 6e 2f  4f 66 66 20 70 6c 75 67  |pe":"On/Off plug|
00000120  2d 69 6e 20 75 6e 69 74  22 2c 22 75 6e 69 71 75  |-in unit","uniqu|
00000130  65 69 64 22 3a 22 38 34  3a 32 65 3a 31 34 3a 66  |eid":"84:2e:14:f|
00000140  66 3a 66 65 3a 66 32 3a  35 35 3a 31 32 2d 30 62  |f:fe:f2:55:12-0b|
00000150  22 7d                                             |"}|
00000152
```

...and the cluster info:
[image]

ebaauw, 2020-12-03:
The Zigbee _Manufacturer Name_ name looks valid.  I don’t know why `manufacturername` doesn’t reflect that.

SwoopX, 2020-12-04:
This is the Silvercrest plug, right? The manufacturer code as mentioned above is VENDOR_EMBER.

I'd assume this code passage could be involved
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/c0306356ad6a51c620b12aa59f74e78ca6698521/de_web_plugin.cpp#L2083-L2116

Smanar, 2020-12-04:
Ha yes, can be from this code

```
        "lastannounced": null,
        "lastseen": "2020-11-30T10:41Z",
        "manufacturername":"�P��ZU",
        "modelid": null,
        "name": "On/Off light 16",
```

I can have this kind of JSON too for tuya device.
It's a wired switch ? If yes I think it from the fake switches created for tuya one.

Smanar, 2020-12-04:
Yep, take a look after line starting with "//Make 2 fakes device for tuya switches"

eweinell, 2020-12-04:
Folks, after connecting a second one of these plugs, both show up normally in the PWA. 

REST API now show manufacturer name "_TZ3000_kdi2o9m6" which isnt terribly helpful, but at least leaves out any control characters. Interestingly enough I didn't even restart deconz. So whilst my actual problem seems gone, it still might be worthy to add some safeguard against malformed data here before it comes back to haunt me.

SwoopX, 2020-12-04:
Absolutely, and that's the only silly reason I bought that stuff :)

Appreciate raising this here!

Smanar, 2020-12-04:
And BTW, I m still interested by cluster list, to see how it look for tuya plug !
To see wich one cluster they use for power measurement for exemple.

SwoopX, 2020-12-04:
Here you go

[image]

Smanar, 2020-12-04:
So no power measurement or it use an hidden tuya cluster ? (visible in log)

Here the cluster for w switch
[image]

Good luck to reconize them ....

The 0xF2 endpoint is only for lidl ATM.

alinelena, 2020-12-07:
had the same bad character in manufacturer name for a lidl led strip... had to edit by hand the database to get rid of the offending node.

Smanar, 2020-12-08:
It s too late for you give me a log during the inclusion and the bad manufacture name ?

eweinell, 2020-12-09:
Im afraid so, because the device after a couple of failed attempts never again reported the bad name. So we could either close the issue, or take it is request to suppress faulty characters in the REST response. Also with the bad name, the device was working perfectly normal and could be controlled with the REST API.

Smanar, 2020-12-09:
I think you can close it, manup have corrected the bug for illegal character https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3838

SwoopX, 2020-12-09:
Closing this for the given reason above.

closed 2020-12-09 by SwoopX