dresden-elektronik/deconz-rest-plugin #3805: Websocket stops sending Sensor Events

diocloid, 2020-12-03:
## Describe the bug

**Issue Part A**
After updating to the newest version (from 2.05.77), At irregular time intervals (about 1h) the websocket will stop sending events for buttons pressed. There is no error in home assistant, or in the phoscon web app. It just looks like the devices haven’t sent updates.
In VNC, the deconz screen shows ‘A device firmware update is needed’, and gives me a button to press to start the update. 
The API is working fine at that point, even if the GUI cannot render the map.
After this event I cannot get deconz to send events via the websocket even if I restart the service and or the machine. Only a firmware flash fixes the issue. 
Before the device 

**Issue Part B**
During debugging I also noticed the following :
The firmware 0x26680700 was not changed in this experiment.
Deconz started with: deCONZ --http-port=80 --dbg-info=1 --dbg-aps=2 --dbg-zcl=1
All sensors are already paired (They got paired while running 2.05.88)

1. A fresh install of 2.05.77 sent all events to the websocket.
2. A fresh install of 2.05.78 did not send sensor "button press" events to websocket (Group, Lights, and Configuration Tool Events were sent)
3. A fresh install of 2.05.88 did not send sensor "button press" events to websocket (Group, Lights, and Configuration Tool Events were sent)

The sensor events are only sent after I remove and add the sensor 
[deconz_20588_removeandadd.ws.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639472/deconz_20588_removeandadd.ws.log)
[deconz_20588_removeandadd.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639473/deconz_20588_removeandadd.log)
After restarting deconz the websocket failes sending sensor events again.

I suspect that these parts are related.

## Steps to reproduce the behavior
Issue Part A
1. Install 2.05.88 on Raspbian
2. Start deconz-gui service and pair a Tradfri sensor (button)
3. Start listening to the websocket
4. After a period of time the websocket will stop sending events
5. The GUI will display  ‘A device firmware update is needed’
[Fix] 6. Stop deconz-gui service, Re-Flash Firmware, (Sometimes reboot), start deconz-gui service
[Alternative Fix] 7. Revert to 2.05.77

Issue Part B
1. Install 2.05.77 on Raspbian
2. Start deconz-gui and websocket listener
3. Observe working websocket communication
4. Stop deconz-gui and websocket listener
1. Install 2.05.88 on Raspbian
2. Start deconz-gui and websocket listener
3. Observe no websocket communication for sensors

## Expected behavior
Working websocket communication:
A. Even if a Firmware Update is found 
B. On newest Version

## Screenshots

## Environment
System:
 - Host system: Raspberry Pi armv7l GNU/Linux
 - Running method: Raspbian 
 - Firmware version: 0x26680700
 - deCONZ version: 2.05.88
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
.log -> deCONZ Logs
.ws.log -> Websocket listener log
[deconz_20588.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639287/deconz_20588.log)
[deconz_20588.ws.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639288/deconz_20588.ws.log)
[deconz_20577.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639289/deconz_20577.log)
[deconz_20577.ws.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639290/deconz_20577.ws.log)
[deconz_20577_1.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639291/deconz_20577_1.log)
[deconz_20587.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639292/deconz_20587.log)
[deconz_20587.ws.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5639293/deconz_20587.ws.log)

## Additional context
Devices: Tradfri Buttons and On/Off plug-in units

The only reference I found concerning this issue was [https://community.home-assistant.io/t/conbee-ii-deconz-stops-working-when-firmware-update-is-needed/178620](https://community.home-assistant.io/t/conbee-ii-deconz-stops-working-when-firmware-update-is-needed/178620) which has not been resolved.

diocloid, 2020-12-04:
Further analysis:
The missing websocket communication for sensors (switches) occurs when upgrading from 2.05.84 to 2.05.87.

Smanar, 2020-12-04:
On the not working you have that

```
22:05:46:383 New websocket 192.168.1.210:64214 (state: 3) 
22:05:46:649 New websocket 192.168.1.202:50052 (state: 3) 
22:05:46:774 New websocket 192.168.1.201:59812 (state: 3) 
```

SwoopX, 2020-12-04:
The clients are initializing the connection.

Smanar, 2020-12-04:
But he is using same clients on 2 tests.

diocloid, 2020-12-04:
@Smanar That is correct, I open the websocket connection from a nodejs testclient.
```
const WebSocket = require('ws');

const host = '192.168.1.202';
const port = 8088;

const ws = new WebSocket('ws://' + host + ':' + port);

ws.onmessage = function(msg) {
    console.log(JSON.parse(msg.data));
}
```

@Smanar 
The problem is not, that a websocket connection is not established. It is established every time  - See Issue Part B
The problem is, that the sensor events are not sent to the websocket. Light and Configuration Tool events are sent. - See the combination of deconz and websocket logs

majorsl, 2020-12-05:
Possibly similar to the issue I have? #3707

SwoopX, 2020-12-05:
So, only **sensors** stop sending events? Is that also regardless of the type or dedicated to some specific (e.g. switches)?

Smanar, 2020-12-05:
Funny bug ...

Just by curiosity what is the value of "websocketnotifyall" in the gateway config ? (you can see it in the API or on log at deconz start)

diocloid, 2020-12-05:
@Smanar it is true in both instances.
2.5.78 and 2.5.88
From the API: "websocketnotifyall": true,

diocloid, 2020-12-05:
@SwoopX That is correct. I am reading Events from
r: 'lights',
t: 'event',
and 
r: 'groups',
t: 'event'

But not 
r: 'sensors',
t: 'event',

Thanks for the good idea.
Until now I have only been able to test with 4 different buttons. I will try to test with a movement sensor.

diocloid, 2020-12-05:
@majorsl From what I understand reading through Issue #3707 mine seems to be very similar.
Different environment, so that elimiates hardware issues.
I am working of the stable releases in [deconz.dresden-elektronik.de/raspbian/stable/](http://deconz.dresden-elektronik.de/raspbian/stable/). So the last working version is 2.05.84. The next version 2.05.86 seems to have introduced the bug to stable.

PS.
For debugging I highly recommend running your own listener. (NodeJS Code: [Issue Reply](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3805#issuecomment-738945081) It eliminates errors from other programs such as Home Assistant.

diocloid, 2020-12-05:
> 
> 
> So, only **sensors** stop sending events? Is that also regardless of the type or dedicated to some specific (e.g. switches)?

I have now tested it with an "Ikea Tradfri Motion Sensor" aswell. Same behavior. Presence events are sent in version 2.05.84 but not 2.05.86

The event data I received in version 2.05.84:
`
{
  e: 'changed',
  id: '8',
  r: 'sensors',
  state: {
    dark: true,
    lastupdated: '2020-12-05T18:53:45.993',
    presence: true
  },
  t: 'event',
  uniqueid: '00:0b:57:ff:fe:9b:b1:c6-01-0006'
}
`

SwoopX, 2020-12-06:
Hm, that's very odd. Could you try bringing data side by side for comparison? Launched your test client code on the left and I tail the deconz debug output on the right

[image]

However, I run a bare ubuntu installation here. Debug output is generated by starting deconz as follows `/usr/bin/deCONZ --dbg-info=2 --dbg-error=2 > debug3`
I grep the lines of interest from the generated debug file with `tail -f debug3 | grep changed`

SwoopX, 2020-12-06:
**Update**

Downgraded my Conbee II firmware. The message on the gateway page was instantly there and all websocket events (including sensors)  are continuing to happily drip in.

manup, 2020-12-07:
Hmm that's odd the firmware shouldn't cause this.
Can you please provide a log with the latest firmware version where *Help &rarr; Debug view* &rarr; APS and INFO is checked?

The interesting part is if the APS indication is received when the motion sensor triggers.

diocloid, 2020-12-07:
An Update: As a semi-permanent, although unsatisfying, solution I moved the whole setup to new hardware with a fresh install of raspbian (deconz provided image)
In the new setup all events are sent via websocket as expected (with newest version of firmware and deconz) 
Therefore I suspect that, in the not-working setup, some setting or libriary got in a very wierd state. Thats the techical description 😄 

I will generate the log next weekend. As this now involves taking the network down and moving the raspbee hat to a different mashine.

closed 2021-02-08 by Mimiix