dresden-elektronik/deconz-rest-plugin #3806: Network stops - "A Firmware update is needed" - Update not possible

michi1g, 2020-12-04:
## Describe the bug
I updated to the lastest beta of the gui(2.07) and now I'm getting disconnected and prompted with "A Firmware update is needed" unfortunately pressing the "Update Firmware" Button doesn't do anything.
My entire network is unreachable until I restart the gui.

## Expected behavior
If there really is a firmware update available, it should just update.

## Screenshots
will provide one when the problem reoccurs.

## Environment

 - Host system: RasPi
 - Running method: Raspian
 - Firmware version: 26680700
 - deCONZ version: 2.07.00
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, a label printer Brother QL-700

## deCONZ Logs
No logs available. Will try to get a log when 
## Additional context
It's a mixed network with xiaomi, aqara, Eurotronics, Philips Hue and some noname devices.

LeoeLeoeL, 2020-12-04:
In this release it needs to be installed manually as described in the Wiki https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually .

Smanar, 2020-12-04:
He is on raspberry.
The manual method is not only for virtual machine ?

michi1g, 2020-12-05:
I can't update manually as there is newer firmware available...

Smanar, 2020-12-05:
If you make the manual procedure, you can use the same firmware you already have, it s just command line, and if it don't work show the error message pls.

diocloid, 2020-12-05:
May be of no help, but I am experiencing the same issue. API calls keep working for me, but no events are sent.
My Environment
    Host system: Raspberry Pi armv7l GNU/Linux
    Running method: Raspbian
    Firmware version: 0x26680700
    deCONZ version: 2.05.88
    Device: RaspBee II
    Do you use an USB extension cable: no
    Is there any other USB or serial devices connected to the host system? no

I have also found a reference to the same Problem here: [https://community.home-assistant.io/t/conbee-ii-deconz-stops-working-when-firmware-update-is-needed/178620](https://community.home-assistant.io/t/conbee-ii-deconz-stops-working-when-firmware-update-is-needed/178620) It sadly has not been resolved.

thomasmaxson, 2020-12-07:
I am having the exact same problem. Manual update "works", but after a short time (less than 10 minutes), I "not-connected" and am prompted for an update.

Smanar, 2020-12-07:
And you have the problem too on bigger hardware like a PC ?

closed 2021-02-08 by Mimiix