dresden-elektronik/deconz-rest-plugin #3807: Raspbee 2 Installation issue on Raspberry Pi 4B

gaetanentz, 2020-12-04:
Hi Guys,

I'm facing an issue installing my Raspbee 2 using the official Install :
[https://phoscon.de/en/raspbee2/install](url)

## Describe the question or issue you are having

Everything goes well until the 'make' instruction.
Something's going wrong with the file directory and I'm getting following error:

make[1]: *** /lib/modules/5.4.72-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:17: build] Error 2

## Screenshots

  If applicable, add screenshots to help explain the problem.
[image]

As I'm a newbie in all these things, I do not know how to fix it.

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian
 - Firmware version: (26xxyy00)
 - Device: RaspBee II

derkahless, 2020-12-04:
Try to reboot prior make.

That happens after a upgrade/dist-upgrade with newer kernel without reboot.

gaetanentz, 2020-12-04:
Also tried that but did not work.
In fact I add 42 things upgradeble ==> I did a full upgrade, reboot and then it worked

gaetanentz, 2020-12-04:
Everything was not up to date.
sudo apt full-upgrade 
And worked fine

closed 2020-12-04 by gaetanentz