dresden-elektronik/deconz-rest-plugin #3811: Pairing Tradfri driver with deCONZ AND Tradfri dimmer

1100101, 2020-12-04:
## Describe the question or issue you are having
I have a TRADFRI transformer 30W (Firmware v1.2.245) that was paired to a TRÅDFRI wireless dimmer (1.2.248), and was also listed in Phoscon. I could control the light either through deCONZ/Phoscon, and with the dimmer as well.

At some point the transformer was not controllable via Phoscon anymore, so I attempted to re-pair it with the application.
But now I'm at a loss as to how to get the same situation back as before:

 - If I hit the reset button on the transformer while searching for lights is active in Phoscon, the light shows up again and can be controlled, BUT I can't use the remote anymore.
 - If I pair the dimmer to the light (as per manual), I can control the light again with the dimmer, but Phoscon does not "see" it anymore.

I know this had worked before, but for the life of me I don't know how to get the old behavior back.

## Screenshots

## Environment

 
- Host system: Raspberry Pi 3B
- Running method: Raspbian Buster
- Firmware version: 26680700
- deCONZ version: 2.05.84, 2.05.88, 2.07.00 (tried all three versions)
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? Yes: Busware CULv3 868MHz

## deCONZ Logs

## Additional context

capstan1, 2020-12-08:
You have to pair both devices to phoscon via their sections (lights and controls), then add the switch as a control under the light-groups setting! 
When you pair them directly, they won't be connected to phoscon any more.

1100101, 2020-12-09:
Unfortunately that does not work: while the dimmer is detected as a switch, and reports changes via REST API, when I create a group and add a light,the dimmer is not available in the switch editor.

MattWestb, 2020-12-09:
@1100101 The switch editor is only for HA switches (That is sending click click) and not for light switches that is sending commands to lights (or "converting" lightsitches to HA ones).
If one light controlling device (the zigbee name for light witches / remotes) is paired OK the best is using the old "Wireless Light control" app there you is having one new groupe with the name of the switch and add the light to the group and all is working also then deCONZ is offline alla Zigbee Light standard.

MattWestb, 2020-12-09:
@1100101 Your devices is very out of date. Ikea have all controlling devices updated to zigbee 3 without the old motion sensor that sill is old lightlink (1.X) and your remote is 1.X.

1100101, 2020-12-10:
I’ve dug out the Tradfri gateway and connected the 30W driver as well as the dimmer to it. Works as expected, and it shows the dimmer as being updated (for over 4 hours now).

I’ll try using the wireless light control app.

One question: when pairing a tradfri dimmer, deCONZ says that „once pairing is successful, all lights paired to the remote will blink shortly“.

Is that still the case? I didn’t see any lights blinking.

1100101, 2020-12-22:
After updating the dimmer via Tradfri gateway, it works now as before.

case closed.

closed 2020-12-22 by 1100101