dresden-elektronik/deconz-rest-plugin #3816: ZHAOpenClose/Xiaomi Mijia Contact Sensor battery issue

ReviloEgros, 2020-12-04:
## Describe the question or issue you are having

I have a bunch of Xiaomi Mijia Contact sensors which are reporting the battery status every hour or so. I used this to determine if a sensor is alive or not regarding to the timestamp when the last battery status was received. Now i've switched from ConBee I to ConBee II because the casing of my ConBee I was broken . But now the battery status isn't transferred that frequently anymore, so my check_if_alive routine in FHEM is reportig all sensors as dead which they aren't. So my question is, Is this a bug or a feature?

## Screenshots

## Environment

 - Host system: PC
 - Running method: Ubuntu
 - Firmware version: 26680700
 - deCONZ version: 2.07.00
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? JeeLink, CUL

## deCONZ Logs

## Additional context

Smanar, 2020-12-04:
IDK, device are working same on conbee 1 and conbee2.
No other update ?
Since how many time have you switched ?

On phoscon what is the last "Refreshed" time ?

SwoopX, 2020-12-12:
@Smanar I guess you mean `lastseen`. This is exposed and picked up by FHEM, but for sensors it is not properly implemented yet.

I pretty much do the same, but on the income of temperature readings. Also giving them a grace time of 120 mins before I mark them as lost, which grants a very constant temperature over that period + one missed hourly "default" report.

closed 2021-03-05 by github-actions[bot]