dresden-elektronik/deconz-rest-plugin #3818: What is the state of IKEA Kadrilj or IKEA Fyrtur working on Decoz 2.05.88?

fribse, 2020-12-06:
## Describe the question or issue you are having

## Describe the question or issue you are having

I have a few IKEA curtains, I had tried to attach them to deconz a while back, and only the curtain contacting the deconz stick directly worked, and that failed after a deconz restart.
So now I've had to move them back to the IKEA hub which is terribly unstable, and the latest attempt is that I've bought a smartthings hub to use for the IKEA curtains, but that is not a good solution either, they stay connected, but does not report the 'degree of open', and uses the cloud.
I would very much like to move the IKEA extender, curtain and button BACK to deconz, but I need to know if it's stable or not?

## Screenshots

None

## Environment

 - Host system: Intel NUC w. PVE
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: yes - USB3
 - Is there any other USB or serial devices connected to the host system? Yes, AEOTECH Z-stick Gen 5 Mark 1

## deCONZ Logs

None

## Additional context

MattWestb, 2020-12-06:
The concept IKEA have made is that the signal repeater is acting as one relay for the commands sent from the remote and "caching" them to the blinds that is one end device (not one router).
This is made by binding cluster to one group then pairing the devices with touchlink.
Then pairing in "classical mode" (with Trust Center( TC)) its not possible getting the signal repeater bonded to one group and its with un luck not relaying commands to the blindes.

Have the blinds direct connected to the coordinator is very bad then its losing the functionality then its off line (updates, broken SD-Card).

Is connected to the signal repeater or some other router in the network is very possible its working good also then the coordinator is offline.

I think the best practice is first adding the signal repeater then the remote to it (paring it very near and hope its using it as its parent) and then the blinds also very near the signal repeater and hope its using it as its parent. Then doing the group binding for the devices to the same group.

For the moment its not possible forcing deCONZ pairing devices thru one specific devices (ZHA can) and also devices can changing its parent then they is liking.

I have sending one friendly request to IKEA for implanting the groupe binding of the signal repeater and 5 button remote for TC joined devices for fixing this problem (and the scenes for the remote) but have not getting any response.

fribse, 2020-12-07:
OK, great info @MattWestb , thankyou very much.
I added 1 blind now, so now I have one on IKEA, one on Smartthings, and one in deconz :-D
So far the deconz thing is behaving properly.

fribse, 2020-12-07:
Darned, I tried moving one more, and despite me holding the blind clean up to the ikea range extender, it chose another router :-(
Works so far though.

MattWestb, 2020-12-08:
The problem if it having deCONZ as router and deCONZ it is going offline then its not working until it have finding one new parent. If some other router is working normally (not losing its children) it should working OK.

Then working with radio waves its not always nearest / strongest that is best (Mother nature is making much bad thing here) ;-)

frassefraze, 2020-12-09:
> Darned, I tried moving one more, and despite me holding the blind clean up to the ikea range extender, it chose another router :-(
> Works so far though.

I have two blinds and experienced the same issues. One worked just fine, the other one wouldn't connect to the extender. However this time my stubbornness won the fight. I tried over and over again, letting it rest without the battery for some time (possibly not related) and finally it worked.  

But at first my blinds would report back their state/position, if I looked in the cluster info the data got updated, but not in Phoscon. I could only control them by binding them to a group/remote. I got the feeling it started working after I did a get request of http://x.x.x.x:40850/api/XYZXYZXYZ/lights, once again possibly not related, but that was the last thing I did before it started working.

fribse, 2020-12-10:
Well, the router it chose is a bit further away (4 m) than the repeater which is right beside it (1 m), but the router is a battery backed up sirene, so that's probably a super router for it, even in a power outage 😄 

I got the curtains to update by binding them as @ebaauw suggested in another thread, and then also remove the battery for 10s, and replace it.
So they now update the position perfectly.

Oddly though, it seems that the curtains sometimes falls asleep somehow, and doesn't react on commands, but after being awakenned, they react again, that is quite odd.

fribse, 2020-12-12:
Thankyou for all your help @MattWestb and @frassefraze 

The best procedure I've found was:

Make sure all three items are close. When adding the button and the cover, place the range extender right next to it, and away from other zigbee routers to try and get the items paired directly to the extender. This is the most tricky part, and can require multiple attempts. The problem is also that the deconz vnc view is not updated immediately with routes established, so it's something that can take several days.

1. Add the Range Extender as a light by resetting it (pin hole for 10 seconds)
2. While still searching for lights, go to Deconz, and 'Read' all the cluster-info's out, make sure that it updates each of the entries (might require multiple reads).
3. When everything is done, end the searhing for lights.
4. Add the button as a button by resetting it (four button presses within 5 seconds)
5. Go to Deconz and 'Read' all the cluster-info, especially the power configuration, it should grey out the 'no battery', if it does not, activate the button by pressing one of the directions, it probably requires multiple read's to have it all populated.
6. Add the cover as a light by resetting it (holding down the two buttons for 10 seconds or something like that until it starts flashing).
7. While still searching for lights, go to Deconz, and 'Read' all the cluster-info's out, make sure that it updates each of the entries (might require multiple reads). You might need to activate the blind by pressing the up or down to get all info read out.
8. When everything is done, end the searhing for lights.
9. Go to Deconz, open the binding panel, and bind the 'Windows Covering' (0102) as a source, and the Conbee II's 0001 as the destination.
10. When that is shown with success, remove the battery from the cover, and reinsert it.

Now it is paired properly, and will report position correctly.

closed 2020-12-12 by fribse