dresden-elektronik/deconz-rest-plugin #3819: Sonoff SNZB-03 Model: 66666

Alissh, 2020-12-06:
Hi,
I want to request an other Model identifier as #2991 --> this issue have Model identifier: MS01
new/other Model identifier: 66666 with some new attributes-Ids (Basic Cluster 0x0014)

## Device

- Product name: Sonoff SNZB-03.
- Manufacturer: eWeLink.
- Model identifier: 66666.
- Device type :
  - Sensor (Motion Sensor)

## Screenshots
Node-View:
[image]

Node-Info:
[image]

Basic:
[image]

Identify:
[image]

IAS Zone:
[image]
Power Configuration:
[image]

SwoopX, 2020-12-06:
Supported with 2.07.00-beta

closed 2020-12-06 by SwoopX

Alissh, 2020-12-06:
@SwoopX: Thanx for your fast responds.

deCONZ support: Model identifier: MS01
But it looks like deCONZ have some problems with 
Model identifier: 66666
or did i something wrong in the "configuration"?

Regards

SwoopX, 2020-12-06:
> But it looks like deCONZ have some problems with
> Model identifier: 66666

What are those problems?

Alissh, 2020-12-06:
i did the Steps, how i got the TRV Eurotronic Spirit to Phoscon:
- Start "Add sensor" in Phoscon
- go to deCONZ and do Cluster basic read
- Wait for time-out in Phoscon
- do "New try" in Phoscon

but i don't see the Sensor in Phoscon.
Only in deCONZ and in the Home Assistant appears a new "unknown Device".

is this not the same issue as #3300 but now with new ModelID: 66666? (compared Basic Cluster screenshots above and in #2991)

Or do you have some other idea what i can do to get the sensor work ?

Regards

SwoopX, 2020-12-07:
How does the device in deconz GUI look like

Alissh, 2020-12-07:
you mean this view with your question?

[image]

Alissh, 2020-12-07:
if i understand the merge from #3300 a little then you do a translating for the "Model Identifier" to a correct REST-API Device-Type-mapping.
I think this is missing for this type of Sonoff SNZB-03 sensor, someone set the model Identifier to 66666 and not correctly to MS01 or MSO1 or ms01, before it left the manufacturer.

SwoopX, 2020-12-07:
This is just universally stupid from Sonoff...

Regardless, have you tried a device reset and re-pairing it already?

Alissh, 2020-12-07:
Yes i have reset it more then 10 times and re-pairing it. No result.

Now the Sensor have a Green "Link" in deCONZ but doesn't react any more:
[image]
Error-Log: 
```19:49:02:639 ZCL read cluster: 0x0000, attribute: 0x0000
19:49:02:640 ZCL read cluster: 0x0000, attribute: 0x0001
19:49:02:640 ZCL read cluster: 0x0000, attribute: 0x0002
19:49:02:641 ZCL read cluster: 0x0000, attribute: 0x0003
19:49:02:642 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:642 APS-DATA.request id: 90, addrmode: 0x02, addr: 0x1f48, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 0 len: 11 tx.options 0x00
19:49:02:643 ZCL read cluster: 0x0000, attribute: 0x0004
19:49:02:644 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:645 APS-DATA.request id: 91, addrmode: 0x02, addr: 0x1f48, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
19:49:02:645 Delay APS request diff = 2
19:49:02:646 ZCL read cluster: 0x0000, attribute: 0x0005
19:49:02:646 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:648 APS-DATA.request id: 92, addrmode: 0x02, addr: 0x1f48, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
19:49:02:649 Delay APS request diff = 6
19:49:02:649 Delay APS request diff = 6
19:49:02:650 ZCL read cluster: 0x0000, attribute: 0x0006
19:49:02:650 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:651 APS-DATA.request id: 93, addrmode: 0x02, addr: 0x1f48, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x00
19:49:02:651 Delay APS request diff = 8
19:49:02:652 Delay APS request diff = 9
19:49:02:652 Delay APS request diff = 9
19:49:02:653 ZCL read cluster: 0x0000, attribute: 0x0007
19:49:02:653 ZCL read cluster: 0x0000, attribute: 0x0008
19:49:02:654 ZCL read cluster: 0x0000, attribute: 0x0009
19:49:02:654 ZCL read cluster: 0x0000, attribute: 0x000A
19:49:02:654 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:655 reject aps request to enddevice node queue is full (4)
19:49:02:655 ZCL read cluster: 0x0000, attribute: 0x4000
19:49:02:656 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:657 reject aps request to enddevice node queue is full (4)
19:49:02:658 ZCL read cluster: 0x0000, attribute: 0x0010
19:49:02:659 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:659 reject aps request to enddevice node queue is full (4)
19:49:02:660 ZCL read cluster: 0x0000, attribute: 0x0011
19:49:02:660 ZCL read cluster: 0x0000, attribute: 0x0012
19:49:02:661 ZCL read cluster: 0x0000, attribute: 0x0013
19:49:02:661 ZCL read cluster: 0x0000, attribute: 0x0014
19:49:02:662 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:662 reject aps request to enddevice node queue is full (4)
19:49:02:663 ZCL read cluster: 0x0000, attribute: 0xFFFD
19:49:02:663 ZCL cmd-req nwk: 0x1F48, profile: 0x0104, cluster: 0x0000 cmd: 0x00
19:49:02:664 reject aps request to enddevice node queue is full (4)
19:49:02:732 APS-DATA.confirm id: 90, status: 0xE9 MAC_NO_ACK
19:49:02:968 Delay APS request diff = 325
19:49:02:969 Delay APS request diff = 326
19:49:02:971 Delay APS request diff = 327
19:49:02:996 Delay APS request diff = 353
19:49:02:997 Delay APS request diff = 354
19:49:02:998 Delay APS request diff = 355
19:49:03:294 Delay APS request diff = 36
19:49:03:296 Delay APS request diff = 38
19:49:03:317 APS-DATA.confirm id: 91, status: 0xE9 MAC_NO_ACK
19:49:03:583 Delay APS request diff = 325
19:49:03:584 Delay APS request diff = 326
19:49:03:894 Delay APS request diff = 50
19:49:04:099 Delay APS request diff = 255
```

Alissh, 2020-12-09:
@SwoopX it looks like, the communication "Delay" is, because of the "Only react every 60 seconds".
Now I only need the "Template" from the REST-API to get the sensor as Entity in Phoscon and Home-Asistent.