dresden-elektronik/deconz-rest-plugin #3820: Eurotronic Spirit, lost connection

Alissh, 2020-12-06:
## Describe the question or issue you are having

Hello,

I am using the Eurotronic Spirit #1098 in deCONZ and since 2.05.88 i can set the temperature without any problem, Thanx.

But the Thermostat <--> ConBee II lose the connection.
On the display of the thermostat i see the "i have a Connection" but in deCONZ the green-line to the node is missing and no Request get to the thermostat (red blinking on node thermostat).
When i pull the battery out and in and do the InS(Install) on the thermostat, the communication-links comes back and i can control it.

Now my Question:
Did i something wrong in the "Communication settings" between thermostat and deCONZ, is it an issue from the Thermostat or an issue from the ConBee II Firmware?
Some Idea what i can do, to get the device always reachable for the ConBee II / deCONZ?

Regards

## Screenshots
"Default device Request" Screenshots from the thermostat when it is reachable:
Node Info:
[image]
Basic:
[image]
Power Configuration:
[image]
Identify:
[image]
Thermostat:
[image]
[image]
[image]
[image]
[image]
[image]
Time:
[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: no

## deCONZ Logs

Sorry, no logs

## Additional context

ebaauw, 2020-12-06:
I assume the communication from the gateway to the TRV has been lost, but when you update the thermostat setting manually, it's still reflected by the gateway?

I've had this issue for years, with TRVs randomly becoming unreachable from the gateway, but happily continuing to report. I'm pretty sure it's an incompatibility between the parent router firmware and the TRV firmware.  The parent router has disowned the TRV, but the TRV still assumes it has a working parent.

Next to power cycling the TRV, power cycling or forcing a leave/rejoin of (pressing `0` in the GUI) the router that the TRV mistakingly takes for its parent seems to remedy the situation.  However, it might be hard to find out which router that is (you'd need to sniff the Zigbee MAC traffic between the TRV and its assumed parent).  I could only prevent these issues structurally by moving my TRVs to a second network, formed by a different RaspBee, with only a Trådfri repeater as additional router.  

What routers do you have in your network, besides the ConBee II?  I haven't been able to pinpoint for which routers the issue occurs, except that the TRVs seem to like my RaspBee and Trådfri repeater, and the other way round.

Alissh, 2020-12-06:
@ebaauw Thanx for your fast responds.

I am new to ZigBee and only use the ConBee II:
The Parent-Router is only a raspberry 4 with "Home Assistent OS" and ConBee II Stick.
I lose the whole connection, no report from TRV to ConBee II or from ConBee II to TRV.
In deCONZ the Basic Cluster Info window is then empty without any values, if i press Attributes "read" nothing happens.
If i change temperature on TRV i don't see it in deCONZ or Home Assistent.
Only power off/on the TRV works.
Can there by a communication "overflow"? because the battery of the TRV discharge very fast too, since last change only 4-5 Weeks left.

My small network: 
- 5 temperature sensors
- 1 Motion Sensor: 0xE513 --> Sonoff SNZB-03 (some problems with MI:66666)
- 1 TRV: Thermostat 6 --> Eurotronic Spirit
[image]

ebaauw, 2020-12-06:
That's a different issue than mine.

I'd replace the batteries before trying anything else.

SwoopX, 2020-12-06:
I had pretty much the same experience with it and that's why I dumped it in the cupboard. On the other hand, the device does seem to have quite some firmware bugs whereas not everybody is equally affected.

Alissh, 2020-12-06:
@SwoopX  what type of TRV do you use?

SwoopX, 2020-12-06:
I currently use the AVM301, but tha's a different technology underneath (DECT). Got some Danfoss Ally here though, which seem to me more promissing. However, I'm still looking for the perfect device fulfilling all my wishes (supporting on-device schedules and groups).

moobid, 2020-12-12:
I am also experiencing lost connections. I have two Eurotronic Spirit ZigBee devices, and both loose connection after a restart of Phoscon or after some time.
I am using Phoscon Version: 2.07.00 / 15.11.2020. beta version.
I would be happy to help, to get this issue fixed.
I have also asked for advice on this issue here (German): on 09.12.2020, but so far none of the suggestions helped :/ https://forum.iobroker.net/topic/28785/how-to-eurotronic-spirit-zigbee-mit-conbee-ii/146

ChristophHornung, 2020-12-15:
@Alissh I have the same problem. I reported it [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3911). Are you also seeing the "is known by 0 neighbors" in the log?

Alissh, 2020-12-29:
Update: since i have some router, like LED-band and 230V plug-switches in the network and the Eurotronic Spirit ZigBee connected to one of this routers, i have no connection lost any more.
So it must be a problem between ConBee II and the Eurotronic Spirit ZigBee (direct connection problem).

VivantSenior, 2021-01-03:
Try to downgrade Conbee II firmware to [0x26580700](https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26580700.bin.GCF) ([reference](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3387#issuecomment-751625190)). It worked for me. [Instructions how](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) to down / upgrade.

jeankone, 2021-01-25:
Also got the same problems here. After restart of the Phoscon-GW, i have to re-pair my 2 thermostates. otherwise they'll loose connection anyways, after a week or so.

b3ng3, 2021-01-26:
i tried the fix with the ikea repeater, worked for 2 day, not it lost the connection again :-( this is so annoying, even with the repeater

ChristophHornung, 2021-01-26:
There is a new (old) firmware for the Spirit available [here](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/releases/tag/20190408). It seems to have an older date code which might not have the bugs the later versions had.

b3ng3, 2021-01-26:
Thank you, I will try it. Maybe this works

jeankone, 2021-01-30:
Was anybody able to update? Following procedures and it just says "idle" and nothing happens.

Gupy92, 2021-02-06:
@jeankone Hi there. I got the same Problem. The progress stays on idle.

ChristophHornung, 2021-02-27:
The issue still persists, and it seems there is no way to update the Spirit at the moment as it does not seem to accept the firmware linked above. Has anyone got information about whether it is possible to return the Spirits? The situation does not seem to improve at all.

Mimiix, 2021-03-01:
@ChristophHornung I think it's best to either contact your seller or the company themselves.  Or try the git that was metnioned.

I do like to close this issue, as this is something in the device firmware and on deCONZ side we can only do so much.

closed 2021-03-01 by Mimiix

Gupy92, 2021-03-06:
I wrote to Eurotronics and i got help of them. I resend my Spirits and they will exchange with devices with newer firmware.