dresden-elektronik/deconz-rest-plugin #3821: Debug View HowTo Close?

Alissh, 2020-12-06:
## Describe the question or issue you are having

When i open the new "Debug View", i don't find the "Close" button.
A 2nd click in the Help --> Debug View don't close it too.
Bug, wrong "doing" or "Home Assistent deCONZ Add-on" bug?

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.05.88)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

No Logs
## Additional context

SwoopX, 2020-12-06:
A right click on the title bar doesn't give you any menu?

Alissh, 2020-12-06:
There is no right click menu on title bar.
Only double click before "title bar text" collapse the window.

cllatser, 2020-12-08:
Press the ESC key. :) Same thing happened to me.

Alissh, 2020-12-08:
@cllatser thanx for your fabulous tip, it worked!

closed 2020-12-08 by Alissh

Dreamoffice, 2021-02-16:
how can i copy the text out of the debug window? 
select all > copy but i can t paste it, clipboard is still empty

Smanar, 2021-02-16:
And ctrl+c ?

pattyland, 2021-09-10:
As this issue it at the top of Google Results: Clipboardsync does not work with Web VNC. You need a real Deskop VPN Client and activate the VNC port in the addon config