dresden-elektronik/deconz-rest-plugin #3824: KNYCKLAN Open/Close remote

frassefraze, 2020-12-07:
## Device

- Product name: TYPE E1841 -  KNYCKLAN Open/Close remote
- Manufacturer: IKEA of Sweden
- Model identifier: KNYCKLAN Open/Close remote
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch

## Screenshots
[image]
[image]
[image]

Mimiix, 2020-12-07:
Missing screenshots of all other clusters.

frassefraze, 2020-12-07:
For some reason I took that as optional, sorry about that. Updated the original post with additional clusters.

MattWestb, 2020-12-07:
Pleas read / (manual) Update SW Build ID from basic cluster (0x4000).
I looking very like the 2 On/Off and Open/Close controllers but have different firmware ID (0x11c9) so new OTA images is coming soon.

E2001 and E2002 STYRBAR Remote control is also on the way :-))

frassefraze, 2020-12-08:
SW Build ID: 2.3.019

Updated all of the values one by one below (shouldn't the top "Read" button pick them all up?).
[image]

MattWestb, 2020-12-08:
Thanks very much i putting the info on known firmware but without known updates.
For My deCONZ is more or less not reading all attributes from the basic cluster automatic or manual must making it manual record by record like you was doing (Undocumented future).

Do you have the receiver E1842 in in deCONZ 2 ? 
Im interested of device ID and FW version if you can getting it. [KNYCKLAN receiver](https://zigbeealliance.org/zigbee_products/knycklan-receiver/)

frassefraze, 2020-12-08:
Sure, here is the receiver data

[image]
[image]
[image]

MattWestb, 2020-12-08:
Gret !!

I need the Image Type ID from the OTA cluster attribute 0x0008.

frassefraze, 2020-12-08:
There you go ( Image Type ID - 0x1103 ) 
[image]

MattWestb, 2020-12-08:
Tackar och bockar Jonas !

Added to known firmware in the Wiki.

MattWestb, 2020-12-09:
Is the the KNYCKLAN receiver working in deCONZ and you can sending on / off commands to it and also adding it in one groupe and sending commands ?
If yes try bind the remotes on/off cluster (0x0006) to the the same group in deCONZ GUI.

Its dont getting it nice implanted in Phoscon app but perhaps the funktion working locally and also from HA.

frassefraze, 2020-12-09:
Yes the KNYCKLAN receiver is working just fine in deconz, I can turn it on and off. 
But I actually just want to use the KNYCKLAN remote as a remote for other stuff picking up the events. I usually (not so much right now) use KNYCKLAN for outdoor watering and start it from HA using a timer, so I don't forget to turn it off, or I just use the cord connected "remote".

MattWestb, 2020-12-09:
Do you not giving your "Adventsgran" weather ?!?!?!?
And Rudolph the reindeer also need weather then doing the express deliveries of smart things in the north !!!

I dont knowing if its possible getting the events from the remote in deCONZ without the devs have implanting it in the code but I think not then its need being whitelisted and configured in the codebase for working OK.

But if you like you can loading the E1743 firmware and userdata and you can using it as the normal on/off switch (but need little souldring and SWD flasher for doing it) but its more expensive than buying one new (around 6€).

MattWestb, 2020-12-10:
Im sorry but i think adding support for new devices can taking little time for the moment then its not so easy getting the code being done by the company's devs. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765
I you or other users can doing it then its can being done faster (company policy ?)

Mhv  MW

frassefraze, 2021-01-01:
Still want this…

Mimiix, 2021-01-01:
@SwoopX what do we need to make this happen?

oywino, 2021-01-06:
@frassefraze can you please post copies of the deconz_event  from the button ?  (short press on/off as well as long press on/off)

frassefraze, 2021-01-06:
If the events showed up I wouldn't have created this post. Are they supposed to show up even if the device isn't supported or even added in phoscon ? I can only partly see it in deCONZ. But maybe I'm looking in the wrong place…

Mimiix, 2021-01-06:
It isn't supported yet. However, i think it might be the same as the on/off switch and the fyrtur up/down.

We have to wait for a dev to integrate it.

oywino, 2021-01-06:
If you can see it in deCONZ, then you're normally good to go. It doesn't have to be visible in Phoscon to operate. Have you tried the REST-API ?

Mimiix, 2021-01-06:
@oywino That's not correct. The screenshot indicates that it is not supported. You can see that because of its name being hex-valued.

oywino, 2021-01-06:
I see, my bad. I thought it popped up as a switch or a light perhaps. 
Guess we just have to wait then .....

SwoopX, 2021-01-06:
It would be best to have that device in hands as Ikea has some twists of it's own. The usual "whitelist and go for it" is probably not enough here, so ideally somebody does it with some experience in Ikea devices.

frassefraze, 2021-02-25:
I still want to be able to use my little droplet remote, so I'm keeping this active.

oywino, 2021-02-25:
Bump +1

frassefraze, 2021-03-19:
Yeah I'm stubborn… I still want this :)

Mimiix, 2021-03-23:
This remote seems to be the same as the On/Off and Open/Close remote. @ebaauw @SwoopX Can't we take over those configs?

ebaauw, 2021-03-23:
On/off and open/close remotes may look similarly mechanically, but are very different devices at Zigbee level.  As with any wireless controller, we need to know which commands it sends on what button actions.  Looking at the STYRBAR, it seems IKEA are even putting more variation in their wireless controllers...

Mimiix, 2021-03-23:
@ebaauw So it would be best if you had one hands on?

ebaauw, 2021-03-23:
That would be easiest, indeed.

dabepp, 2021-06-18:
Bump! Any update on the Styrbar E2001 FCC ID FHO-E2001?

rayitaly, 2021-06-28:
Bump! Any update on the Styrbar E2001 FCC ID FHO-E2001?

Mimiix, 2021-06-28:
Isn't that another switch?

dabepp, 2021-06-29:
> Isn't that another switch?

Unfortunately no. I can't add it do COnbee II,  neither as others -> switch nor as IKEA remote...

Mimiix, 2021-06-29:
No, i mean that a Styrbar remote is a different switch than an Knycklan.

dabepp, 2021-07-08:
> No, i mean that a Styrbar remote is a different switch than an Knycklan.

Yes, I've just hitchhicked the device request. Sorry. I guess I just create another request, right?

closed 2021-08-08 by github-actions[bot]