dresden-elektronik/deconz-rest-plugin #3825: Slow Response in HA when deCONZ is Running

dlindnegm, 2020-12-07:
I've been running the default ZHA zigbee integration on a Rpi4 for about 5 months with about 20 sensors, lights and switches.    Everything has worked perfectly until I decided to try installing the official deCONZ add-on.  As soon as I start the add-on my zigbee network slows to a crawl.  Lights and switches either don't turn on/off at all, or they have 2-5 second delays.  The status of the lights and switches in HA often are either greyed out or show the incorrect state.  As soon as I stop the add-on everything returns to normal.  

## Environment
deCONZ Add-on version is 6.5.0
ConBee Version: 2.05.88 / 10/15/2020
Conbee 2 firmware is 26680700

 - Host system: Raspberry Pi 4
 - Running method: HassOS 4.17 with deCONZ Add-on
 - Firmware version: 26680700
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: Yes
 - Second USB device is Zooz Z-Wave Plus S2 USB Stick ZST10

## deCONZ Logs
Log Errors from HA: 
No response to 'Command.aps_data_request' command with seq id '0x85'
11:28:25 AM – /usr/local/lib/python3.8/site-packages/zigpy_deconz/api.py (WARNING) - message first occurred at 11:20:47 AM and shows up 70 times
Unexpected transmit confirm for request id 132, Status: TXStatus.SUCCESS
11:26:04 AM – /usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py (WARNING) - message first occurred at 11:22:53 AM and shows up 3 times
No watchdog response
11:25:32 AM – /usr/local/lib/python3.8/site-packages/zigpy_deconz/zigbee/application.py (WARNING)
Updating zha sensor took longer than the scheduled update interval 0:00:30
11:25:16 AM – Sensor (WARNING) - message first occurred at 11:24:46 AM and shows up 2 times

deCONZ Log:
11:22:57:894 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'11408e4911593d0000003d0000', 1607365377 FROM devices WHERE mac = 'e2:0d:b9:ff:fe:03:c7:8a'
11:22:58:054 DB UPDATE device_descriptors SET data = x'0a5ec00001020700000010040003000500060008000200001900', timestamp = 1607365378 WHERE device_id = (SELECT id FROM devices WHERE mac = 'e2:0d:b9:ff:fe:03:c7:8a') AND endpoint = 10 AND type = 4
11:22:58:055 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 10, 4, x'0a5ec00001020700000010040003000500060008000200001900', 1607365378 FROM devices WHERE mac = 'e2:0d:b9:ff:fe:03:c7:8a'
11:22:58:159 ZDP discovery done in 200 ms11:22:59:973 APS-DATA.indication from unknown node 0x000D6F000E339C12
11:23:03:543 device state timeout ignored in state 3
11:23:04:290 ZCL attribute report 0x000D6F0005077149 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
11:23:08:046 saved node state in 1 ms
11:23:08:615 sync() in 568 ms
11:23:10:434 device state timeout ignored in state 3
11:23:11:931 APS-DATA.indication from unknown node 0x000D6F000E339C12
11:23:18:043 device state timeout ignored in state 3
11:23:20:549 device state timeout ignored in state 3
11:23:23:543 device state timeout ignored in state 3
11:23:24:060 Current channel 15
11:23:24:071 Device TTL 7082 s flags: 0x7
11:23:28:886 device state timeout ignored in state 3
11:23:34:622 DB UPDATE device_descriptors SET data = x'11408e4911593d0000003d0000', timestamp = 1607365414 WHERE device_id = (SELECT id FROM devices WHERE mac = 'e2:0d:b9:ff:fe:0a:a4:7b') AND endpoint = 0 AND type = 2
11:23:34:623 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'11408e4911593d0000003d0000', 1607365414 FROM devices WHERE mac = 'e2:0d:b9:ff:fe:0a:a4:7b'
11:23:34:833 DB UPDATE device_descriptors SET data = x'0a5ec00001020700000010040003000500060008000200001900', timestamp = 1607365414 WHERE device_id = (SELECT id FROM devices WHERE mac = 'e2:0d:b9:ff:fe:0a:a4:7b') AND endpoint = 10 AND type = 4
11:23:34:834 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 10, 4, x'0a5ec00001020700000010040003000500060008000200001900', 1607365414 FROM devices WHERE mac = 'e2:0d:b9:ff:fe:0a:a4:7b'
11:23:34:943 ZDP discovery done in 229 ms11:23:35:874 APS-DATA.indication from unknown node 0x000D6F000E339C12
11:23:39:543 device state timeout ignored in state 3
11:23:45:045 saved node state in 1 ms
11:23:45:389 sync() in 343 ms
11:23:45:851 APS-DATA.indication from unknown node 0x000D6F000E339C12
11:23:47:916 device state timeout ignored in state 3
11:23:50:543 device state timeout ignored in state 3
11:23:52:876 APS-DATA.indication from unknown node 0x000D6F000E339C12
11:23:55:703 DB UPDATE device_descriptors SET data = x'11408e4911593d0000003d0000', timestamp = 1607365435 WHERE device_id = (SELECT id FROM devices WHERE mac = 'e2:0d:b9:ff:fe:03:c7:40') AND endpoint = 0 AND type = 2
11:23:55:705 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'11408e4911593d0000003d0000', 1607365435 FROM devices WHERE mac = 'e2:0d:b9:ff:fe:03:c7:40'
127.0.0.1: new handler Process
localhost - - [07/Dec/2020 11:26:13] "GET //vnc_lite.html?scale=yes HTTP/1.1" 200 -
11:23:55:867 DB UPDATE device_descriptors SET data = x'0a5ec00001020700000010040003000500060008000200001900',172.30.32.2 - - [07/Dec/2020:18:26:13 +0000] "GET /novnc/vnc_lite.html?scale=yes HTTP/1.1" 200 3396 "http://192.168.0.115:8123/api/hassio_ingress/A9smEEXjM19iakY_yRwryrin_Jhm6DnMfeWUM9DDbmo/ingress.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
Ignoring interrupted syscall
127.0.0.1: new handler Process
localhost - - [07/Dec/2020 11:26:14] "GET //websockify HTTP/1.1" 101 -
localhost - - [07/Dec/2020 11:26:14] 127.0.0.1: Plain non-SSL (ws://) WebSocket connection
localhost - - [07/Dec/2020 11:26:14] 127.0.0.1: Version hybi-13, base64: 'False'
localhost - - [07/Dec/2020 11:26:14] 127.0.0.1: Path: '//websockify'
localhost - - [07/Dec/2020 11:26:14] connecting to: 127.0.0.1:5900
localhost - - [07/Dec/2020 11:26:34] 127.0.0.1:5900: Client closed connection
localhost - - [07/Dec/2020 11:26:34] 127.0.0.1:5900: Closed target
172.30.32.2 - - [07/Dec/2020:18:26:39 +0000] "GET /novnc/websockify HTTP/1.1" 101 120495 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
[... 35 lines cut ...]
  File "/usr/lib/python2.7/dist-packages/websockify/websocket.py", line 581, in handle
    SimpleHTTPRequestHandler.handle(self)
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "/usr/lib/python2.7/dist-packages/websockify/websocket.py", line 547, in do_GET
    SimpleHTTPRequestHandler.do_GET(self)
  File "/usr/lib/python2.7/SimpleHTTPServer.py", line 45, in do_GET
    f = self.send_head()
  File "/usr/lib/python2.7/SimpleHTTPServer.py", line 99, in send_head
    self.send_response(200)
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 406, in send_response
    self.send_header('Server', self.version_string())
  File "/usr/lib/python2.7/BaseHTTPServer.py", line 412, in send_header
    self.wfile.write("%s: %s\r\n" % (keyword, value))
  File "/usr/lib/python2.7/socket.py", line 328, in write
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
Ignoring interrupted syscall

## Additional context
Occasionally I can get into the Phoscon or deCONZ app via the HA GUI, but most times they are unresponsive. When I can connect with the deCONZ app it usually won't connect to the ConBee.  Once or twice I have been able to connect and see the network mesh with all of my devices.  As soon as I stop the Add-on all of my zigbee devices start working again.

closed 2020-12-07 by close-issue-app[bot]

Kane610, 2020-12-07:
You can't run zha and deconz on the same conbee at the same time

dlindnegm, 2020-12-07:
OK - I hadn’t seen that mentioned anywhere but it does make sense.  So if I wanted to start using deCONZ I’d need to remove the ZHA integration and reset/re-pair all of my devices to the deCONZ app?  My ultimate goal here was to be able to use an ikea Tradfri remote, and both deCONZ and zigbee2mqtt are mentioned as options without needing to use the Tradfri gateway.  I was able to get deCONZ installed, but don’t look forward to re-pairing everything if that’s what’s required.  Do you happen to know if zigbee2mqtt will have the same issue of not being compatible with zha?

Thanks!

Kane610, 2020-12-07:
Yeah, sorry different set ups completely