dresden-elektronik/deconz-rest-plugin #3827: ID-Lock

johnniemalm, 2020-12-07:
## Device

- Product name: ID-Lock
- Manufacturer: Id-lock
- Model identifier: 150
- Device type : Please remove all unrelated device types. 
  - Lock

https://idlock.no/wp-content/uploads/2020/11/User-Manual-Zigbee-module_EN_v0.4.pdf

## Screenshots

Mimiix, 2020-12-07:
@SwoopX / @Smanar is the above pdf enough?

SwoopX, 2020-12-07:
Oh, a rare surprise. That should do

Smanar, 2020-12-08:
@SwoopX have you started something ?
I have already started the addition of 3 devices
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3750
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3639

And all are using standard zigbee command, so the same code.

SwoopX, 2020-12-08:
@Smanar no, just taken a look at the doc so far.

Smanar, 2020-12-08:
Just by curiosity, you haven't at least
- The manufacture number
- The model ID
- The manufacture name 

From the doc you have "Datek" "ID Lock 150" but still missing the manufacture number.

johnniemalm, 2020-12-08:
Hi, cant find any other info about this, and havent been able to connect it to Deconz. Sorry

johnniemalm, 2020-12-08:
The manufacture number = 0402946
The model ID = ZB-IDL-01A
The manufacture name ID Lock Zigbee Module
The device MAC adress = ?, at least the 8 first char.

Smanar, 2020-12-09:
>and havent been able to connect it to Deconz

You haven't deconz with GUI, or you can't include it ?

The manufacture number is a number < 65000
Because I m searching the missing information on other project without sucess.

Electronisse, 2020-12-10:
Hi, Im very new to Deconz in general, and dont really know what I'm doing yet.
Anyway, Im running Hassio/HassOS with Deconz 6.6.0 addon.
But I think I might have some good information.
I managed to connect my ID Lock 150 to Deconz and can also use the unlock command so far. Nothing visible in Phoscon.
I also tried to add an extra xml-file with the clusters/attributes from the pdf, but i couldnt get them to work.

Just tell me if there is anything I can do, would be awesome to get this lock integrated in Hassio!
Here are some screenshots, hope that helps:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-12-10:
Nice thx.

So https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
git clone --branch kiwiset https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

So you will have (If I m right)
- More setting/command on deconz on cluster 0x0101 (but you need to replace manualy the xml file).
- The device can be included
- on/off to open/close the device
- a ZHAopenclose sensor created that mimic the door lock position

The sensor is still WIP, if you take a look in the pdf file, I think I will use it to show attribute 0x0000 and 0x0003, or if you have other ideas ....

Electronisse, 2020-12-10:
Thanks!
Im now stuck on getting it built, which doesnt seem possible to do inside a Hassio system?
I tried to search a solution for this, but didnt find any that worked for me.
qmake returns command not found. So does apt. My next idea would be to run raspbian in a vm and compile it on there?
Unfortunately I am beginner when it comes to linux as a whole.
I have no running linux machines apart from the Rpi4 running HassOS.
Maybe i could use a Rpi2 with clean raspbian img? I have one that is not used atm.

attribute 0x0000 and 0x0003 is the only ones i would actually use as sensors, apart from the battery level in percent but i guess that comes from the power configuration cluster. So your idea for sensors seem good.

Smanar, 2020-12-11:
Yep right the battery is from an other cluster, and I m not sure it s already implemented.

And yes it's possible with VM, but don't ask me how to configure it ^^
And yes it s possible too on a raspberry 2, it will be realy slow (almost with desktop), but enought for test.

jyrki69, 2021-01-14:
Any news on this? Can anyone confirm if Idlock with the Zigbee module is actually working in Deconz/HomeAssistant or not?
Im moving all my devices from Zwave to Zigbee and IDlock is on of the few devices left on Zwave.

Thanks

Smanar, 2021-01-14:
The code is not tested, so still in waiting list.
I m on 4 door locker in same time, and all in waiting list. Some of them are working but I have no return for a part.

If I realy don't have return, i will make a pr in blind mode for them and give up for the last part.

closed 2021-03-05 by github-actions[bot]

Smanar, 2021-03-08:
Have done a PR, if someone cantest it for this device ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4540

Electronisse, 2021-03-11:
I finally got qt5 and the code running. Cloned "branch doorlock https://github.com/Smanar/deconz-rest-plugin.git" today.
Can confirm that state and command through the API works as expected with my ID Lock 150. Battery works aswell.

Smanar, 2021-03-12:
Nice, thx a lot ^^.

marsin-crypto, 2021-04-12:
Hi. I am using deCONZ 6.8.0 2.10.04 / 3/14/2021 firmware 26660700 installed in HA on rpi. 
ID lock 150 appears in the deCONZ as node 0x1AC7, but I'm not able to see it in the GUI of PHOSCOM
Could you please guide me here a bit more.I read thread above but dont know where to start.

Mimiix, 2021-04-13:
@marsin-crypto Can only be after version 2.11.0

Smanar, 2021-04-13:
And it will be not on phoscon, you will need third app to use it.

marsin-crypto, 2021-04-13:
Thanks  for reply, zigbee module for ID-lock was made by Datek, seems they have a gateway and app, ID-lock on the webpage mentions Samsungs smartthing and some other hubs. Any more clues what sort of app? Ideally it should be integrated in HA...

Smanar, 2021-04-13:
lol, by third app, I mean app using deconz, not app using official gateway ^^.
It s just because phoscon don't support them yet, it s a new sensor type. For HA I think it s ok https://github.com/home-assistant/core/commit/c80f34a75454610bc337f6b22790e5bf52542308

Smanar, 2021-04-13:
I Think it s better ask on HA forum, idk since wich one version HA the device is included, and personnaly I don't know HA at all.
Else you can use direclty the api with command line to act on the device.

Alxandr, 2021-07-29:
Hi. Considering buying one of these zigbee modules for my ID locks. Did you get it working with home assistant?

Smanar, 2021-07-30:
I haven't the device, so I can't answer myself.
But the primary code is already in the actual version, so the first basic command can work.
The next version with more command is still in waiting list https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4876

alexpomsft, 2021-07-31:
@Smanar great work on the lock integration! And it's great to see progress being made on additional features for the locks.

I do have a question and would love some tips on how to debug it. About a week ago my ID Lock 150 connected via zigbee to a conbee ii started acting strange from within Home Assistant, connected via deconz. Debugging into it, I've found that calling lock.unlock will actually lock the device if the lock is already unlocked. I've also noticed that I cannot unlock the lock at all from within Home Assistant. Neither of these issues were previously present when I first setup the lock and integration approximately two months ago. I'm running on the latest versions of both deconz (v2.12.3) and Home Assistant (core-2021.7.4), but only upgraded after things started going strange. I think :)

I figure that there are two next debugging steps: manually try the deconz REST API and look at some deconz logs. Any tips on what I should be looking for?

Or is there a known issue that may be causing this?

Smanar, 2021-08-01:
>I've also noticed that I cannot unlock the lock at all from within Home Assistant

Yep for this issue, you make a try using direcly the API or at least using the GUI

>I've found that calling lock.unlock will actually lock the device if the lock is already unlocked

You mean is you send the "unlock" command, and if the device is already "unclocked" it lock the device ?
It s more an "toggle" command for me, but the API use only "lock" and "unlock" ....

But yes If you have this kind of issue, trying with the api is a good way to debug.

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

With curl
`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://IP:PORT/api/KEY/sensors/ID/config`

alexpomsft, 2021-08-03:
Thanks a ton! I've repro'd the issue via REST. I'm able to send ```{"lock": true}``` and have the device lock successfully. However, when I send ```{"lock":false}``` I receive a 200 response but the physical lock never moves. After this, the GET request for the sensor has ```{"config": {"lock": false}},"state":{"lockstate":"locked"}}```. 

Any tips on debugging further?

Smanar, 2021-08-03:
Arf, I think you have find something.
Are you able to compile code to test something ?

alexpomsft, 2021-08-03:
Not easily. Is there a development docker image that you recommend that could quickly get me going?

Smanar, 2021-08-03:
No problem, @SwoopX have confirmed the issue.
We will have soon a PR to correct that.

And BTW thx a lot, this problem can cause so much issues.

Smanar, 2021-08-04:
But this bug appeared in version 2.12 ?
So you are already using a beta version ?

alexpomsft, 2021-08-04:
I'm currently on v2.12.3. It's not quite clear to me when this issue started to appear, and if it was tied specifically to a deconz update. Sorry for note keeping better notes :(

Smanar, 2021-08-04:
>I'm currently on v2.12.3

Ok so it s normal, It s because you have asked for a "development docker image" so for me you was on a classic image, not the last one possible.
So yes now I m sure at 100% it will be corrected by the PR

alexpomsft, 2021-08-14:
I've confirmed that v2.12.4 fixes the issue!