dresden-elektronik/deconz-rest-plugin #3828: Heiman remote controller HS1RC (Model: RC_V14) shows as RC-EF-3.0

Mincka, 2020-12-07:
The original support request for Heiman HS1RC is https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2697
The support for this remote controller was added in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2813

I followed the instructions to add the `0x0501` endpoint and the device is joined.
However, I am unable to get events in Home Assistant since the model ID is not `RC_V14` but `RC-EF-3.0`.
The manufacturer appears as Heiman.

The purchase product is this one:
https://www.aliexpress.com/item/4000179913087.html

It is not written Heiman anywhere on the package or the product. Few people in the comments say that it's working in Deconz (2_05_81). Maybe different version of the remote are sent by the vendor...

The exact manual included with the remote is this one:
http://anyk.com/Private/ProductFiles/16542db9e0b0d4daac80.pdf

There is a M414-11Ever1.0 reference

Related issues and PR from other projects:
* https://github.com/Koenkk/zigbee-herdsman-converters/pull/1831
* https://github.com/pipiche38/Domoticz-Zigate/commit/0079bc9402523f122cee861f27e45f263fa29a09

Could it possible that just adding `RC-EF-3.0` as another model id for `RC_V14` could be enough to support events in Home Assistant?

## Device

- Product name: Smart Remote Controller-EF
- Manufacturer: HEIMAN? Sumring?
- Model identifier: RC-EF-3.0
- Device type : Please remove all unrelated device types. 
  - Remote

## Screenshots

[image]

[image]

vegetate7, 2020-12-10:
Have same device. 
Not only 'Model ID' different.
Also different 'MAC prefix' and 'Manufacturer code'
Missed 'Node info' panel screenshot
[image]

And 'Basic cluster' after 'Read' button pressed
[image]

vegetate7, 2020-12-11:
Ok, I've added this model into code, along with Heiman 'RC-EM' remote. Now it's joins, shows in REST-API as a sensor.
I can get events on keypress.
But the device itself sends every keypress 6 times with 3 sec interval:

```
07:14:54:461 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:14:57:475 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:15:00:503 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:15:03:022 DeviceAnnce of SensorNode: 0x842E14FFFEF82766 [1]
07:15:03:408 Bind response success for 0x842e14fffef82766 ep: 0x01 cluster: 0x0001
07:15:04:044 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:15:04:318 ZCL configure reporting rsp seq: 68 0x842E14FFFEF82766 for ep: 0x01 cluster: 0x0001 attr: 0x0021 status: 0x00
07:15:07:068 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:15:08:081 [INFO] - No button handler for: RC-EF-3.0 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
07:15:08:082 ZCL attribute report 0x842E14FFFEF82766 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
07:15:10:096 [INFO] - Button 1002 Arm day/home zones only RC-EF-3.0
07:15:57:529 [INFO] - No button handler for: RC-EF-3.0 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021
07:15:57:529 ZCL attribute report 0x842E14FFFEF82766 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```
the 'No button handler' message is generated on device sending battery status, as I understand. 0x021 == 33 dec, 3.3 Volt.

SwoopX, 2020-12-11:
Close, but no. But it is indeed a battery report.

Have you added the IAS ACE cluster on the coordinator and removed the battery from the device?

vegetate7, 2020-12-11:
> Have you added the IAS ACE cluster on the coordinator and removed the battery from the device?

Yes. I've added cluster in the beginning, and did some reset/rejoin/battery remove / delete device / delete sensor from REST after that.

SwoopX, 2020-12-11:
Hm, hard to tell what's going on there without holding the device in hands and sniffing the traffic. Maybe it (also) requires an IAS ACE binding. The first version supported didn't have that allures and just required the IAS ACE cluster on the coordinator for the match descriptor request.

vegetate7, 2020-12-11:
Shame on me. I completely not familiar with ZigBee cluster bindings. But can perform tests, if need.
Also, may be device awaits some kind of ACK for messages?

SwoopX, 2020-12-16:
No, I guess I know what's the issue here. Received another keyfob today from a different vendor and sniffed the traffic. It also repeats the commands...

I guess I know what to do but it may take some time.

vegetate7, 2020-12-16:
Ok, will wait.
I've made NodeRed filter for this, but it's very inconvenient, based on 20 seconds timeout on same button press.

SwoopX, 2020-12-16:
Yeah, that works with some very basic code 👍

vegetate7, 2021-02-11:
Still not merged? Is it awaiting some changes?

SwoopX, 2021-02-15:
Added in 2.10.00-beta

closed 2021-02-15 by SwoopX

zbrig, 2021-02-22:
Hi,
I upgrade to 2.10 (Firmware 0x26680700) to test the support of Heiman RC-EF-3.0.
Now, I can see it in deconz (jeedom) but unfortunetly, buttonevent and last updated values stay at "null" and "none".
Note that the Remote controller is not seen in phoscon (but I think that is normal), in deconz Gui I can see it and read node cluster without issues.
I don't know how to go deeper in troubleshooting. 
Let me how I add help,
Cheers

SwoopX, 2021-02-22:
What do the logs say (info + info L2?)

zbrig, 2021-02-23:
I don't know if this in relevant
_"105":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"34b77f6ec48d9a5d090a8b60cd2f670c","lastseen":"2021-02-23T08:24Z","manufacturername":"Heiman","mode":1,"modelid":"RC-EF-3.0","name":"RC-EF-3.0","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"84:2e:14:ff:fe:f9:50:01-01-0501"},_
_[2021-02-23 09:37:05][DEBUG] : {"00212EFFFF061977":{"config":{"battery":77,"on":true,"reachable":true},"e":"changed","id":"105","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:f9:50:01-01-0501"}}
[2021-02-23 09:37:10][DEBUG] : {"00212EFFFF061977":{"config":{"battery":77,"on":true,"reachable":true},"e":"changed","id":"105","r":"sensors","t":"event","uniqueid":"84:2e:14:ff:fe:f9:50:01-01-0501"}}_

This is the only log (in debug mode)I found with my remote when I include it.
When I push a button on my remote I don't have any log created.

[image]

Asmolet, 2021-03-15:
Hi, 
I have the same problem... I see the remote control in Deconz but nothing append when I push the buttons and there is no log.
Can you help us SwoopX ?

Thx

Mimiix, 2021-03-16:
@Asmolet Please provide the logs Swoop requested :)

Asmolet, 2021-03-16:
I just have this log :

`
[2021-03-16 18:22:18][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:27][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:33][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:37][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:48][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:52][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:53][DEBUG] : {"00212EFFFF0675FE":{"e":"changed","id":"1","r":"sensors","state":{"dark":false,"daylight":false,"lastupdated":"2021-03-16T17:22:53.166","status":180,"sunrise":"2021-03-16T05:56:29","sunset":"2021-03-16T17:52:48"},"t":"event","uniqueid":"00:21:2e:ff:ff:06:75:fe-01"}}
[2021-03-16 18:22:53][DEBUG] : Something received and no value updated, performing sync to create commands if needed
[2021-03-16 18:22:53][DEBUG] : 127.0.0.1:8484/api/0C894FF2A2/sensors type : POST
[2021-03-16 18:22:53][DEBUG] : {"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"da29498abf15b2b82d5a511c25be7837","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":false,"lastupdated":"2021-03-16T17:22:53.166","status":180,"sunrise":"2021-03-16T05:56:29","sunset":"2021-03-16T17:52:48"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:06:75:fe-01"},"2":{"config":{"battery":100,"on":true,"reachable":true,"temperature":2700},"ep":1,"etag":"7b2c930d6ce52c5be1883b3efb3190b3","lastseen":"2021-03-16T18:14Z","manufacturername":"LUMI","modelid":"lumi.sensor_magnet.aq2","name":"OpenClose 2","state":{"lastupdated":"2021-03-16T16:08:17.416","open":false},"swversion":"20161128","type":"ZHAOpenClose","uniqueid":"00:15:8d:00:03:c7:74:1f-01-0006"},"3":{"config":{"battery":100,"on":true,"reachable":false,"temperature":2800,"tholddark":12000,"tholdoffset":7000},"ep":1,"etag":"7e70d6c094c28a88b627e75be9a10d3a","lastseen":null,"manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"LightLevel 3","state":{"dark":false,"daylight":true,"lastupdated":"2021-03-11T10:22:19.712","lightlevel":26561,"lux":453},"swversion":"20170627","type":"ZHALightLevel","uniqueid":"00:15:8d:00:05:cf:23:f5-01-0400"},"4":{"config":{"battery":100,"duration":90,"on":true,"reachable":false,"temperature":2800},"ep":1,"etag":"9c84bdf1f3d9ac1055dd146e77d59580","lastseen":null,"manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Presence 4","state":{"lastupdated":"2021-03-11T10:22:19.720","presence":false},"swversion":"20170627","type":"ZHAPresence","uniqueid":"00:15:8d:00:05:cf:23:f5-01-0406"},"5":{"config":{"battery":100,"on":true,"pending":[],"reachable":true,"sensitivity":11,"sensitivitymax":21,"temperature":2100},"ep":1,"etag":"4a0cc1c1b9cb0b7003ad07155cd20fa2","lastseen":"2021-03-16T18:14Z","manufacturername":"LUMI","modelid":"lumi.vibration.aq1","name":"Vibration 5","state":{"lastupdated":"2021-03-15T19:29:27.272","orientation":[78,-1,12],"tiltangle":77,"vibration":false,"vibrationstrength":12},"swversion":"20180130","type":"ZHAVibration","uniqueid":"00:15:8d:00:05:77:6e:c1-01-0101"},"6":{"config":{"battery":70,"on":true,"reachable":true},"ep":1,"etag":"d2b0c5d2875d6e1c2fd017c5a96dbd04","lastseen":"2021-03-16T17:22Z","manufacturername":"Heiman","mode":1,"modelid":"RC-EF-3.0","name":"RC-EF-3.0 6","state":{"buttonevent":null,"lastupdated":"none"},"swversion":"2019.9.10","type":"ZHASwitch","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:22:53][DEBUG] : 127.0.0.1:8484/api/0C894FF2A2/lights type : POST
[2021-03-16 18:22:53][DEBUG] : {"1":{"etag":"20c73b13786f8c4ce20e11e8d73c1a04","hascolor":false,"lastannounced":null,"lastseen":"2021-03-16T17:20Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:75:fe-01"}}
[2021-03-16 18:23:06][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:23:27][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[2021-03-16 18:23:40][DEBUG] : 127.0.0.1:8484/api/0C894FF2A2/sensors/6 type : POST
[2021-03-16 18:23:42][DEBUG] : {"00212EFFFF0675FE":{"config":{"battery":70,"on":true,"reachable":true},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:05:87:50-01-0501"}}
[... cut ...]
[2021-03-16 18:33:07][DEBUG] : {"1":{"etag":"baf932cb272e5e1b47b3fcd39bb0d9c6","hascolor":false,"lastannounced":null,"lastseen":"2021-03-16T17:33Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:06:75:fe-01"}}
[2021-03-16 18:33:15][DEBUG] : {"00212EFFFF0675FE":{"e":"changed","id":"2","r":"sensors","state":{"lastupdated":"2021-03-16T17:33:15.162","open":false},"t":"event","uniqueid":"00:15:8d:00:03:c7:74:1f-01-0006"}}`

Nothing append when I push the button...

SwoopX, 2021-03-16:
Those are no logs, those are websocket messages.

Start deconz with `/usr/bin/deconz --dbg-info=2 > debug.txt` and share the textfile. Before doing that, stop the running services with `sudo systemctl stop deconz` and `sudo systemctl stop deconz-gui`.

Asmolet, 2021-03-17:
I use deCONZ on RPI with Jeedom, your commands are not working...
Otherwise I use deCONZ on Win10.

katiuskt, 2021-03-17:
Just received a new Heiman HS1RC-EF and just updated deConz docker image and I'm not being able even to be detected by Phoscon GW.
Sorry for the question, but is that the issue being managed in this topic?

If I press a button, I can see this in docker log so it seems deConz knows about the HS1RC-EF device somehow but no idea why it isn't added (I reset it just in case with the same result):
10:23:21:459 [INFO] - No button map for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), command: EMERGENCY (0x02), payload: None, zclSeq: 80

SwoopX, 2021-03-17:
Well, deconz is shouting at you guys what the issue is :) There's no button map defined for the device.

In the best case, the commands send upon button press are identical to the `RC-EM`. Then, you'd just need to find the file `button_maps.json` and add the remote's modelid accordingly (and restart deconz) to the map below. Please take note that you need to run at least deconz version 2.10.03-beta, otherwise you'll get multiple button press events upon a single press.

```
"rcv14Map": {
            "vendor": "Heiman",
            "doc": "Smart controller HS1RC-M and HS1RC-E",
            "modelids": ["RC_V14", "RC-EM"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        }
```

If the commands differ, a new map must be created.

katiuskt, 2021-03-17:
> Did you try to listen to `deconz_event` in Home Assistant in the Developper Tools?
> 
> You need to identify the event ID for each button then use them in in your automations as a trigger:
> 
> ```
> trigger:
> - event_data:
>     event: 1002
>     id: tradfri_on_off_switch_julien
>   event_type: deconz_event
>   platform: event
> ```
> 
> I can't wait to try it when the Deconz 2.10 add-on will be released for HA. :)

Thank you @Mincka. I wasn't aware of the "deconz_event" event but it doesn't seem to monitor the pressed keys. I'll keep trying with @SwoopX suggestion (my docker is still 2.10 beta2, though) and yours.

katiuskt, 2021-03-17:
> Well, deconz is shouting at you guys what the issue is :) There's no button map defined for the device.
>
> In the best case, the commands send upon button press are identical to the `RC-EM`. Then, [...]

Thank you @SwoopX. I'll try to figure out where is that file into the docker image and I'll check it once the beta3 docker is available (latest tag is still pulling the beta 2 apparently).

Mincka, 2021-03-17:
> Well, deconz is shouting at you guys what the issue is :) There's no button map defined for the device.
> ...
> If the commands differ, a new map must be created.

I am not sure to understand anymore the kind of support we will get with the next Deconz release for this remote.

I though that your PR brought support for RC-EF-3.0 as it adds a reference there:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/bindings.cpp#L3672

But now I don't understand anymore why this reference was not also added here in the PR:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/button_maps.json#L1047

Is it because you don't know yet if the button maps are the same?

SwoopX, 2021-03-17:
How should I know what commands the device sends and how it behaves?

Mincka, 2021-03-17:
I am just asking because I thought that the purpose of #4209 was just to add another reference of the same device. But since the button maps are not updated yet, I guess that's the reason you called it "initial support". 

So, if I understand correctly, either we confirm that the button maps are the same and we can just add this model ID to the current button maps for this kind of remote, either we need to create specific button maps for this model. In both cases, we will need another PR for button_maps.json before we can get events in Home Assistant.

katiuskt, 2021-03-17:
Sorry guys, but I'm a kind of newbie here and... I've found the 'button_maps.json' file inside the docker (marthoc/deconz) container (path: '/usr/share/deCONZ/devices/button_maps.json') and it already contains:
[image]

So from @SwoopX previous comment I understand the idea is just to add the string "RC-EF-3.0" to the modelids array (along side the "RC_V14" and the "RC-EM" ones) but... I just discovered the docker image doesn't contain vi nor nano so... is it possible to edit the button_maps.json on it?

If not, which is the expected approach to this? Because I understood this issue raised by @Mincka was intended to provide that modelid into the "standard code" (sorry if I don't use the proper words) of deCONZ.

SwoopX, 2021-03-17:
> I am just asking because I thought that the purpose of #4209 was just to add another reference of the same device. But since the button maps are not updated yet, I guess that's the reason you called it "initial support".

Correct

> So, if I understand correctly, either we confirm that the button maps are the same and we can just add this model ID to the current button maps for this kind of remote, either we need to create [...]

Well, not quite. The PR would be required to have it for everybody persistently available in future release. The charm of the button mpas is though that you can edit it yourself anytime. That's why this part is not so "super" urgent.

SwoopX, 2021-03-17:
@katiuskt If you can get me confirmation on all 4 buttons, I'll make sure it gets included in the next stable (given that's not in build already).

katiuskt, 2021-03-18:
> @katiuskt If you can get me confirmation on all 4 buttons, I'll make sure it gets included in the next stable (given that's not in build already).

Hi @SwoopX, yes, I can confirm every button pressed show consistent information into the deconz docker log (one press shows four or five messages but that's already known by you and to be tested on beta3 as I'm at beta2 right now)

1002 -> Arm home
2002 -> Disarm
3002 -> Emergency
4002 -> Arm all

Thank you

```
08:12:16:380 [INFO] - Button 3002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Emergency, payload: None, zclSeq: 123
08:12:16:580 [INFO] - Button 2002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Disarm, payload: 0000FF, zclSeq: 124
08:12:19:610 [INFO] - Button 2002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Disarm, payload: 0000FF, zclSeq: 125
08:12:20:227 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 126
08:12:20:228 ZCL attribute report 0xEC1BBDFFFE0FA7E3 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:12:22:641 [INFO] - Button 2002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Disarm, payload: 0000FF, zclSeq: 127
08:12:25:279 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: 0x01, payload: 0F010B12, zclSeq: 0
08:12:26:170 DeviceAnnce of SensorNode: 0xEC1BBDFFFE0FA7E3 [1]
08:12:27:329 [INFO] - Button 2002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Disarm, payload: 0000FF, zclSeq: 1
08:12:30:346 [INFO] - Button 2002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Disarm, payload: 0000FF, zclSeq: 2
08:12:31:151 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 3
08:12:31:151 ZCL attribute report 0xEC1BBDFFFE0FA7E3 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:12:32:908 [INFO] - Button 4002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm all zones, payload: 0300FF, zclSeq: 4
08:12:35:937 [INFO] - Button 4002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm all zones, payload: 0300FF, zclSeq: 5
08:12:36:203 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: 0x01, payload: 0F010B12, zclSeq: 6
08:12:39:467 DeviceAnnce of SensorNode: 0xEC1BBDFFFE0FA7E3 [1]
08:12:40:601 [INFO] - Button 4002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm all zones, payload: 0300FF, zclSeq: 7
08:12:43:627 [INFO] - Button 4002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm all zones, payload: 0300FF, zclSeq: 8
08:12:44:451 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 9
08:12:44:451 ZCL attribute report 0xEC1BBDFFFE0FA7E3 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:12:45:427 [INFO] - Button 1002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm day/home zones only, payload: 0100FF, zclSeq: 10
08:12:48:429 [INFO] - Button 1002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm day/home zones only, payload: 0100FF, zclSeq: 11
08:12:49:501 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: 0x01, payload: 0F010B12, zclSeq: 12
08:12:51:958 DeviceAnnce of SensorNode: 0xEC1BBDFFFE0FA7E3 [1]
08:12:53:091 [INFO] - Button 1002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm day/home zones only, payload: 0100FF, zclSeq: 13
08:12:56:122 [INFO] - Button 1002 - RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ACE (0x0501), action: Arm day/home zones only, payload: 0100FF, zclSeq: 14
08:12:56:934 [INFO] - No button handler for: RC-EF-3.0, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 15
08:12:56:934 ZCL attribute report 0xEC1BBDFFFE0FA7E3 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
```

Asmolet, 2021-03-19:
Thanks SwoopX & Co :)

I confirm that works in Jeedom ! I just made the modification of the button_maps.json by adding RC-EF-3.0 
in the paragraph rcv14map in my RPI.

Mincka, 2021-03-19:
Do we know if this kind of remote also supports other kinds of events? Such as "button press" (1001 usually), "long press", "long press release", "double press"...

katiuskt, 2021-03-21:
Double click o long press doesn't seem to get another values: I don't get any other code rather than the 1002, 2002, 3002, 4002 mentioned above

ads1230, 2021-05-21:
Hey, is anyone having issues with commands being sent 4x per press? Causing issues with toggle automations.

Mincka, 2021-05-22:
Yep, same here. I guess this device was not excluded in this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4716
If it's the same kind of issue, maybe it could be extended for this remote.

SwoopX, 2021-05-30:
No, looks like there's a new bug which prevents processing the command sent by the switch and giving the required respose. We're looking into that...