dresden-elektronik/deconz-rest-plugin #3829: (Lidl) Livarno Lux Remote Control

derrohrbach, 2020-12-07:
Note: This is a device request for the remote control included with the new lidl lamps

## Device

- Product name: Livarno Lux Remote Control
- Manufacturer: _TYZB01_bngwdjsr
- Model identifier: TS1001
- Device type : Please remove all unrelated device types. 
  - Remote

## Screenshots

[image]
[image]
[image]
[image]

Mimiix, 2020-12-07:
Hey!

Your missing screenshots of the other clusters 😅

derrohrbach, 2020-12-07:
[image]
[image]
[image]
[image]
[image]

derrohrbach, 2020-12-07:
Others were mostly empty

davidkrass, 2020-12-08:
I am also interested as i have 4 of these remote controls flying around now. If any further information is required i could also try to help.

aerox845, 2020-12-16:
really would like to add them also cause i also have 5 of them around

Mimiix, 2020-12-17:
For a dev to integrate it , we need someone with a sniffer or someone providing the device to a dev. Otherwise, it will be hard.

davidkrass, 2020-12-17:
Will it help that it is already integrated into Zigbee2MQTT? 
(see https://github.com/Koenkk/zigbee-herdsman-converters/blob/ffc473f3a790dbf950d83791d6b135a3eb267802/devices.js#L15630)

Mimiix, 2020-12-18:
@SwoopX @Smanar  up for you to decide

aerox845, 2020-12-18:
i can totally help or try to if anything is needed

ebaauw, 2020-12-19:
My colleague got a LIDL starter set, but only needed the gateway for his Xmas lightstrip.  I now have my hands on the colour light and on the remote.  Looks like the screenshots above, but a different MAC address prefix, 0xBC33AC.

The device keeps state.  It remembers whether _On_ or _Off_ was last pressed, and cycles through _On_ presses.
- The _On_ button sends an _On_ command to group 0x0000 at first.  On the next press, it sends a unicast _On/Off_ cluster 0xFD command to the coordinator, with a one-byte payload, that cycles from 0x00, 0x01, 0x02, 0x03.  From the manual: this is used by the LIDL gateway to recall scenes.
- The _Off_ button sends an _Off_ command to group 0x0000, and resets the _On_ sequence;
- The _DimUp_ and _DimDown send _Step_ on press, and _Move_/_Stop_ on hold/release to group 0x0000

Looks like we should be able to support (using the mapping from the Hue dimmer switch, even though the buttons are arranged differently):
- 1002 for _On_.  Not sure if we want to expose the cycling: it's not really double press, treble press, quadruple press, but I support we could cycle 1002, 1004, 1005, 1006;
- 2002 and 2001/2003 for _DimUp_;
- 3003 and 3001/3003 for _DimDown_;
- 4002 for _Off_.

I tried binding the client _On/Off_ and _Level Control_ cluster to another group, as well as to the coordinator, but the switch keeps sending commands to group 0x0000.  Never tried that before, but it seems a valid group ID, and I can that group to lights using the server _Groups_ cluster in the GUI.  I'm not sure how to support that group from the REST API.

After adding a group to the remote's server _Group_ cluster, it sends the commands to that group.  Haven't seen that before...

_Battery Percentage Remaining_ is reportable (at least responds to configure reporting).  Default values make no sense.  Not sure if a binding to the coordinator is needed.

The client _OTAU_ cluster seems functional.  It uses manufacturer 0x1002 and image 0x1602 - same as the Christmas light.  Different version, though.

rotkiv, 2020-12-19:
+1 from me.
really would like to add them also.

Mimiix, 2020-12-19:
I am locking this issue for now. We don't need any more +1s. @ebaauw I'll unlock when it's in a first beta.

ebaauw, 2020-12-19:
Hm don't know why my commit isn't shown here, but it adds support for the LIDL remote:
```
$ ph get /sensors/9
{
  "config": {
    "battery": 100,
    "group": "10",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "49739281abb50b9a04dc6fbcc2f94595",
  "lastseen": "2020-12-19T23:28Z",
  "manufacturername": "Tuya",
  "mode": 1,
  "modelid": "TS1001",
  "name": "TS1001 9",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2020-12-19T23:28:24.106"
  },
  "type": "ZHASwitch",
  "uniqueid": "bc:33:ac:ff:fe:4c:95:49-01-1000"
}
```
I should create the group on pairing, add it to the remote, and setup binding and attribute reporting for the battery.

```
$ ph get /groups/10
{
  "action": {
    "alert": "none",
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "9"
  ],
  "etag": "ac0b4e6120190710f2fefe24cb7f3ba9",
  "id": "10",
  "lights": [],
  "name": "TS1001 9",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup",
  "uniqueid": "bc:33:ac:ff:fe:4c:95:49-01"
}
```

closed 2020-12-23 by manup