dresden-elektronik/deconz-rest-plugin #3835: Danalock V3 Lock state binding lost

aryelevin, 2020-12-08:
## Describe the question or issue you are having
Hi Everyone,

After adding the Danalock V3 to my network, I had to manually bind the Door Lock cluster to the Coordinator in order to get realtime updates of the lock state, this works great.
BUT! today I had a power outage for 2 mins of so and after the server (with the doconz) restarted, I checked and I found that the binding was just lost.
I had this issue out of a sudden a few days ago also, and now I know this happened after a power outage.
Please note that the Danalock itself is battery powered, so the issue seems to be with deCONZ since it related to power outage as it seems for me...

Any idea of what's wrong and how to solve this?

Thanks In advance.

## Screenshots
NA

## Environment
 - Host system: PC
 - Running method: Ubuntu 20 in Virtual Machine
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes - Aeotec Z Stick (I use -dev param in deCONZ to avoid issues with this)

## deCONZ Logs
NA

## Additional context
NA

Smanar, 2020-12-08:
I have see a comment about that, have you tried to remove and replace the battery ?

aryelevin, 2020-12-08:
@Smanar I now did this and it continues to work as expected.
I'll try to do a power cycle on my deCONZ system and will report soon.
Thanks

aryelevin, 2020-12-09:
OK, so after doing a complete restart (shutdown and restart) of the server it seems that it works good, so no issues so far...
I'll continue to monitor this to see if it comes up again and will reopen the issue.

closed 2020-12-09 by aryelevin