dresden-elektronik/deconz-rest-plugin #3837: ConBee II not discovering anything on raspberry Pi 4B

vad, 2020-12-08:
## Describe the bug

Looks like a duplicate of  #1643 but that has been closed without solutions. I have a ConBee II that used to work on a RPi3, and works on a Ubuntu laptop. It doesn't work at all on a RPi 4.

What I mean for "doesn't work":

- dmesg ok
- "In Network" in deCONZ GUI
- it doesn't discover anything (I have aqara and OSRAM devices) using Phoscon App

As I said, discovery works if I do the exact same operations on my laptop.

## Steps to reproduce the behavior

I tried several times.

## Expected behavior

I would expect conbee II to discover my devices.

## Screenshots

## Environment

 - **Host system**: Raspberry Pi 4B
 - **Running method**: Raspbian, firmware and kernel updated to latest versions
 - **Firmware version**: 0x26680700
 - **deCONZ version**: 2.05.88-raspbian-buster-stable
 - **Device**: ConBee II
 - **Do you use an USB extension cable:** no
 - **Is there any other USB or serial devices connected to the host system? If so: Which**? SSD root disk

## deCONZ Logs

```
15:14:13:897 Mgmt_Lqi_req zdpSeq: 184 to 0x00212EFFFF0500EF start index 0
15:14:13:898 APS-DATA.request id: 78, addrmode: 0x03, addr: 0x00212effff0500ef, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
15:14:13:954 APS-DATA.confirm id: 78, status: 0x00 SUCCESS
15:14:13:955 APS-DATA.confirm request id: 78 -> confirmed, timeout 23028416
15:14:13:966 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
15:14:13:967 APS-DATA.indication request id: 78 -> finished
15:14:13:967 APS-DATA.request id: 78 erase from queue
15:14:13:968 ZDP status = 0x00 -> SUCCESS
15:14:13:968 ZDP Mgmt_Lqi_rsp zdpSeq: 184 from 0x00212EFFFF0500EF total: 0, startIndex: 0, listCount: 0
```

## Additional context

I use conbee II + Aqara Humidity/T+ OSRAM Smart+ Plug.

Mimiix, 2020-12-08:
Hi,

-  Can you add more logs? Preferably 1 minute or so. Upload to pastebin :)
- Also, I recommend using a extension cable and trying it again.

vad, 2020-12-08:
it works with an extension cable :see_no_evil: thank you! :+1:

closed 2020-12-08 by vad