dresden-elektronik/deconz-rest-plugin #3840: Hue Tap button slow response after adding Smart plug

posixx, 2020-12-08:
## Describe the bug

On my system running latest stable version my hue taps are suddenly responding slowly. On the log i see the following during button press of a hue tap:

20:40:13:713 [INFO] - Button 16 ZGPSWITCH
20:40:13:716 rule event /sensors/18/state/lastupdated: 0 -> 0
20:40:13:717 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2]
20:40:13:717 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2] failed with status 404 Not Found
20:40:16:841 GP srcId: 0x0041D27E cmd: 0x10 frameCounter: 4715
20:40:16:841 [INFO] - Button 16 ZGPSWITCH
20:40:16:845 rule event /sensors/18/state/lastupdated: 0 -> 0
20:40:16:846 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2]
20:40:16:846 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2] failed with status 404 Not Found
20:40:18:938 GP srcId: 0x0041D27E cmd: 0x10 frameCounter: 4716
20:40:18:939 [INFO] - Button 16 ZGPSWITCH
20:40:18:943 rule event /sensors/18/state/lastupdated: 0 -> 0
20:40:18:944 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2]
20:40:18:944 trigger rule 10 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule2] failed with status 404 Not Found
20:40:19:033 [INFO] - No button handler for: SML001 endpoint: 0x02 cluster: 0x0402 command: 0x0A payload[0]: 000
20:40:19:034 ZCL attribute report 0x0017880102007221 for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000
20:40:20:204 GP srcId: 0x0041D27E cmd: 0x22 frameCounter: 4717
20:40:20:204 [INFO] - Button 34 ZGPSWITCH
20:40:20:209 rule event /sensors/18/state/lastupdated: 0 -> 0
20:40:20:210 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1]
20:40:20:210 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1] failed with status 404 Not Found
20:40:21:950 GP srcId: 0x0041D27E cmd: 0x22 frameCounter: 4718
20:40:21:950 [INFO] - Button 34 ZGPSWITCH
20:40:21:955 rule event /sensors/18/state/lastupdated: 0 -> 0
20:40:21:956 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1]
20:40:21:956 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1] failed with status 404 Not Found

The strange thing is; i use home assistant with appdaemon for handling the buttons, not deconz.

This started today after having added a Hue Smart Plug to my system (LOM001). I only have hue devices on my deconz environment. Could the addidtion of the LOM001 be of any influence? This is the only non-light i have on the system..

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

posixx, 2020-12-08:
Could this be due to the recent ZGP proxy changes? I tried removing intergation from home assistant and add it again, but now the switches are reacting even worse; some clicks don't work anymore. All motion sensors are working and reacting instantly, so it doesn't seem to be related to deconz <-> appdaemon communitaction as my motion sensors are also managed by appdaemon instead of deconz rules.

posixx, 2020-12-08:
Ok, debugging some more; Button presses are reported withing appdaemon immediately. So ZGP proxy doesn''t seem to be the problem...

posixx, 2020-12-08:
Also the calls to deconz for activating / turning on lights are instantly called, so the delay is within deconz itself. Looks like the rule error messages are related? 

trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1] failed with status 404 Not Found

Hopefully some can explain what and why these happen.

posixx, 2020-12-09:
Update: this morning the delay was still there. I unplugged the Smart Plug and immediately all Taps worked instantly again. Then i plugged the Smart Plug in again and everything keeps working OK upon this moment. Very strange.. Will keep this report open for some days; if all keeps working i will close it.

jensflorian, 2020-12-10:
> 
> 
> Ok, debugging some more; Button presses are reported withing appdaemon immediately. So ZGP proxy doesn''t seem to be the problem...
I have the same issue. After upgrading and creating ZGP proxys, rule response to ZGPSwitches is slower (did not improve after 12hours). I also have LOM001 smart plugs in my system.

posixx, 2020-12-10:
@jensflorian can you unplug the LOM001's in your system and try? On mine the problem didn't return after unplugging and re-plugging the LOM001..

ebaauw, 2020-12-10:
> 20:40:21:956 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1] failed with status 404 Not Found

ZGPSwitch devices don’t have the cluster in their `uniqueid`.

posixx, 2020-12-10:
> > 20:40:21:956 trigger rule 9 - 00:00:00:00:00:41:d2:7e-f2-0021[Rule1] failed with status 404 Not Found
> 
> ZGPSwitch devices don’t have the cluster in their `uniqueid`.

So this is expected behaviour? If so, thx can be safely ignored then..

ebaauw, 2020-12-10:
I’m not sure if it’s expected.  Where does the `uniqueid` value with cluster come from?  I think that would need to be corrected.

jensflorian, 2020-12-10:
Cluster 0x0021 is BatteryPercentageRemaining - my rules were created in phoscon app, so this could be the culprit. Will test unplugging LOM001 tonight.

posixx, 2020-12-10:
No, i even upgraded last night going from v2.05.88 to v2.7.1 and the problem didn't return

posixx, 2020-12-10:
I'm sorry, spoke too soon. just now the problem reappeard. Again unplugging the LOM001 fixes the problem immediately. So there is definitely something going on with the LOM001 implementation!

jensflorian, 2020-12-10:
I also noticed some rules failing with status 404 not found. These were orphaned rules created by phoscon. Using ZGP debug view I found that up to four rules were firing on those ZGPswitches with slow response. Obviously phoscon doesn't clean up properly when you use the switch editor to create rules.

I used 'ph delete /rules/XXX' to remove the superfluous rules. Now the response of the ZGPswitches has markedly improved.

posixx, 2020-12-10:
I never used phoscon for creating rules, i don't know how to do that. I only have created groups and scenes within deconz; everything else is controlled by home assistant. So in my case there are no rules to delete.

posixx, 2020-12-12:
Ok i added 3 new lights to my gardenhouse so i can get ZGP Proxy working from there, but after i included the lights in deconz i'm back with the slow response problem. This time the unplugging of the LOM001 doesn't help. So it looks like this problem arrises when adding devices to an existing system. 

I have enables some debugging options in deconz, and when the delay is happening, the following message is logged in deconz:

command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_INDICATION2, seq: 141

So when the button presses are reacting directly, this message isn't logged. So it is due to the slow responses.. Hopefully this helps pointing in the correct direction to solve this nasty bug.

posixx, 2020-12-16:
UPDATE: as soon as i'm changing something on my gateway (adding lights or switches) all switches are reacting slowly and log the CMD_APS_DATA_INDICATION2 error. I was able to solve this by unpowering all lights + LOM001 which i added last 2 weeks. Immediately buttons are instantly working. 

However; after i power on these lights again, the ZGP Proxies which were created before are gone. In order to have the Gira switch working again i have to start inclusion again, pressing buttons on the Gira, and stop inclusion. But after this my buttons are slow again logging CMD_APS_DATA_INDICATION2. Chicken and Egg.. Please solve this timeout bug as this is a very disturbent one..

posixx, 2020-12-17:
I decided to leave the system for a while to see if this behaviour changes. And indeed, this morning the delay is gone! That's good news. The bad news; ZGP proxies do not work anymore.,

closed 2021-03-05 by github-actions[bot]