dresden-elektronik/deconz-rest-plugin #3843: LEDVANCE LED Panel Connected+ 625x625 UGR19

AndreasPollinger, 2020-12-08:
Tunable White LED Panel is wrongly detected as color light. Phoscon app therefore provides the wrong control sliders for the light.

- Product name: LEDVANCE LED Panel Connected+ 625x625 UGR19
- Manufacturer: LEDVANCE
- Model identifier: Panel TW 620 UGR19
- Device type : Light (Tunable White)

## Screenshots

[image]
[image]

Please let me know if additional screenshots/information is required.

AndreasPollinger, 2021-01-01:
refreshing...

Mimiix, 2021-01-01:
Can you check if it works correctly with the rest api?

If that's the case  I move the issue to the phoscon repo.

AndreasPollinger, 2021-01-09:
It works with the rest API. I can set the value of the color temperature there trough the "ct" paramter and the light reacts accordingly.
However I still think that it is not only an issue with the phoscon app, as I am having the same problem with the light in openhab.
From the REST API I get the following information for the LEDVANCE CONNECTED+ lamp:
[image]
There is no "ct" value there but the "sat" & "xy" value.
For my LEDVANCE SMART+ lights (also tunable white LED Panels) which are working fine everywhere I get the following from the REST API:
[image]

AndreasPollinger, 2021-03-01:
Refeshing...

AndreasPollinger, 2021-03-29:
Refreshing...

Smanar, 2021-03-29:
The problem is IDK if in the actual deconz version it s possible to block some field in the JSON.
We need something special to reconize the device.

AndreasPollinger, 2021-04-25:
Hey Smanar,
thank you for looking into this.

What do you mean by you need something special to reconize the device?
The model identifier is shown in my first post. So you need something additional?

closed 2021-05-30 by github-actions[bot]