dresden-elektronik/deconz-rest-plugin #3845: Lidl TS0505A

slusken, 2020-12-08:
## Describe the question or issue you are having
I bought 2 E14 TS0505A from Lidl and I can pair them and can see them in Phoscon and also in HA.
But the RGB colors are so far of:
255,0,0 = the light turns red
0,255,0 = the light turns green
0,0,255 = the light turns blue
255,255,0 = the light is still green.

But if you mix it like 255,130,0 then you get something that looks like yellow?
Also brightness 1 with RGB color and you can barely see that the light is on... good!
Brightness 1 with warm white and you cant even look into it... not so good!

Also transition doesn't work with HA.

Dont know if this is the correct place to post this info but the other post regarding TS0505A was closed.

## Screenshots

## Environment

    Host system: PC Debian
    Running method: Marthoc Docker container
    Firmware version: 26390500
    deCONZ version: 2.07.00 / 2020-11-15
    Device: ConBee I
    Do you use an USB extension cable: yes
    Is there any other USB or serial devices connected to the host system? If so: Z-wave stick

## deCONZ Logs

## Additional context

Mimiix, 2020-12-09:
This behaviour is seen with the ledstrip aswell.

slusken, 2020-12-09:
I talked to a guy on Facebook and he had the same problem with this lamp using Smartthings.
Then he "changed DH" and it solved the problem...
So the question is where can I do that in deconz and what is the DH? :)

Mimiix, 2020-12-10:
I have no clue. I tried some stuff with my ledstrip, but i see the same.

You should ask _the guy_  what DH is 😅

Anthrix, 2020-12-10:
DH = Device Handlers This is from Samsung Smart SmartThings this part allows to create custom code for device control.

slusken, 2020-12-11:
> 
> 
> DH = Device Handlers This is from Samsung Smart SmartThings this part allows to create custom code for device control.

So can this be something that could be fixed in deconz? or do I need to ask him if he can provide any more information?
Because he said that changing the DH solved the problem with the color and the brightness that don't work.

Smanar, 2020-12-11:
I think it mean they have a custom file that correct that, But I haven't find it on their github.

Anthrix, 2020-12-11:
> So can this be something that could be fixed in deconz? or do I need to ask him if he can provide any more information?
> Because he said that changing the DH solved the problem with the color and the [...]

There is DTH people using for SmartThings to get working lidl lights.
[https://github.com/Obi2000/smartthings/blob/master/Sengled%20RGBW%20Bulb.groovy](https://github.com/Obi2000/smartthings/blob/master/Sengled%20RGBW%20Bulb.groovy)

ebaauw, 2020-12-12:
By the looks of it, that’s just the usual RGB to CIE 1931 colour conversion.  I don’t see any (other) magic there.  Make sure HA sets the colour using `xy` instead of `hue` and `sat`.

h-koenig, 2020-12-12:
full ACK: just yesterday I did some color tests with exactly that E14 TS0505A, 
directly using the REST API, not phoscon.  selecting various colors via X/Y coordinates works ok.

but also playing with phoscon (2.07.00 beta) with the HSB ui looked ok to me...

slusken, 2020-12-13:
I still cant get the transition to work with HA.
I am using Marthoc Docker container with the deCONZ version: 2.07.01.

Is this something that's on HA or is the transition executed by deconz?

ebaauw, 2020-12-19:
Transition is executed by the light firmware.  Or not...

closed 2021-03-05 by github-actions[bot]