dresden-elektronik/deconz-rest-plugin #3847: Use TS0205 tuya smoke sensor as a siren

saxn-paule, 2020-12-09:
## Feature request type
  - Extend the REST-API with new API endpoints, attributes or capabilities.

## Description
I would really appreciate, if the TS0205 could be used as a controllable siren.
Maybe the siren could be added as a light that could be turned on via Home-Assistant.

## Considered alternatives

## Additional context

Thanks in advance
Paul

Smanar, 2020-12-09:
Can I have more informations ? TSXXXX mean nothing for tuya.
Do you have a link to the device ?
A cluster capture from deconz ?

saxn-paule, 2020-12-09:
See here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3395
and here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3557

Smanar, 2020-12-09:
It s possible a zigbee device use the cluster 0x0500 IAS for a siren instead of the IAS WD cluster ?
Do you have something in the cluster 0x0500 to test the siren ?

From the documentation, the device manage itself the siren no ?

saxn-paule, 2020-12-09:
I don't understand what you are talking about. Clusters, IAS,.... don't know.

I run Deconz and Phoscon in a Docker container on my headless server. 

I manage all the devices via Home-Assistant.

Smanar, 2020-12-09:
Ha you haven't deconz with desktop ?
I have take a look on your 2 links, and It seem the device have not a cluster to use a siren, so I have read the documentation, and I m not sure but it seem it s the device itslef that trigger the siren, not possible another device can trigger the siren without smoke.

saxn-paule, 2020-12-09:
Mhh very disappointing.
Then I don't know what the Zigbee function should be used for. No meshing, no remote triggering,... 

Wait, if there is fire in a room, then in addition to the siren, my smartphone could tell me, that I should leave the building. Great...... Not! 

Thank you for your investigations.

Smanar, 2020-12-09:
>Wait, if there is fire in a room, then in addition to the siren, my smartphone could tell me, that I should leave the building. Great...... Not!

Yes, it s the working mode, but seriously if you are at home, you will listen first the siren before your phone ^^.
The siren seem working naturaly, if it detect smoke > siren+ notification to gateway.

pdelrot-samea, 2020-12-29:
Bonjour,
With connected smoke detectors one of the group may detect smoke, start its own siren, and tells the others to start their sirens too.
I found this feature on connected smoke detectors for my house few years ago (using a proprietary network).

A way may exist to control the siren from Zigbee messages, only if the manufacturer designed the firmware for that.

I have one of these Tuya TS0205 (Raspbee2, deconz and HomeAssistant), I'll let you know.

Pascal

Mimiix, 2020-12-29:
@kane610 Maybe you got a solution here as its related to the HA integration?

pdelrot-samea, 2020-12-29:
https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/module-sdk-development-access/access-standard-zigbee/tuya-zigbee-smoke-sensor-access-standard?id=K9ik6zvm2tg0l

Kane610, 2020-12-29:
> @Kane610 Maybe you got a solution here as its related to the HA integration?

Once it's exposed as controllable over the rest api it's possible to implement as a switch in hass. Or be used using the deConz configure service in hass.

prinsmikal, 2024-02-24:
I really hope they will fix this option in the future, because when you buy many of these and have them displaced around the house. It would be really important that the other fire alarms would also be activated so to alert everyone in the house. Has a new firmware to this sensor been deployed?