dresden-elektronik/deconz-rest-plugin #3849: Heiman HS2SK-E

twobeass, 2020-12-09:
## Device

- Product name: HS2SK-E
- Manufacturer: Heiman
- Model identifier: SmartPlug
- Device type : Please remove all unrelated device types. 
  - Light
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]

The Device is only shown as light. The power measurement features are not available via API

SwoopX, 2020-12-09:
The device is fully supported

closed 2020-12-09 by SwoopX

Mimiix, 2020-12-09:
@SwoopX User reported on Discord that he cant use the power measurements from the websocket/api.

twobeass, 2020-12-09:
Correct, like Mimix said there are no Infos about Power Measurements in the REST API

twobeass, 2020-12-10:
I am sorry, after a full system restart, i got two new Sensors via REST named "type": "ZHAConsumption" and "type": "ZHAPower" that were missing before... 
Topic closed, Device is fully supported...