dresden-elektronik/deconz-rest-plugin #3850: No neighbor links visible in deCONZ 2.07.01

Einstein2150, 2020-12-09:
## Describe the question or issue you are having

After `apt upgrade` to the latest Version 2.07.01 everything seems working fine. After a full reboot of the raspberry pi the connection lines "Neighbor Links" are gone but the network is working normal.

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.07.01)
 - Device: RaspBee I
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

none

## Additional context

The issue occurred first after the reboot of the pi. After the first manual restart of the deCONZ GUI (without reboot of the whole system) the lines were visible...

Einstein2150, 2020-12-09:
Now after nearly exact 1 hour after boot the lines start drawing ...

[image]

What is causing this delay?

manup, 2020-12-09:
Odd, there is a new button `Neighbor Links` which can be used to disable drawing them but after startup it defaults to true.
(The reason for the button are large networks where drawing can be painfully slow)

jensflorian, 2020-12-09:
I can't find 2.07.01 in https://deconz.dresden-elektronik.de/raspbian/beta/

SwoopX, 2020-12-09:
Stable

Einstein2150, 2021-01-13:
Visualisation-bug was fixed by installing the latest raspbian update. Neighbor links are drawing now right after the startup of deCONZ.

closed 2021-01-13 by Einstein2150