dresden-elektronik/deconz-rest-plugin #3854: Can't update to the newest version

MasterofKungfu9999, 2020-12-09:
## Describe the question or issue you are having

I can't updat to the newest version of phoscon. I i use the steps pointed out in the wiki or on https://www.phoscon.de/de/conbee2/install#raspbian there it says that this version i use is the newest one. How can i get "Sonic Boomerang"?

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

MasterofKungfu9999, 2020-12-09:
Here is what i see when i use "apt-get update"

pi@raspberrypi:~ $ sudo apt-get update
OK:1 http://phoscon.de/apt/deconz buster InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
OK:3 http://archive.raspberrypi.org/debian buster InRelease
OK:4 https://deb.nodesource.com/node_10.x buster InRelease

With "udo apt-get upgrade deconz" i get this:

udo apt-get upgrade deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
deconz ist schon die neueste Version (2.07.01-raspbian-buster-stable).
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Das folgende Paket wurde automatisch installiert und wird nicht mehr benötigt:
  rpi-eeprom-images
Verwenden Sie »sudo apt autoremove«, um es zu entfernen.
Die folgenden Pakete sind zurückgehalten worden:
  binutils binutils-arm-linux-gnueabihf binutils-common libbinutils python-rpi.gpio raspi-config
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 6 nicht aktualisiert.

Mimiix, 2020-12-09:
Sonic boomerang is the latest version... 01.

Closed.

closed 2020-12-09 by Mimiix