dresden-elektronik/deconz-rest-plugin #3858: Network was not found after upgrading

Danny-P, 2020-12-09:
## Describe the bug
I upgrades my docker installation of deconz to the latest version.
When I try to login in http://: I can see the login Screen and log-in. But after that I can only see a screen which tells "Network was not found"
Docker logs tells:

```
23:07:39:037 dev can't create file: /dev/ttyACM0 via mknod (missing permissions)
23:07:39:084 dev can't create file: /dev/ttyACM0 via mknod (missing permissions)
23:07:39:983 failed to reconnect to network try=5

```
## Steps to reproduce the behavior
Upgrade to latest version. 
Try to log in on webapp: http://<ip>:<port>
Login is working, but directly after that the error page is shown.

## Expected behavior
Log-In was successfull and the app is working in the normal way and I can control lights and so on.

## Screenshots
[image]

## Environment
I am using a Conbee II Stick on a Linux machine.
I am using mathoc docker container in latest version
No USB extension cable
No other USB devices connected for testing.

## deCONZ Logs
```
23:07:39:037 dev can't create file: /dev/ttyACM0 via mknod (missing permissions)
23:07:39:084 dev can't create file: /dev/ttyACM0 via mknod (missing permissions)
23:07:39:983 failed to reconnect to network try=5

```
## Additional context

Mimiix, 2020-12-10:
Did you try sudo?

Danny-P, 2020-12-10:
I was able to solve that issue. 
There seems to be a problem with the docker run command I used. 

Until yesterday I used something like that: 
`docker run --env=DECONZ_DEVICE=/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2197591-if00 --env=UDEV=off --volume=/opt/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ --volume=/root/.local/share/dresden-elektronik/deCONZ --restart=always --device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2197591-if00:/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2197591-if00 --label='io.balena.architecture=amd64' --detach=true marthoc/deconz:stable`

But the device parameter seems not to work any longer in this way. So I used
`docker run --env=DECONZ_DEVICE=/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2197591-if00 --env=UDEV=off --volume=/opt/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ --volume=/root/.local/share/dresden-elektronik/deCONZ --restart=always --device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2197591-if00:/dev/ttyACM0 --label='io.balena.architecture=amd64' --detach=true marthoc/deconz:stable`

Now it is working again.

Mimiix, 2020-12-10:
@phdelodder or @marthoc can explain here :)?

fl0pp, 2020-12-10:
I have the same problem after upgrading firmware from 264A0700 to 26660700.

Screenshot before upgrade:
[image]

I have the following configuration in deCONZ add-on in Home Assistant:
`device: /dev/ttyACM1`

Host Operating System	Ubuntu 20.04 LTS
Home Assistant 0.118.5
Installed Add-ons	deCONZ (6.6.0), File editor (5.0.0), Let's Encrypt (4.11.0)
Docker Version	19.03.11
Supervisor Version	2020.12.6
Operating System Version	5.4.0-56-generic

Log:
```
21:18:25:851 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
21:18:25:857 CTRL. 3.27.221:18:25:880 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2198832-if00
21:18:25:905 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2198832-if00 / serialno: , ConBee II
21:18:25:905 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
21:18:25:960 parent process s6-supervise
21:18:25:960 gw run mode: docker/hassio
21:18:25:960 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
21:18:25:961 DB sqlite version 3.27.2
21:18:25:962 DB PRAGMA page_count: 53
21:18:25:962 DB PRAGMA page_size: 4096
21:18:25:962 DB PRAGMA freelist_count: 4
21:18:25:962 DB file size 217088 bytes, free pages 4
21:18:25:962 DB PRAGMA user_version: 7
21:18:25:962 DB cleanup
21:18:25:964 DB create temporary views
21:18:25:976 started websocket server at port 8081
21:18:25:977 [INFO] - Found file containing button maps. Parsing data...
21:18:25:979 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
21:18:25:982 [INFO] - Button maps loaded.
21:18:25:982 dlg action: Read binding table
21:18:25:983 found node plugin: libde_rest_plugin.so - REST API Plugin
21:18:25:984 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
[20:18:26] INFO: Starting Nginx...
2020/12/10 20:18:26 [notice] 329#329: using the "epoll" event method
2020/12/10 20:18:26 [notice] 329#329: nginx/1.14.2
2020/12/10 20:18:26 [notice] 329#329: OS: Linux 5.4.0-56-generic
2020/12/10 20:18:26 [notice] 329#329: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/12/10 20:18:26 [notice] 329#329: start worker processes
2020/12/10 20:18:26 [notice] 329#329: start worker process 1641
21:18:28:060 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
21:18:28:083 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2198832-if00 / serialno: , ConBee II
21:18:28:215 New websocket 172.30.32.1:37352 (state: 3) 
21:18:29:112 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2198832-if00 / serialno: , ConBee II
21:18:29:129 COM check bootloader
QObject::connect: Cannot queue arguments of type 'QTextBlock'
(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
21:18:29:141 
R21B18 Bootloader
21:18:29:141 Vers: 2.07
21:18:29:142 build: Jun 17 2019
21:18:29:142 , 08:48:53
21:18:31:135 COM check bootloader
21:18:31:146 
R21B18 Bootloader
21:18:31:147 Vers: 2.07
21:18:31:147 build: Jun 17 2019
21:18:31:147 , 08:48:53
172.30.32.2 - - [10/Dec/2020:20:18:31 +0000] "GET /api/config?_=1607631511247 HTTP/1.1" 200 264 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [10/Dec/2020:20:18:31 +0000] "GET /api/hassio_ingress/kXmBbXD9Gfehgy7l2keq8BxAlY-lx1DE8SLQv_JZYRs/api/config?_=1607631511270 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
21:18:31:892 Announced to internet http://dresden-light.appspot.com/discover
21:18:33:141 COM check bootloader
[... 23 lines cut ...]
R21B18 Bootloader
21:18:39:192 Vers: 2.07
21:18:39:192 build: Jun 17 2019
21:18:39:192 , 08:48:53
21:18:41:164 COM check bootloader
21:18:41:176 
R21B18 Bootloader
21:18:41:176 Vers: 2.07
21:18:41:176 build: Jun 17 2019
21:18:41:176 , 08:48:53
21:18:41:179 device disconnected reason: 5, index: 0
172.30.32.2 - - [10/Dec/2020:20:18:41 +0000] "GET /api/config?_=1607631521364 HTTP/1.1" 200 264 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [10/Dec/2020:20:18:41 +0000] "GET /api/hassio_ingress/kXmBbXD9Gfehgy7l2keq8BxAlY-lx1DE8SLQv_JZYRs/api/config?_=1607631521395 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
21:18:42:037 wait reconnect 15 seconds
21:18:42:066 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2198832-if00 / serialno: , ConBee II
21:18:43:037 wait reconnect 14 seconds
21:18:44:037 wait reconnect 13 seconds
21:18:45:037 wait reconnect 12 seconds
21:18:46:037 wait reconnect 11 seconds
172.30.32.2 - - [10/Dec/2020:20:18:46 +0000] "GET /api/config?_=1607631526432 HTTP/1.1" 200 264 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
172.30.32.2 - - [10/Dec/2020:20:18:46 +0000] "GET /api/hassio_ingress/kXmBbXD9Gfehgy7l2keq8BxAlY-lx1DE8SLQv_JZYRs/api/config?_=1607631526455 HTTP/1.1" 403 140 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36"
21:18:47:037 wait reconnect 10 seconds
21:18:47:038 start reconnect to network
21:18:48:037 wait reconnect 9 seconds
```

SwoopX, 2020-12-10:
Have you clicked on the button for the update?

fl0pp, 2020-12-10:
Yes, I clicked the button to update.
I found a related issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3385, so I flashed the old firmware  `GCFFlasher -d COM3 -t 60 -f ../firmware/deCONZ_ConBeeII_0x26580700.bin.GCF` following https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

This fixed it for me.

SmartHomeNewbie, 2020-12-12:
> I was able to solve that issue.
> There seems to be a problem with the docker run command I used.
>
> Until yesterday I used something like that:
> `docker run [...]

simonjro, 2020-12-12:
Also having this issue since upgrading today:

`R21B18 Bootloader
23:48:39:112 Vers: 2.07
23:48:39:112 build: Jun 17 2019
23:48:39:112 , 08:48:53
23:48:39:118 device disconnected reason: 5, index: 0
23:48:39:999 wait reconnect 15 seconds`

labaland, 2020-12-13:
me2

dico, 2020-12-15:
I also encountered this problem today. Using it with Home Assistant. Not sure why this happened now. Couple of days since I upgraded anything. Just tried to add some new lights today and suddenly Phoscon shows "Network was not found...". I have seen someone confirming that downgrade works. Not sure how I even can downgrade or change the docker-parameters with HA/Hass.io.

simonjro, 2020-12-15:
> I also encountered this problem today. Using it with Home Assistant. Not sure why this happened now. Couple of days since I upgraded anything. Just tried to add some new lights today and suddenly [...]

I found the issue wasn't with the DeCONZ version but the Conbee II firmware version and that's what I was able to downgrade. Easiest done using a PC, I followed [this](https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/) guide, hope it helps.

SwoopX, 2020-12-15:
You guys clicked on the "update firmware" button in Phoscon?

simonjro, 2020-12-15:
> You guys clicked on the "update firmware" button in Phoscon?

I did yes.

dotbanana, 2020-12-15:
Same here, clicked update firmware in Phoscon and now i get "Network was not found" after logging in. 
I don't use Docker but manually installed Deconz on raspian.  
Anyone an idea how to fix this properly without downgrading?

dico, 2020-12-15:
I wasted all day on this... Reading logs, github-issues... Restarted Home Assistant and docker-container several times. Tried to flash a new firmware from several Windows PCs, but none of them could recognize the Conbee-stick. At the end I just gave up, ordered a Conbee II and stucked the old one back in my NUC with HA.... and it started working 👍😒

Bananabombo, 2020-12-16:
> Same here, clicked update firmware in Phoscon and now i get "Network was not found" after logging in.
> I don't use Docker but manually installed Deconz on raspian.
> Anyone an idea how to fix this properly without downgrading?
> 
> Edit: Fixed it by doing an update manually as described here: https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/
> But: Problem is not fixed immediatly after a restart. I thought at first it did not help but after waiting 2 minutes or so it suddenly starts working.

Same for me as well. This window ("Network was not found...") seems to be like a timout window regarding the connection. I just bought a Conbee, installed it, updated the firmware and boom after login there was just this screen.

Sometimes it even works all of the sudden.

Mimiix, 2020-12-16:
For those having this issue after clicking firmware update button, follow this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Using any virtualisation software is **not** recommended and causes issues as you describe.

As OP's issue is solved, i'll close this issue.

closed 2020-12-16 by Mimiix

monpelaud, 2020-12-16:
Hi,
Same issue with deCONZ_RaspBeeII_0x26680700.bin.GCF  firmware and no issue with RaspBeeII_0x26660700.bin.GCF firmware.
For update and downgrade firmware of my RaspBeeII I use Raspbian Buster Desktop SD-Card Images from https://phoscon.de/en/conbee2/sdcard#downloads.
No virtualisation software and I update manually my RaspBeeII with ssh command line, not by clicking firmware update button.
1) Update firmware deCONZ_RaspBeeII_0x26680700.bin.GCF issue Network was not found
2) Downgrade RaspBeeII_0x26660700.bin.GCF no issue, works fine
3) Update firmware deCONZ_RaspBeeII_0x26680700.bin.GCF issue Network was not found
4) Downgrade RaspBeeII_0x26660700.bin.GCF no issue, works fine
etc...
There is really an issue with deCONZ_RaspBeeII_0x26680700.bin.GCF firmware, it's not due to update process!

Best regards

hzmmbm, 2020-12-17:
Hi,
I have a setup with Conbee1 & Phoscon SD Card image. Last week I updated the Image using the apt upgrade method.
2.07.00 / 15.11.2020 and after I clicked the Firmware update Button in GUI Phoscon APP.
Firmware is now 26390500

After the update I got the issue Network was not found.
a restart solved this.
Until yesterday it was running fine but in the evening I recognized that Zigbee switches are dead.
Login into the Phoscon APP shows up issue Network was not found again.
i erstarrtest twice but didn't help. after plug out an in it worked again.
**Is the solution I read in the previous post to update the firmware again manually?**
Also wondering because Conbee was running nearly 2 years without issues.

Thanks

Mimiix, 2020-12-17:
@hzmmbm  You could be having interference issues. Anyway: Please open a new user question issue.

Tillmann-E, 2020-12-18:
Same issue here. Updated my new Conbee2 manually to 26680700. 
Trying to downgrade in docker failed with flashing error. 
Thought that I had successfully bricked the conbee...
Downgrading on Windows 10 luckily worked, now back and running again.

kurtjy, 2020-12-21:
Yeah, me too, I clicked update from within hassio, after that 'network was not found'
Shutdown HA, and did the upgrade on windows following https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually  Works flawlessly, thanks for pointing me there

Thing is I did not 'try do downgrade' I updated to last firmware found here http://deconz.dresden-elektronik.de/deconz-firmware/ being for conbee2: deCONZ_ConBeeII_0x26680700.bin.GCF 

So yeah problem is not with the firmware imho, more like when I tried to update it from HA, it didn't succeed..

lukaszbel, 2020-12-22:
I have the same problem :( what's a shit... WIn10 also doen't work

```
Reboot device COM3 (ConBee II)
action: update firmware after 8 ms
flashing 126132 bytes: |=
dev: ERROR writing image (15 ms)
Reboot device COM3 (ConBee II)
action: update firmware after 10 ms
flashing 126132 bytes: |=
dev: ERROR writing image (15 ms)
Reboot device COM3 (ConBee II)
action: update firmware after 9 ms
flashing 126132 bytes: |=
dev: ERROR writing image (16 ms)
Reboot device COM3 (ConBee II)
action: update firmware after 10 ms
flashing 126132 bytes: |=
dev: ERROR writing image (14 ms)
Reboot device COM3 (ConBee II)
action: update firmware after 7 ms
flashing 126132 bytes: |=
dev: ERROR writing image (14 ms)
retry, failed
```

andriej, 2020-12-22:
@lukaszbel it's not the firmware for conbee looking at the size.
[this is the proper one](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26680700.bin.GCF)

lukaszbel, 2020-12-22:
Eh so why phoscon answered mi this version? Not works, but something is wrong. Nodered doesn't see the correct status os Ikea motion sensor.

lukaszbel, 2020-12-22:
Ok after a few restarts it is working. Dresden Elektronik do some with that.

Mimiix, 2020-12-22:
@lukaszbel I'm getting the button disabled for this reason. However, depending on your install method it is _doomed_ to fail. VM's or Docker is not recommended at all to update it with as the documentation tells.

lukaszbel, 2020-12-22:
I do not use venv or Docker env. I have instaled standalone deconz on RPI Debian distribution.

faaafo, 2020-12-25:
Hi Phoscon / Conbee / Deconz / whathever

What's the solution?

dico, 2020-12-25:
> Hi Phoscon / Conbee / Deconz / Dresden Elektronik / whatever
> 
> What's the solution?

I'm not a developer here, but as I wrote above here; I wasted a whole day reading about this - trying to fix my own setup. It looks like there are several solutions. In another Github-thread, someone mentioned that there was a security-update in the new firmware, that would take all from 5 minutes and up to one hour, before it was ready to use. The solution here was just to wait.

Another solution I found was changing some of the docker run command.

For many there is enough with just a restart.

This error occured for me when trying to add new lights. I had not upgraded the firmware for a long time. So it isn't only a firmware issue (if it's a firmware issue at all...?) The solution for me was just to unplug the Conbee and plug it in again. Which is a very annoying solution, after using a whole day trying to fix it. I also gave up in this process and ordered a Conbee II, that I migrated to a couple of days ago.

monpelaud, 2020-12-26:
I solved my issue by upgrading firmware of my RaspBeeII to deCONZ_RaspBeeII_0x26690700.bin.GCF.
Now it works fine! I dont no why with deCONZ_RaspBeeII_0x26680700.bin.GCF it didn't work.
I used the same manual process for every firmware upgrade but after several upgrade/downgrade to firmware deCONZ_RaspBeeII_0x26680700.bin.GCF, it's never worked fine. I upgraded to deCONZ_RaspBeeII_0x26690700.bin.GCF and now all works fine !!!
Why ?  I don't know.

SpoofEx, 2021-01-01:
I experienced the same issue using deCONZ for several months succesfully on my Home Assistant system.

I followed the instructions on [how to upgrade the deCONZ firmware manually](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) 
[image]

After the firmware upgrade to [deCONZ_ConBeeII_0x26680700.bin](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26680700.bin.GCF) from `30 november 2020` I still experienced the same problem:
[image]

After I rebooted Hassio (Home Assistant) on the issue resolved:
[image]

It might sound an obvious thing to do, but now you know ;)

jaruba, 2021-01-04:
I can't update the Conbee II either, for me, it looks like the last version that can connect is `26570700` (from 27th April 2020), all versions after that (5 versions) simply cannot connect.

I tried restarting after installing a new version, i tried unplugging the Conbee and pluging it in again.. No use, before updating I used `264a0700` (which is the previous version from the one that works now), I stuck with this version for months because I couldn't update before either.

I would really like to find some way to use a more recent version..

Mimiix, 2021-01-04:
@jaruba please open a new issue with all detailed info requested in the template.

jaruba, 2021-01-04:
@Mimiix 
I created an issue for it: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4102
Thanks

Jalle19, 2021-01-25:
I have the same issue. I'm running Home Assistant Operating System in a KVM virtual machine with the Conbee II stick forwarded to the VM. I used to have a similar setup at home and upgrading the firmware definitely used to work. This is a new setup with a new stick, so it seems like a regression in the updater?

fpoussin, 2021-01-26:
it seems to be cause by the env var DECONZ_DEVICE being ignored