dresden-elektronik/deconz-rest-plugin #3860: LOM004 Philips hue plug is sending a brightness attribute (bri)

MattL0, 2020-12-10:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Firmware version: 26680700
 - deCONZ version: 2.07.01
 - Device: Conbee II

SwoopX, 2020-12-10:
Can you share a screenshot of the node with the clusters expanded in deconz GUI?

If it doesn't have/support bri, we can filter it out...

ebaauw, 2020-12-10:
It’s quite common for on/off plugs to carry a _Level Control_ cluster, so they can respond to _Move to Level ( with On/Off)_, and be controlled by switches that send that command instead of _On_ and. _Off_.

Mimiix, 2020-12-10:
@ebaauw So this is something that Homeseer needs to sort-out?

wvuyk, 2020-12-10:
There was another post on this github (sorry cannot find it) where the remark made was that for on/off plugs the bri attribute will or should be omitted. And indeed for most plugs the bri attribute is not present. So far I can see plugs used, the LOM004 is the exception to this. 
I can try to recognise the plugs based on the type, but I remember that the type can differ too based on the model? Too many different methods to determine what device is presented....

MattL0, 2020-12-10:
> Can you share a screenshot of the node with the clusters expanded in deconz GUI?
> 
> If it doesn't have/support bri, we can filter it out...

I would want to.. but I am running headless here, with minimal install

ebaauw, 2020-12-10:
> So this is something that Homeseer needs to sort-out?

I surely hope not; it should be fixed by the REST API Plugin.  Check the closed issues, we've done so for several plugs already (including the beloved innr SP120).

MattL0, 2020-12-11:
I deployer a docker install. so it was easy to launch vnc . 

Is that what you need ? 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ebaauw, 2020-12-11:
That should do it.

closed 2020-12-14 by manup