dresden-elektronik/deconz-rest-plugin #3861: Firmware not connected, Conbee2 with Homeassistant using Raspbee

alexejan, 2020-12-10:
## Describe the question or issue you are having
Firmware not connected.

## Screenshots
[image]

## Environment

 - Host system: (Raspberry Pi 3 B+)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (26580700)
 - deCONZ version: ( 6.5.0)
 - Device: ( ConBee II)
 - Do you use an USB extension cable: (yes) 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
09:22:09:415 COM check bootloader
09:22:11:336 failed to reconnect to network try=5
09:22:11:425 COM timout query bootloader, assume application
09:22:12:852 device state timeout ignored in state 2
09:22:15:835 device state timeout ignored in state 2
09:22:16:336 try to reconnect to network try=6
09:22:18:391 dev /dev/ttyAMA0
09:22:20:835 device state timeout ignored in state 2
09:22:21:336 try to reconnect to network try=7
09:22:25:835 device state timeout ignored in state 2
09:22:26:336 try to reconnect to network try=8
09:22:28:396 dev /dev/ttyAMA0
09:22:30:835 device state timeout ignored in state 2
09:22:31:336 try to reconnect to network try=9
09:22:35:835 device state timeout ignored in state 2
09:22:36:335 try to reconnect to network try=10
09:22:38:389 dev /dev/ttyAMA0
09:22:40:836 device state timeout ignored in state 2
09:22:41:336 reconnect network failed, try later
09:22:41:337 start reconnect to network
09:22:43:836 device state timeout ignored in state 2
09:22:46:336 try to reconnect to network try=1
09:22:46:836 device state timeout ignored in state 2
09:22:48:394 dev /dev/ttyAMA0
09:22:49:836 device state timeout ignored in state 2
09:22:51:335 try to reconnect to network try=2
09:22:55:835 device state timeout ignored in state 2
09:22:56:336 try to reconnect to network try=3
09:22:58:392 dev /dev/ttyAMA0
09:23:00:836 device state timeout (handled)
09:23:00:841 device disconnected reason: 1, index: 0
09:23:01:336 failed to reconnect to network try=4
09:23:01:337 wait reconnect 15 seconds
09:23:02:335 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
09:23:02:336 device state timeout ignored in state 1
09:23:02:337 wait reconnect 14 seconds
09:23:03:335 wait reconnect 13 seconds
09:23:04:335 wait reconnect 12 seconds
09:23:05:335 wait reconnect 11 seconds
09:23:06:336 failed to reconnect to network try=5
09:23:06:337 wait reconnect 10 seconds
09:23:07:335 wait reconnect 9 seconds
09:23:08:336 wait reconnect 8 seconds
09:23:08:390 dev /dev/ttyAMA0
09:23:09:335 wait reconnect 7 seconds
09:23:10:336 wait reconnect 6 seconds
09:23:11:336 failed to reconnect to network try=6
09:23:11:337 wait reconnect 5 seconds
09:23:12:336 wait reconnect 4 seconds
09:23:13:336 wait reconnect 3 seconds
09:23:14:335 wait reconnect 2 seconds
09:23:15:335 wait reconnect 1 seconds
09:23:15:399 dev /dev/ttyAMA0
09:23:15:400 COM: /dev/ttyAMA0 / serialno: , RaspBee
09:23:15:413 COM check bootloader
09:23:16:336 failed to reconnect to network try=7
09:23:17:421 COM timout query bootloader, assume application
09:23:18:392 dev /dev/ttyAMA0
09:23:18:849 device state timeout ignored in state 2
09:23:21:335 try to reconnect to network try=8
09:23:21:836 device state timeout ignored in state 2
09:23:24:836 device state timeout ignored in state 2
09:23:26:335 try to reconnect to network try=9
09:23:28:397 dev /dev/ttyAMA0
09:23:30:835 device state timeout ignored in state 2
09:23:31:336 try to reconnect to network try=10
09:23:35:835 device state timeout ignored in state 2
09:23:36:335 reconnect network failed, try later
09:23:36:337 start reconnect to network
09:23:38:394 dev /dev/ttyAMA0
09:23:38:836 device state timeout ignored in state 2
09:23:41:335 try to reconnect to network try=1
09:23:41:835 device state timeout ignored in state 2
09:23:44:835 device state timeout ignored in state 2
09:23:46:336 try to reconnect to network try=2
09:23:48:390 dev /dev/ttyAMA0
09:23:50:835 device state timeout ignored in state 2
09:23:51:336 try to reconnect to network try=3
09:23:55:836 device state timeout ignored in state 2
09:23:56:335 try to reconnect to network try=4
09:23:58:397 dev /dev/ttyAMA0
09:24:00:835 device state timeout ignored in state 2
09:24:01:336 try to reconnect to network try=5
09:24:05:835 device state timeout (handled)
09:24:05:841 device disconnected reason: 1, index: 0
09:24:06:336 failed to reconnect to network try=6
09:24:06:337 wait reconnect 15 seconds
09:24:07:335 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
## Additional context
Using ethernet cable

Added in config.txt
enable_uart=1
dtoverlay=pi3-miniuart-bt

HA
device: /dev/ttyAMA0

Works fine on Windows 10.

SwoopX, 2020-12-10:
Try updating through Windows and then try again

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26680700.bin.GCF

Mimiix, 2020-12-10:
Also make sure to have the correct device serial location in the configuration page of the addon.

alexejan, 2020-12-10:
@Mimiix Is device serial location the same as in device: /dev/ttyAMA0 ?

alexejan, 2020-12-10:
Thank you @SwoopX @Mimiix 
It is working now!
Like you sad. I hade to update the firmware.  And then add the device serial in the configuration page (/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_serialnumberHERE-if00)

Mimiix, 2020-12-10:
Case closed 👍

closed 2020-12-10 by Mimiix