dresden-elektronik/deconz-rest-plugin #3862: unknown neighbour messages after upgrade to v2.7.1

posixx, 2020-12-10:
## Describe the bug

I upgraded yesterday from v2.05.88 to v2.7.1. Currently a lot of "unknown neighbour" messages scroll though the log. Everything is working as expected though.. Any possibility to reset or clean?

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Intel NUC
 - Running method: HASS
 - Firmware version: latest
 - deCONZ version: 2.7.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? ZWave / DSMR

## deCONZ Logs

11:51:28:326 unknown neighbor 0x001788010274C18D (0x7ADA), lqi: 95, rssi: -82
11:51:33:011 unknown neighbor 0x0017880102741E25 (0x155C), lqi: 55, rssi: -87
11:51:34:225 unknown neighbor 0x001788010274B2F8 (0xD13F), lqi: 55, rssi: -87
11:51:35:320 unknown neighbor 0x001788010274C154 (0x5981), lqi: 111, rssi: -80
11:51:38:697 unknown neighbor 0x001788010274B372 (0xAD67), lqi: 127, rssi: -78
11:51:39:712 unknown neighbor 0x00178801027BDB77 (0xB9C4), lqi: 191, rssi: -70
11:51:51:826 unknown neighbor 0x001788010274C86B (0x1B02), lqi: 71, rssi: -85
11:51:52:938 unknown neighbor 0x00178801038D6173 (0xB2AF), lqi: 2, rssi: -93
11:51:55:127 unknown neighbor 0x0017880102741D63 (0x0DE7), lqi: 87, rssi: -83
11:51:58:411 unknown neighbor 0x001788010274B23F (0xDCD9), lqi: 71, rssi: -85
11:52:00:613 unknown neighbor 0x001788010274C1C6 (0xE031), lqi: 15, rssi: -92
11:52:06:135 unknown neighbor 0x0017880102307374 (0xA11F), lqi: 10, rssi: -92
11:52:08:329 unknown neighbor 0x00178801027D27BE (0x161B), lqi: 255, rssi: -52

## Additional context

posixx, 2020-12-10:
These are also logged once in a while:

12:17:20:836 0x001788010274C182 error APSDE-DATA.confirm: 0xE9 on task

Mimiix, 2020-12-10:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

The 0xE9 is described there

manup, 2020-12-10:
The unknown neighbor message is just informative, it's printed when a command was received from a node one hop away which isn't found in the neighbor table itself. It's not related to a problem just some log print to help figure things out :)

posixx, 2020-12-10:
Ok but will this resolve itself? Never seen these before, now about one every 3 seconds...

manup, 2020-12-10:
After startup the neighbor tables of the coordinator get queried which can take few seconds up to minutes.
For neighbors which are listed there the debug print won't appear anymore.

However neighbor tables are limited, so for commands received from devices which aren't in there the print will still happen. I think for the next version raising the debug level might be good to not get too many lines here.

posixx, 2020-12-10:
@manup; is it possible somehow to clear these tables so the gateway will fill with actual lights being found?

manup, 2020-12-11:
The neighbor tables (all routers and coordinators have one) are maintained automatically by the respective firmware. No need to clear them. They hold entries to active end-device children and routers which are in direct 1-hop range.

closed 2020-12-11 by posixx