dresden-elektronik/deconz-rest-plugin #3864: Sonoff motion sensor (snzb-03) not full discovery

SuperSpe, 2020-12-10:
## Describe the question or issue you are having

So I just installed the raspbee II module and scan for 3 sensors, all sonoff, 2 motion sensor (snzb-03) and 1 temperature/umidity (snzb-02).
For the temperature sensor all is ok, I see it on phoscon app inside sensors menu and in HA. 
Instead for the two motion sensors, the discovery was ok but I cannot see them on the sensors menu of phoscon app. 
I can see them on the deconz app and in HA too...they are working. I try with an automation and is ok.
Is it normal?

## Screenshots
[image]
[image]
[image]

## Environment
 - Host system: Raspberry Pi 3
 - Running method: hassio: 4.15
                                home assistant: 0.118.4
 - Firmware version: (26680700)
 - deCONZ version: (6.6.0)
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
no log

## Additional context
thanks

Mimiix, 2020-12-10:
Hi!

Phoscon needs to add support themselves to show them. This device isn't fully supported yet. Therefore, it won't show. 

So yes, it's normal :)

#caseclosed!

closed 2020-12-10 by Mimiix

SuperSpe, 2020-12-10:
should I open an issue to phoscon?

Mimiix, 2020-12-10:
After somone requests it :) https://github.com/dresden-elektronik/phoscon-app-beta

SuperSpe, 2020-12-10:
How to request it? I cannot see on the the issue section how to request new item

Mimiix, 2020-12-10:
Yep.

SuperSpe, 2020-12-10:
ok thanks I'll do