dresden-elektronik/deconz-rest-plugin #3870: 2.7.1 GUI (still) doesn't handle bitmap fields correctly

ebaauw, 2020-12-10:
## Describe the bug

GUI doesn't handle bitmap fields correctly since a couple of releases.  Still not fixed in 2.7.1.
- In _Cluster Info_ panel, the field doesn't show at all.
- In _Attribute Editor_ windows, the check boxes show when the values is 0, the value isn't updated on _Read_ and no checkboxes show when the value is non-0.

## Steps to reproduce the behavior

Run deCONZ.

## Expected behavior

Check boxes should show and reflect value.

## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.7.1
 - Device: RaspBee II
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

manup, 2020-12-11:
Confirmed, it seems to only work in the Cluster Info panel command view but not in the Attribute Info.
I've also seen that sometimes the fields don't show up at all in the Cluster Info panel. I guess this broke when adding more support for manufacturer specific commands and attributes.

Looking forward to fix this in the upcoming beta.

manup, 2021-01-09:
@ebaauw  Can this be closed now, should work with recent version?

ebaauw, 2021-01-14:
Seems to be solved in v2.9.0.  Thanks.

closed 2021-01-14 by ebaauw