dresden-elektronik/deconz-rest-plugin #3874: One IKEA Range Extender shown as an extender, the other one shown as a light?

fribse, 2020-12-11:
## Describe the bug

I have two range extenders, one for each of my blinds currently on deconz.
One shows up in phoscon as a range extender, the other one shows up as a light.
[image]
[image]

## Steps to reproduce the behavior

  1. Remove the extender from deconz
  2. Readd it to deconz

But the weird part is, that one of the extenders will always show as an extender, and the other one always as a light.
I've gone to the deconz graphical view, and did a lot of 'read' on the different clusters, but that didn't change anything.

## Expected behavior

Extenders should be registered as extenders.
## Screenshots

## Environment

 - Host system: Intel NUC with PVE
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.7.1
 - Device: ConBee II
 - Do you use an USB extension cable: yes USB3
 - Is there any other USB or serial devices connected to the host system? Aeotec Z-stick Gen 5 I

## deCONZ Logs

## Additional context

Extender registered as a light:
[image]
Excender registered as an extender:
[image]

Mimiix, 2020-12-11:
Intresting!

Funny how the Date Code is the only different thing. 

The Light seems to be updated to the latest firmware tho. That could be the difference?

SwoopX, 2020-12-11:
Obviously, the model identifier also deviates. The list of clusters per node would be interesting now...

ebaauw, 2020-12-11:
And a listing of the API resources.

fribse, 2020-12-11:
@SwoopX , not sure what you mean, but enlighten me, and I'll get the info pulled out 😄 
The one registered as a light
[image]
The one registered as an extender:
[image]

The api list of lights:
```
{
  "1": {
    "etag": "b253e752f44f9c89b3eb20364bcbccdf",
    "hascolor": false,
    "lastannounced": "2020-09-19T11:33:58Z",
    "lastseen": "2020-09-19T11:36Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI control outlet",
    "name": "Nathalie Mirror",
    "state": {
      "alert": "none",
      "on": true,
      "reachable": false
    },
    "swversion": "2.0.022",
    "type": "On/Off plug-in unit",
    "uniqueid": "00:0d:6f:ff:fe:9c:30:f8-01"
  },
  "10": {
    "etag": "727d17407a4b1f33e34cd908844204c5",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-12-11T10:53Z",
    "manufacturername": "Heiman",
    "modelid": "WarningDevice",
    "name": "Siren2",
    "state": {
      "alert": "none",
      "reachable": true
    },
    "swversion": "2018.03.21",
    "type": "Warning device",
    "uniqueid": "00:0d:6f:00:16:1b:b7:ca-01"
  },
  "11": {
    "etag": "6b9c6809134bd36f1724111116016289",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-12-11T10:31Z",
    "manufacturername": "dresden elektronik",
    "modelid": "ConBee II",
    "name": "ConBeeII stick",
    "state": {
      "reachable": true
    },
    "swversion": "0x264a0700",
    "type": "Unknown",
    "uniqueid": "00:21:2e:ff:ff:05:1f:1a-01"
  },
  "12": {
    "etag": "5a8f9e676d53b63e222c707999703526",
    "hascolor": false,
    "lastannounced": "2020-12-08T12:46:47Z",
    "lastseen": "2020-12-11T10:42Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI transformer 10W",
    "name": "Office",
    "state": {
      "alert": "none",
      "bri": 60,
      "on": true,
      "reachable": true
[... 497 lines cut ...]
      "reachable": true
    },
    "swversion": "1.2.245",
    "type": "Dimmable light",
    "uniqueid": "d0:cf:5e:ff:fe:c8:09:b1-01"
  },
  "9": {
    "etag": "956393a7606481b795a90fb0c2d05211",
    "hascolor": false,
    "lastannounced": "2020-12-07T13:26:31Z",
    "lastseen": "2020-12-11T10:57Z",
    "manufacturername": "_TZ3000_g5xawfcq",
    "modelid": "TS0121",
    "name": "IKEA-hub",
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "84:2e:14:ff:fe:16:1e:1c-01"
  }
}
```

SwoopX, 2020-12-11:
@fribse Thanks, that's what we meant. So we have different mac prefixes and slightly different model identifiers here...

SwoopX, 2020-12-11:
@fribse Doe they both have the same manufacturer code in the node info panel?

fribse, 2020-12-11:
This is the one registered as a light:
[image]
This is the one registered as a range extender:
[image]
So for me it looks like it.

ebaauw, 2020-12-11:
The REST API Plugin exposes both correctly, and identically.  I would suspect this to be a Phoscon issue.  Note the different mac address prefix, maybe Phoson (incorrectly) uses more than the `type` to recognise the repeater?  And changes `A` to `Å` after it has?

jensflorian, 2020-12-11:
Note also the differences between signal repeater and Signal Repeater

ebaauw, 2020-12-11:
Good catch!

SwoopX, 2020-12-11:
Agreed that this is related to Phoscon.

Mimiix, 2020-12-11:
Moving this to Phoscon 👍