dresden-elektronik/deconz-rest-plugin #3876: Support for Bticino K4027C broken

zinoalex, 2020-12-11:
## Describe the bug
With the latest release (2.7.1) the K4027C (shutter control) switch does not work anymore in home assistant exept stop and up command.
The only way to make it work is to use the deconz vnc gui.
Also in HA if you send the "down" command activate the Up action on the switch.
Any idea why?
I've seen that with 2.6.0 release they introduced the support for L4027C, maybe is because of that.

## Steps to reproduce the behavior

Home Assistant use cover buttons

## Expected behavior
## Screenshots
## Environment
Host system: Dell Optiplex 3070
Running method: Win 10 Pro (Virtual box with Home Assistant)
Firmware version: (0x26680700)
deCONZ version: (2.7.1)
Device: (ConBee II)

## deCONZ Logs

## Additional context

Mimiix, 2020-12-11:
Can you check the output of the events?

zinoalex, 2020-12-11:
Hi, with down command in gui:
11:28:58:007 ZCL attribute report 0x00047400007DB9BD for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
Stop command:
11:29:02:697 ZCL attribute report 0x00047400007DB9BD for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000

zinoalex, 2020-12-11:
Seems a problem of reporting the correct lift value 
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3549

zinoalex, 2020-12-11:
Maybe @Smanar knows how to address the issue

Smanar, 2020-12-11:
It seem I have this problem every 2 month ...

The first issue was https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532
Your device have his firmware "out of the box" or you have updated it ? It s called "Sw Build ID" on capture, and he have 0x0B.

>The only way to make it work is to use the deconz vnc gui.

And the order is not reversed in it ?

zinoalex, 2020-12-11:
SW build 0016, with deconz gui the order is correct but then in HA and phoscon app the buttons doesn't work anymore

Smanar, 2020-12-11:
I m asking to the other user with version 0B to be sure it s a version problem.
Just bu curiosity you know how to use the API, to send only "open" or "lift" command but never both in same time ?

zinoalex, 2020-12-11:
No i just press the button in the gui

bouksy, 2020-12-12:
dont know if it's related, but I cant no more associate during the integration process.
I've seen that deconz package (I'm on ubuntu 20.04 server) has been recently updated 
I did as before (activate Authenticate App) and nothing happens except "Unknown error occurred" which is not quite usefull :)

Version 2.7.1
Firmware: 26680700

Smanar, 2020-12-12:
No I realy don't see what you are talking about. "Authenticate App" is to enable association with third app, and I realy don't see where you are seing ""Unknown error occurred"

@zinoalex not possible to custom command send by HA
Can you at least show a complete JSON and say the covering position.

bouksy, 2020-12-12:
I'll send you print screens asap

bouksy, 2020-12-12:
ok to sum up the situation.
First I saw this morning that I the switchs were greyed in the UI (therefore I couldn't swich on my lights anymore) . 
Then I tried to delete deconz in the integration page and add it again.
And then when I tried to do it, i got the error.

[image]

I got an error:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_protocol.py", line 422, in _handle_request
    resp = await self._request_handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_app.py", line 499, in _handle
    resp = await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/aiohttp/web_middlewares.py", line 118, in impl
    return await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/forwarded.py", line 71, in forwarded_middleware
    return await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/request_context.py", line 18, in request_context_middleware
    return await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/ban.py", line 72, in ban_middleware
    return await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/auth.py", line 127, in auth_middleware
    return await handler(request)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/view.py", line 129, in handle
    result = await result
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/config/config_entries.py", line 169, in post
    return await super().post(request, flow_id)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/http/data_validator.py", line 60, in wrapper
    result = await method(view, request, *args, **kwargs)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/helpers/data_entry_flow.py", line 106, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/data_entry_flow.py", line 155, in async_configure
    result = await self._async_handle_step(flow, cur_step["step_id"], user_input)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/data_entry_flow.py", line 213, in _async_handle_step
    result: Dict = await getattr(flow, method)(user_input)
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/deconz/config_flow.py", line 141, in async_step_link
    api_key = await async_get_api_key(session, **self.deconz_config)
  File "/srv/homeassistant/lib/python3.8/site-packages/pydeconz/utils.py", line 22, in async_get_api_key
    response = await async_request(session.post, url, auth=auth, data=data)
  File "/srv/homeassistant/lib/python3.8/site-packages/pydeconz/utils.py", line 89, in async_request
    _raise_on_error(response)
  File "/srv/homeassistant/lib/python3.8/site-packages/pydeconz/utils.py", line 105, in _raise_on_error
    raise_error(data["error"])
  File "/srv/homeassistant/lib/python3.8/site-packages/pydeconz/errors.py", line 58, in raise_error
    raise cls("{} {}".format(error["address"], error["description"]))
pydeconz.errors.pydeconzException: / link button not pressed

I will try to install a previous version of deconz...

bouksy, 2020-12-12:
ok i've tested with previous version of deconz ....so it ain't related :(... I l keep up my research

Smanar, 2020-12-12:
Ha ok, so nothing to see with this device, it s a deconz problem.
Do you have deconz on the same machine VM than HA, if yes "Authenticate App" is useless on local host.

Have you check if phoscon is working ? It can be just deconz that is out of order.

bouksy, 2020-12-12:
ok then I finally managed to get it up. I did try a lot of things, therefore dont know what worked.
But I did reinstall deconz, reset the gateway, and loaded again my backup.
and FYI, it is indeed on the same PC (just bought a NUC recently)
sorry I don't have any bticino device. I'm indeed starting to think about changing my door bell but this one is a abit too expensive:p
anyway many thanks

Smanar, 2020-12-13:
@zinoalex do you have the device json (with all values) and the shutter position at this moment ?

zinoalex, 2020-12-13:
Well, i guess you need to tell me how to do it

zinoalex, 2020-12-14:
After some tests and upgrading HA to the latest version, I think that the problem is the closed/open state report that is now inverted with the latest version of deconz. The commands work as expected but are not usable straight away, if you want to close it the close button is greyed out (because the report state is closed) so you need to press open-stop-close and then it works. 
Long story short we need to invert the status reporting. Do you think I need to upgrade the switch firmware?

Smanar, 2020-12-14:
Yes it s a long story.

Before the last modification targetLiftZigBee was always reverted for only legrand device
After it it was reverted only if the firmware version is inferior at 33 (0x21) but for Legrand AND bticino.

The problem is for Legrand I have checked all firmware and new and I know the version that have the modification but not for Bticino.

@dpmworld  have the problem on his firmware 0x0B, it s for that the code was made. But You have 0x16, I can use your value for bticino, like I have done for Legrand.

So I can made change in blind mode, using this hypothesis, but for that you realy need to compile the code to test it on your side.
Or I can send you command to test using curl to check all command ?

zinoalex, 2020-12-15:
I don't know how to compile the code i'm afraid. The only thing I can do is to upgrade the switch firmware and see if something changes.

dpmworld, 2020-12-15:
@Smanar the firmware is 00B
If anyone know where to find legrand firmware I would be pleased to try an upgrade....
[image]

Smanar, 2020-12-15:
Ok So I have return from netatmo, Legrand and Bticino use same fimware, so 

> 0x0B > need reverse (code reverse) WORKING
> 0x1A > need to reverse (code reverse) WORKING
> 0x16 > No need to reverse (code reverse) NOT WOKING
> 0x21 > No need to reverse (code don't reverse) WORKING

So we have a problem, 0x1A is > 0x16 ....

dpmworld, 2020-12-15:
L4027C is working flawlessy

zinoalex, 2020-12-15:
Tomorrow i'll upgrade, i will let you know

Smanar, 2020-12-15:
@zinoalex do you need the firmware or have the gateway ?

If you upgrade it, it will work (you will have version > 0x21), but it don't solve the problem ...
I have asked to another user that have several firmware version to have his return about it.

zinoalex, 2020-12-15:
I have the original gateway buy Is a bit of pain to unpair It, upgrade, ecc
ecc. If you have the firmware It should be easier

Smanar, 2020-12-15:
Pm me on discord, I have the Legrand 0x23 ATM.
And you need the last deconz version, a bug was corrected recently.

Smanar, 2020-12-16:
BTW someone can say me if it have a value on attribute 0x0000, on output cluster (gray) 0x00019 , for this device ?
It need to be the gateway mac adress.

zinoalex, 2020-12-16:
[image]

zinoalex, 2020-12-17:
update for everyone interested.
I've upgraded to latest firmware 001c (v28 on bticino app) with original gateway, up/down commands works but the report of position is reversed so it makes troubles with HA. 
0016 was v22 for bticino. 
Now with new version the green power field appeared on deconz gui and some values are read/write. 
For info: option 2 on legrand specific cluster turns on the blue led inside the switch to make it visible during night

Smanar, 2020-12-17:
And the last version is 001C ? no more recent version ? From netatmo, you need to have the version 35 (0x23)

> 0x0B > need reverse (code reverse) WORKING
> 0x16 > No need to reverse (code reverse) NOT WOKING
> 0x1A > need to reverse (code reverse) WORKING
> 0x1C > No need to reverse (code reverse) NOT WOKING
> 0x21 > No need to reverse (code don't reverse) WORKING

0x1C is more "normal" I think I can use the value 0x1C for reverse/don't reverse, it will work for bticino and Legrand

Smanar, 2020-12-23:
Ok so if you want to test the new version, I have set the limit at 0x1C, so for the moment the only no working version will be the 0x16 if I m right

```
git clone --branch netatmo_cov https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

zinoalex, 2020-12-27:
I tried again today with the original gateway and 001c seems the latest release.
If your mods will be included in the next release at the beginning of January I'll wait for that
thanks

Smanar, 2020-12-27:
Can you test it to confirm it works for the version 0x0C ?

zinoalex, 2021-01-02:
no sorry, i'm not a programmer and i'm not able to compile

Smanar, 2021-01-02:
You don't need programmer skill ^^, just an unix machine.
I can make the PR in blind mode, but I can't guarantee result.

closed 2021-01-12 by manup

nicolarosada, 2023-05-15:
Hi,

I am sorry to resume this conversation, but I have version 002c of the K4027C and the version 0x26780700 of the ConBee II and it is reversed. Do you know whther would be possile to fix it?

Thanks for your help!

Smanar, 2023-05-16:
Hello, depend of the "reverse".
If it's the return that is reversed, it will be easy to correct (if you choose "open" it open the covering but this one have closed state)
If it's the request, bad new, will be another story (if you choose "open" and if it close the covering.)

nicolarosada, 2023-05-21:
It is the first you mentioned. When you select open it opens, but the resulting state is closed. And the opposite.

However, I have been using the shutters for aroud 10 days and 3 days ago it inverted the state! I honestly don't know why. I didn't change any configuration, just maybe a HA minor update, but I don't think this is the reason. Do you know if they need some time to "understand" the right state?

I'll install few more shutters in the next weeks, I will monitor the situation. Sory for the post, but it was truly not working when I commented few days ago.

Smanar, 2023-05-22:
Hu ?
No seriously, without deconz update, I don't understand what happen, and you can't have made a OTA update by error.
The code use the "attr/swversion" value for a "code check", but I don't see how the value can be missing during 7 days and appear after .....

But the good news is you have the better situation, it's solvable just with a DDF, but the DDF will be active for all device, so you need to have the issue on all Legrand shutter.

nicolarosada, 2023-05-28:
Yeah I don't understand honestly. Just consider that I don't have much experience with HA, I am just a casual user.

The only thing that I touched is the configuration of the device from the deCONZ app trying to modify some parameters, but at the end I should have restore the original configuration because I was not able to solve the problem. I can double check this when I will install few more shutters the next month.

If there is anything that I can do let me know. Othervise I can give you un update once I will install the new shutters and see if they work. Thank you very much for the very fast replies!

nicolarosada, 2023-07-12:
Ok I eventually understood what happened, it is not related to HA, but I leave it here for reference in case other people experience the same.

I didn’t find any reference on the manufacture website, but I am pretty sure that the bticino shutter modules have a logic inside and they “understand” and memorize the time necessary to open and close the shutter and its state. They probably do that after a few open/close cycles once the shutters are new and then they memorize/use it.

If you install a new module and you complete a few full open/close cycles then everything works fine. However, if you are installing the shutters and calibrating the motor as I did, you might end up doing several partial open/close cycles. This ends up with wrong timing in the module and wrong state of the shutter which is propagated to HA.

I solved it by simply resetting the bticino module and adding it again to HA. Sorry for bothering you with this issue that was not related to HA and thanks for the support.

mariomaz87, 2023-11-27:
Hi, I recently moved my K4027C shutter switch from Bticino Gateway to Deconz and I have this problem. I can control the device as expected only on Deconz Gui. From Phoscon or Home Assistant I can fully close or fully open the device, I can stop the movement and it will report the correct percentage, but I can't set it a to a specific % (as I can in Deconz Gui). I think all the firmwares are up to date: ConBee II 26780700, Gateway 2.24.2, Bticino 0x003643FF. Can you help me? Thanks!

Smanar, 2023-11-27:
Hello, can you share the device json ? Visible in phoscon/help/API Information/light ?

mariomaz87, 2023-11-27:
> Hello, can you share the device json ? Visible in phoscon/help/API Information/light ?
> 
> Edit: Ok have see #7390

If you need it for future reference:

   "4": {
        "config": {
            "groups": [
                "0"
            ]
        },
        "etag": "58fa2315a5226f4996e1a75bf8d7659c",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2023-11-27T16:29Z",
        "manufacturername": "Legrand",
        "modelid": "Shutter SW with level control",
        "name": "Finestra Bagno Vasca",
        "state": {
            "bri": 0,
            "lift": 0,
            "on": false,
            "open": true,
            "reachable": true
        },
        "swversion": "0x003643FF",
        "type": "Window covering device",
        "uniqueid": "00:04:74:00:00:9c:4c:ed-01"