dresden-elektronik/deconz-rest-plugin #3879: ORVIBO Zigbee Dry Contact CM10ZW

cba64, 2020-12-11:
## Device

- Product name: ORVIBO Zigbee Dry Contact CM10ZW
- Manufacturer: Unknown
- Model identifier: Unknown
- Device type : 
  - Switch (zigbee dry contact) 

## Screenshots
I don't know how to do that, but here is the description of the product:
https://shop.smarthome-europe.com/fr/peripheriques/4611-orvibo-module-commutateur-contact-sec-zigbee.html

Smanar, 2020-12-11:
Ha nice a native full dry contact. Realy usefull.
It s battery powered or with sector ?

If it s battery powered we need the screenshoot to do something.

An easy way if you don't have the GUI is installing deconz on a PC machine, switch the key, include the device, take the capture and switch again the conbee.

But you will probably need an Unix machine to test the code.

cba64, 2020-12-11:
Yes it is really useful, I don't understand how it didn't exist earlier ... only out of the box solutions I found were either expansive z-wave or unreliable WIFI devices ... but NOTHING with zigbee, that's a complete mystery to me.

It is powered with 220V, two cables L and N, and then two other cables for the dry contact. 
But the module can do other things.
User Manual can be downloaded here: https://www.domadoo.fr/fr/index.php?controller=attachment&id_attachment=2461

Unfortunately, I'm not able to pair it so far (not sure what category to put in deconz - switch or light or sensor) and not sure how to trigger the pairing mode on the device, or how to change its state to help detection - I tried many things but device does not appear on deconz, which is why I ask for support here).
I do have an unknown device on the network map but I'm new to this, I think this node was there before so...

Mimiix, 2020-12-11:
@Smanar I can't keep this open without the screenshots. 

@cba64 Please follow the guide (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support) to get device support. For reading clusters: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Once you get it added And set up, please open another issue or edit the first post of this one including all information.

Closing for now.

closed 2020-12-11 by Mimiix