dresden-elektronik/deconz-rest-plugin #3880: Complete loss of availability (lights and switches) after firmware update

nexttoyou, 2020-12-11:
## Describe the bug
After the Raspbee II firmware update to version 26680700 none of my lights and switches are accessible any longer.

## Steps to reproduce the behavior
All the lights and switches are showing up in the Phoscon web GUI (headless Raspberry Pi 4B) but cannot be switched on/off, no reaction from anything.

## Expected behavior
A firmware update should neither touch configured lights nor switches.

## Screenshots
Nothing unusual is shown in the web GUI but still nothing is working.

## Environment
- Raspberry 4B
- Firmware version 26680700
- deCONZ version 2.07.01 / 8.12.2020
- Raspbee II

## deCONZ Logs
```
/usr/bin/deCONZ --dbg-info=2
QCoreApplication::arguments: Please instantiate the QApplication object first
qt.qpa.screen: QXcbConnection: Could not connect to display 
Could not connect to any X display.
```
… as there is no display available

The excellent [homebridge-hue](https://github.com/ebaauw/homebridge-hue) plug-in, however, tells me at least the following:

```
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 272: PUT /lights/8/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 272: error 3 resource, /lights/8, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 273: PUT /lights/3/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 273: error 3 resource, /lights/3, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 274: PUT /lights/6/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 274: error 3 resource, /lights/6, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 275: PUT /lights/4/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 275: error 3 resource, /lights/4, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 276: PUT /lights/5/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 276: error 3 resource, /lights/5, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 277: PUT /lights/1/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 277: error 3 resource, /lights/1, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 278: PUT /lights/7/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 278: error 3 resource, /lights/7, not available
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 279: PUT /lights/9/state {"on":false}
[11/12/2020, 19:16:40] [Hue] Lux: gateway request 279: error 3 resource, /lights/9, not available
```

## Additional context
Backup from inside of the Phoscon web GUI has been uploaded but nothing at all happened for the better. System worked perfectly well before.

SwoopX, 2020-12-11:
Wait, just wait and don't touch anything.

Do you recall from which firmware you came?

nexttoyou, 2020-12-11:
Unfortunately not. Could it be found in the backup file from the web GUI?

nexttoyou, 2020-12-11:
By coincidence I did a screenshot once and there the old firmware is indicated: **26520700**

SwoopX, 2020-12-11:
Not too important. Allow it 1h to recover. I'm currently hoping it's currently trying to align the security frame counter.

If that doesn't help, we should fetch some logs. That's done slightly differently on headless.

Currently on mobile

nexttoyou, 2020-12-11:
And yep, it is also indicated in the backup file: **fwversion0x26520700**. Thanks for your fast reply anyway. Will report later.

nexttoyou, 2020-12-11:
One hour later the lights and switches still do not react. After updating the firmware I also waited easily a couple of hours before uploading the backup as the lights and switches refused cooperation. The bedroom lights are concerned, so will read my daily Carl Barks dose with smartphone flashlight this night before sleep …

SwoopX, 2020-12-11:
Hm, from what I read, you already seem to have interfered by loading a backup, still assuming this is/was related to the security frame counter.

So let's move on to the logs. Generate them with `/usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-error=2 > debug`. The output is redirected to the file `debug` placed in the directory where you've executed the command. Let it run for like 5 mins and share the file here.

I assume Phoscon is saying "not in network" on the gateway page?

nexttoyou, 2020-12-11:
I did the firmware update around 12 hours ago. This afternoon then (so at least four hours later) I realised that nothing did work. It might have been the error "not in network" then but I do not remember to well. I then uploaded the backup and afterwards posted the issue here. If I’d known that a backup could create further issues I wouldn’t have done it of course.

[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5681666/debug.txt)

Mimiix, 2020-12-11:
The device is not found.

22:08:28:709 COM failed to open /dev/ttyS0: Permission error while locking the device

You should do a manual update : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

nexttoyou, 2020-12-12:
I saw that and am puzzled by it. I did the firmware update manually again but, unfortunately, the result is still the same.

Would returning to the previous firmware be the solution perhaps?

nexttoyou, 2020-12-13:
So it seems I have to redo everything as even a waiting period of several hours as well a return to a backup didn’t work? Well, not so good news six weeks after the initial setup. If a firmware update with such a simple configuration (five switches and eight lightbulbs – all from Signify/Philips Hue) already leads to such a result doing firmware updates in the future will be something to avoid for me. "Nothing works!", like good old [Catweazle](https://en.wikipedia.org/wiki/Catweazle) used to say. Many thanks again for your help efforts, it's still the thought that counts!

nexttoyou, 2020-12-13:
Nope, re-installing everything doesn’t work too. Now the web GUI isn’t accessible anymore. Complete loss of functionality. Please help, recent debug file included.

[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5684604/debug.txt)

nexttoyou, 2020-12-13:
What I did to resolve the situation:

1) Downgrade to the oldest firmware available for the RaspBee II (deCONZ_RaspBeeII_0x26610700.bin.GCF)
2) Web GUI accessible again but everything else lost -> reset the gateway
3) Upgrade to the most recent firmware available (deCONZ_RaspBeeII_0x26680700.bin.GCF)
4) Redid every bulb (8) and switch (5) from scratch using screenshots as backup obviously useless
5) Made a backup from this (probably useless if it happens again)

closed 2020-12-13 by nexttoyou