dresden-elektronik/deconz-rest-plugin #3885: Orvibo ST30 Temp/Humidity Sensor with diplay

samsam-rolon, 2020-12-12:
[image]

## Device

- Product name: Orvibo ST30.
- Manufacturer: FEI
Basic cluster
[image]

- Model identifier: As screenshot
- Device type : 
  - Sensor
[image]

## Screenshots

PowerConfig cluster
[image]

Temp cluster
[image]

Humidity cluster 01
[image]

Humidity cluster 02
[image]

Smanar, 2020-12-12:
This device is already in code https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2489
Just need to add some lines, will be fast.

samsam-rolon, 2020-12-12:
IDK why there is 2 Relative humidity measurement but it's the cluster 0405 which it's standalone (the last one) which is working.
Here the last screenshot 😄 

[image]

Smanar, 2020-12-12:
Ok so first try, you know the procedure now ^^

```
git clone --branch orvibo https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

This PR
- make the device pairable
- change the name "898ca74409a740b28d5841661e72268d" in "ST30 Temperature Sensor", can have made typo here
- Prevent the creation of humidity sensor for cluster 0x01

samsam-rolon, 2020-12-19:
I think there is a typo error  :S
[image]

Smanar, 2020-12-19:
No it s not a typo ^^
I had this bug too, need to update the system (or just deconz-dev if you can)

samsam-rolon, 2020-12-20:
System is already updated, IDK why. But if i'm changing `dataType_t `in `dataType` it's compiling ! But not working in deconz IHM / API / WebGUI

samsam-rolon, 2020-12-20:
Okay, my bad compilation works now ...

Smanar, 2020-12-20:
Yeah, I don't know the moethod to update only the dev package (not worked) so on mu side have upgraded all the OS

```
sudo apt update
sudo apt full-upgrade
```

samsam-rolon, 2020-12-21:
Yeah, i've done it and it's Okay now for compiling but notthing in GUI or API.

Smanar, 2020-12-21:
You have tried to re-include it ? (Or read again basic attribute with phoscon in permit join) ?
There is some log about this device ?

Smanar, 2020-12-27:
Can you share log during inclusion if it s still not working ?

srtxema, 2021-01-01:
i can´t sync with phoscon. is there any special sync mode?

Smanar, 2021-01-02:
@srtxema you are using the recompiled code ?

samsam-rolon, 2021-01-03:
Sorry @Smanar  ! With holidays, not easy to work on it ! I'll will test deeply next week !

samsam-rolon, 2021-01-10:
Hello @Smanar !
I confirm : re-inclusion let it detected now !
[image]
`json
{
  "config": {
    "battery": 100,
    "offset": 0,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "b7b6b57e79e89d4dd8092b118257fede",
  "lastseen": "2021-01-10T12:03Z",
  "manufacturername": "fei",
  "modelid": "ST30 Temperature Sensor",
  "name": "ST30 Temperature Sensor",
  "state": {
    "lastupdated": "2021-01-10T12:03:44.533",
    "temperature": 1986
  },
  "swversion": "ST01",
  "type": "ZHATemperature",
  "uniqueid": "00:15:8d:00:03:56:9d:2a-01-0402"
}
`

`json

{
  "config": {
    "battery": 100,
    "offset": 0,
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "01375309452f213eccf49fab3a9d7a4c",
  "lastseen": "2021-01-10T12:03Z",
  "manufacturername": "fei",
  "modelid": "ST30 Temperature Sensor",
  "name": "ST30 Temperature Sensor",
  "state": {
    "humidity": 3952,
    "lastupdated": "2021-01-10T12:03:45.142"
  },
  "swversion": "ST01",
  "type": "ZHAHumidity",
  "uniqueid": "00:15:8d:00:03:56:9d:2a-02-0405"
}`

Smanar, 2021-01-10:
>But I didn't see the device on web UI, Only through API

Yep new device, need to wait if the phoscon dev want to add it in whitelist.

Will make the PR today.

srtxema, 2021-01-10:
> @srtxema you are using the recompiled code ?

finally i could. press the button continuosly 3 times for 20 seconds and sync.

stifler6969, 2021-01-27:
I bought this device (ST30) but it was not recognize by jeedom/deconz (inclusion OK). I have deconz 2.07.01.
How i do ?

Mimiix, 2021-01-27:
@stifler6969 You need to manually compile deconz on a newer version. Otherwise, you have to wait until merged and released.

stifler6969, 2021-01-27:
@Mimiix thx for your reply. I will wait...

samsam-rolon, 2021-02-11:
I've just upgrade deconz in 2.07.03 and it's still not working for both orvibo devices added (#3973)

Smanar, 2021-02-11:
The pr is not validated ^^.
Easy to see, the issue will be automaticaly closed when the PR will be validated.

samsam-rolon, 2021-02-12:
Oh yeah ! I've just seen that ! :(

closed 2021-03-03 by manup

samsam-rolon, 2021-03-05:
Hello @Smanar 
I've updated in 2.10.01 and I didn't see the device, even after re-inclusion.
[image]
Pairing still failing in  WebGUI and the name isn't displayed in Software- GUI 

Any idea

Mimiix, 2021-03-05:
@samsam-rolon Need to be on 2.10.2

samsam-rolon, 2021-03-08:
Yeah ... my bad ... latest docker image https://hub.docker.com/r/marthoc/deconz is still not maped on the latest beta .... really painful ^^'

Leobaillard, 2021-03-25:
Hi! First of all, thanks for adding the integration!

After talking about it on the Discord, it would appear that my ST30 has a different ModelID than the one added to the code. Here is the screenshot with the info:
[image]

For reference, the Model ID is: 50938c4c3c0b4049923cd5afbc151bde

I can provide other info if necessary!

Smanar, 2021-03-25:
You can mimic the previous model id to make a PR, if you want ? I think you will have only 2 lines to change

Leobaillard, 2021-03-25:
Yes, I can try to do a PR. I have started looking at the code, but weirdly enough, the manufacturer code is also different (0x1002, which matches VENDOR_EMBER). Is that common for a device that looks exactly the same?
I'm also not sure yet how to test it all as I'm currently using HA and the Deconz plugin, but I can at least work on the PR.

stifler6969, 2021-03-26:
Hi,

I upgraded my deconz in 2.10.04 and i didn't see my ST30 sensor. I tried an new pairing. It's normal ?

Smanar, 2021-03-26:
@Leobaillard 
Yep on my side I have see 3 modifications to do, the whitelist with the new manufacture, and 2 place in the code where you need to mimic previous code

To make modification, you need a linux machine, you have the process here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

@stifler6969 can you show the device model id ? and BTW have you a linux machine to test code modification if you need them ^^ ?

Leobaillard, 2021-03-26:
@Smanar Yes, I gathered some info on the Discord side and started setting up my dev environment. I have resources and knowledge to accomplish that, just need time now. I'll start testing hopefully tonight or this weekend.

Leobaillard, 2021-03-26:
PR made, the device is working on my end :)

Smanar, 2021-03-27:
Nice, thx ^^