dresden-elektronik/deconz-rest-plugin #3886: Device Request - LIdl - Smart Window/Door sensor

albei, 2020-12-12:
### **Device:**

- Product name: Smart Window or Door Sensor.
- Manufacturer: SilverCrest.
- Model identifier: 
- Device type : Sensor
- Other: IAN 341806_1910 SDWZ 1 A1
- https://wikilidl.com/2020/11/09/silverrest-smart-window-or-door-sensor/
### Screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-12-12:
This device is already supported, no ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3767

albei, 2020-12-12:
Oh did not found it on Issues search, probably I was filtering the closed ones.
When I tried to add the sensor in webapp nothing happens nothing gets added.

Smanar, 2020-12-12:
Np, so I have make an experimental code, to avoid to add 5 lignes by tuya device for all mac adress possible every time.

Do you have an Unix machine to test it ?

Explanations here : https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
git clone --branch tuya_general https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then retry to include it.

albei, 2020-12-12:
> Np, so I have make an experimental code, to avoid to add 5 lignes by tuya device for all mac adress possible every time.
>
> Do you have an Unix machine to test it ?
>
> Explanations here : [...]

Sadly I have deconz installed via Hassio, so its the docker image. I dont have a local deCONZ install. 
If its possible to do it on the doker image I will happily test is. But a need a hint on how to do it.

Smanar, 2020-12-12:
You can do nothing on a docker image.
It s possible on a VM, but I have never do it on my side.

albei, 2020-12-12:
I got the container updated and now I see it, and look like it works!
I am gonna setup a rasp image for testing anyway and install it locally for future test. And I will join discord also.
@Smanar many thanks for your work! If you need me to test it give me shout so we can talk faster on discord.

Smanar, 2020-12-12:
So it have worked without testing the new code ?

albei, 2020-12-12:
Yep, I rescan for sensor,pushed the button on the sensor shortly until the led blinked green once and it appeared in the list. I already made automation for it in hassio 😎.
Thanks!

Mimiix, 2020-12-12:
Case closed

closed 2020-12-12 by Mimiix