dresden-elektronik/deconz-rest-plugin #3888: closed Port 8080 RaspBee II

syswombat, 2020-12-12:
## Describe the question or issue you are having
Done a basic installation no Graphical interface just all command line, and followed this
https://phoscon.de/de/raspbee2/install#raspbian

Doing this installation the port 8080 is not opend, so doing a connection with node Red is failling
Could you please tell me if the port 8080 have allways to be open or only to pear node Red with Raspbee II?
And how to open this Port?

## Screenshots

## Environment
System: Raspberry Pi 4 Model B Rev 1.1
with a RaspBee II
Distro: Raspbian GNU/Linux 10 (buster)
Host system: (Raspberry Pi)
Running method: (Raspbian)
Firmware version: (26610700)
deCONZ version: (2.07.01 / 14.9.2020)
Device: (RaspBee II)

## deCONZ Logs

## Additional context

Smanar, 2020-12-12:
You can choose the port you want, and if one is not available deconz choose itself another one

`sudo netstat -tulpn | grep "deCONZ"`

To find your websoket port

Mimiix, 2020-12-12:
If you've started it with sudo, it starts on port 80 afaik

syswombat, 2020-12-12:
Thanks for the feedback, yes with port 80 it works.
Now connected to my zigbee device Worl - i have a bunch of Product from lidl (Bulb, Powerswitch and other Thing)

Mimiix, 2020-12-12:
So case is closed?

syswombat, 2020-12-12:
Last question,
if i have to connect Home Assistant, Zigbee2mqtt or openHAB
will i have to reconect or just copy the code i have in Node Red

Mimiix, 2020-12-12:
Not sure and you should ask at those platform communities.

closed 2020-12-12 by Mimiix