dresden-elektronik/deconz-rest-plugin #3890: HY368 (Tuya / Yidi Radiator Actuator)

Luigi-01, 2020-12-12:
HY368

- Product name: Thermostat valve Zigbee
- Manufacturer: YIDI
- Model identifier: HY368
- Device type : Radiator Actuator Thermostat

## Screenshots
[image]

[image]

[image]

Smanar, 2020-12-12:
Hello, can you show the basic attributes ?
Clic on cluster 0x0000 "basic" and pls show the panel.

Mimiix, 2020-12-12:
Seems to be fully reported already.

Closed