dresden-elektronik/deconz-rest-plugin #3892: Connection to Tuya/Moes TRV (TS0601) lost after update to deCONZ 2.7.1

kuszek, 2020-12-12:
## Describe the bug
On deconz 2.05.88 (HA addon 6.5.0) I had encountered a problem with Tuya/Moes TRV sometimes showing 30 degrees out of nowhere so I restored deCONZ version 2.05.84 (addon 6.4.1) and had been waiting for the update.

After the update to deconz 2.7.1 (addon 6.6.0) I encountered another problem. I couldn't change the thermostat's mode to "heat" or "auto". Only hvac_mode "off" worked. I tried the following:
- using the service deconz.device_refresh
- restarting deCONZ addon
- restarting Home Assistant
- re-adding of the TRV to deCONZ (turning paring mode on TRV again)
- removing deCONZ node and re-adding of the TRV a couple of times

Eventually I again reverted to version 2.05.84 and everything works properly ever since.

## Steps to reproduce the behavior
  1. Go to HA dashboard.
  2. Change thermostat's hvac_mode to "heat" or "auto". OR
      Click on the thermostat card and change hvac_mode to "heat" or "auto".
  3. State of the thermostat doesn't change and error message "Failed to call service climate/set_hvac_mode. /sensors/##/config/mode parameter, mode, not available" is displayed.

## Expected behavior
Thermostatic Radiator Valve mode changes to "heat" or "auto".

## Screenshots

I tried to change a mode on the default HA thermostat card and on simple-thermostat card. There is no difference.
Temperature values are displayed properly though.
[image]

Clicking on the thermostat card and changing of entity's state fails too.
[image]

TRVs are shown as connected properly.
[image]

## Environment
 - Host system: Raspberry Pi 4B
 - Running method:  Home Assistant deCONZ Add-on
 - Firmware version: 26680700
 - deCONZ version: 2.7.1 (12/8/2020)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, APC UPS

## deCONZ Logs

I couldn't copy the debug info so I'm attaching it as a screenshot.
[image]

Deconz log
[deconz log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5683298/deconz.log.txt)

Home Assistant log
[ha log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5683300/ha.log.txt)

Smanar, 2020-12-12:
Hello, do you have the manufacture name of the device (on the basic attribute) to reconize the device ?
Or if you can the complete JSON.

kuszek, 2020-12-12:
Hello, thank you for such a quick reply.

I have three thermostats. I bought one as Shojzj and later I bought two Moes ones.

If by GUI you mean Phoscon, then all three thermostats don't show up there, but in deconz VNC and in Home Assistant they were described as model TS0601, vendor Heiman.

Node list:
[image]

After reading the info from basic cluster it looks like this:

Shojzj (378RT)
[image]

1st Moes (HY368)
[image]

2nd Moes (HY368)
[image]

Node list after basic cluster reading:
[image]

I didn't really get your whole request but I hope this helps.

I don't if this information helps but after updating the addon to version 6.6.0 I updated the firmware from version 26660700 to 26680700. After reverting to addon 6.4.1 I'm still using version 26680700. I have never used version 26670700.

kuszek, 2020-12-12:
@Smanar
I read the whole issue #3697 and I think I now understand what you meant in your post. After reading the proper information from basic cluster and updating the addon to version 6.6.0 everything works fine now. The bug has disappeared.

For the moment there I had a problem with one Moes TRV - it didn't show an actual temperature but after re-adding it again to deCONZ it shows the temperature properly.

And for the first time ever these thermostats appeared in Phoscon.

Smanar, 2020-12-13:
>on the API, not on deconz, never read the manufacture name on the GUI else it will probably make the bug disapear

Lol, yes, too late now to find the problem.

>And for the first time ever these thermostats appeared in Phoscon.

Ha ? Goog new, I don't know phoscon display thermostat now.

closed 2021-03-05 by github-actions[bot]