dresden-elektronik/deconz-rest-plugin #3893: Bug: Samsung Moisture v4 Sensor (recently added to stable deconz) not Identifying Properly

adawalli, 2020-12-12:
## Describe the bug
@SwoopX recently added support for the Samsung Smartthings Moisture Leak v4 sensor (thanks!). However, even after this change making it's way to the latest stable release, it is not getting recognized properly.

I do have logs of the discovery/permit stage and will attach them below

## Steps to reproduce the behavior
1. Open Phoscon App
2. Click Sensors Page
3. Add sensor
4. Click Other
5. Wait 3 minutes for failure message

## Expected behavior
Samsung Smartthings moisture v4 to be identified

## Screenshots
I can rescan the Cluster info if necessary but I have no screenshots otherwise

## Environment

Phoscon Gateway version : 2.07.01 / 12/8/2020

 - Host system: Raspberry Pi 4
 - Running method: Hass.IO + Deconz Addon/Phoscon
 - Firmware version: 26660700
 - deCONZ version:2.07.01 
 - Device: Conbee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? SSD, Aeotec v5

## Additional context
I only have a handful of routers at this point and I am using the USB extension cable.

## deCONZ Logs
```log
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
09:50:00:138 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
09:50:00:143 CTRL. 3.27.209:50:00:196 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2233986-if00
09:50:00:239 dev /dev/ttyAMA0
09:50:00:241 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2233986-if00 / serialno: , ConBee II
09:50:00:242 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
2020/12/12 14:50:00 [notice] 323#323: using the "epoll" event method
2020/12/12 14:50:00 [notice] 323#323: nginx/1.14.2
2020/12/12 14:50:00 [notice] 323#323: OS: Linux 5.4.79-v8
2020/12/12 14:50:00 [notice] 323#323: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/12/12 14:50:00 [notice] 323#323: start worker processes
2020/12/12 14:50:00 [notice] 323#323: start worker process 1548
09:50:00:368 parent process s6-supervise
09:50:00:369 gw run mode: docker/hassio
09:50:00:370 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
09:50:00:371 DB sqlite version 3.27.2
09:50:00:373 DB PRAGMA page_count: 35
09:50:00:373 DB PRAGMA page_size: 4096
09:50:00:374 DB PRAGMA freelist_count: 0
09:50:00:374 DB file size 143360 bytes, free pages 0
09:50:00:374 DB PRAGMA user_version: 7
09:50:00:375 DB cleanup
[... 1263 lines cut ...]
10:00:48:643 CTRL skip polling while permit join is set
10:00:48:648 unknown neighbor 0x24FD5B00010B4137 (0x753A), lqi: 255, rssi: -25
10:00:48:649 FP indication 0x0104 / 0x0500 (0x24FD5B00010B4137 / 0x753A)
10:00:48:649                       ...     (0x24FD5B00010B4137 / 0x753A)
10:00:48:650 Clear fast probe timeout for cluster 0x0500, 0x24FD5B00010B4137
10:00:48:654 [3.1] get IAS Zone type for 0x24fd5b00010b4137
10:00:48:896 CTRL skip polling while permit join is set
10:00:49:154 CTRL skip polling while permit join is set
10:00:49:159 unknown neighbor 0x24FD5B00010B4137 (0x753A), lqi: 255, rssi: -25
10:00:49:160 FP indication 0x0104 / 0x0500 (0x24FD5B00010B4137 / 0x753A)
10:00:49:161                       ...     (0x24FD5B00010B4137 / 0x753A)
10:00:49:162 Clear fast probe timeout for cluster 0x0500, 0x24FD5B00010B4137
10:00:49:166 [3.1] get IAS Zone type for 0x24fd5b00010b4137
10:00:51:122 send permit join, duration: 59
10:00:51:124 send GP proxy commissioning mode
10:00:53:686 Search sensors done
10:00:54:686 Search sensors done
10:00:56:168 CTRL skip polling while permit join is set
10:00:56:427 CTRL skip polling while permit join is set
10:00:56:432 unknown neighbor 0x24FD5B00010B4137 (0x753A), lqi: 255, rssi: -25
10:00:56:675 CTRL skip polling while permit join is set
10:01:00:137 Current channel 25
10:01:00:145 Device TTL 2922 s flags: 0x7
10:01:03:677 CTRL skip polling while permit join is set
```

SwoopX, 2020-12-12:
Thanks for sharing the log data, very helpful in this case. Despite the fact, that you seem to have quite some interference, this is probably not the root cause of your issue. During pairing, the device got stuck in querying the IAS zone type, as it is expected to have a value other than "Standard CIE" (which is the default while not being queried). Either the device does not provide a response to the query, or it responds with a value unexpected by deconz.

Can you please share a screenshot from the IAS zone cluster, when the attributes have been read? The one given in the device support request isn't suitable for that (attributes have NOT been read). For this device, I'd expect the `zone type` to be "Water sensor".

adawalli, 2020-12-12:
@SwoopX This one Cluster is odd - For all the other clusters, I have been able to hit "read" on the attributes table, and then I tap the button on my sensor, and the table slowly populates with new data.

No matter what I have tried on this IAS Zone, the table does not seem to change values at all from what you see below

Any ideas?

[image]

adawalli, 2020-12-12:
I pulled up the DEBUG view this time while I tried to read the attributes - forgive my ignorance if this is unrelated, but I see the following printing out - not sure if this is related to me trying to read those attributes

[image]

SwoopX, 2020-12-12:
> 
> 
> I pulled up the DEBUG view this time while I tried to read the attributes - forgive my ignorance if this is unrelated, but I see the following printing out - not sure if this is related to me trying to read those attributes
> 
> [image]

This appears to be a light, so it's unrelated.

Can you give me another screenshot of the node with the clusters expanded? If it still has its NWK as name (0xWXYZ), then no sensor has been created. In case not, you might want to try entering sensor search in Phoscon and reading the basic cluster.

adawalli, 2020-12-13:
wow, lol - I had a bit of dyslexia I guess. It turns out that one of my lights ended in 317 and my moisture sensor ended in 137 - my bad

I put phoscon in Sensor Add mode and then read the basic attributes back in deconz. Is that what you were asking for?

[image]

adawalli, 2020-12-13:
Wow - I just noticed that it fully added for the very first time ( I even see it in Home Assistant) - Did forcing a read of basic attributes while Phoscon was running help push it along?? Interestingly, it doesn't show up in my sensor menu of Phoscon (but I can see it not only in deconz, but also registered in Home Assistant now)

SwoopX, 2020-12-13:
So, what's the issue after all?

adawalli, 2020-12-13:
The issue is that it's apparently very difficult to get the sensor to actually add. I had already tried upwards of 10 times and it magically worked on this last try without any real changes.

Mimiix, 2020-12-13:
Case closed?

closed 2020-12-14 by adawalli

Mimiix, 2020-12-14:
Case closed!