dresden-elektronik/deconz-rest-plugin #3895: Aqara Motion Sensor doesn't work correctly after server restart

Dominik-1980, 2020-12-12:
## Describe the question or issue you are having

Hello.

I have got some problems with my two aqara motion sensors. Especially with one of them.

When I restart the server everything connects again, two remote switch, two door/window sensors, one temp/hum sensor, one Osram Plug that acts as a repeater and the motion sensors. At the beginning there is a very slow response, but after one or two hours it works like it should be, but the second motion sensor always causes trouble.

I have to repair it many times till it works again as expected.
And even then, it doesn't show every cluster in the deconz-gui that it should.
Only Basic, Illuminance measurement and Occupancy sensing is shown there. Temperature and Battery is not.
That cause me an error shown Empty Battery in Homekit.
What can I do to get all the needed cluster? Please take a look at the screenshots...

## Screenshots

[image]

[image]

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 8.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Smanar, 2020-12-12:
Strange, an both are using same device as router ?
The battery level is good (when it s displayed) ?

Dominik-1980, 2020-12-12:
Yes. Battery is 100%. Same router...
I deleted the sensor once again and repaired it, now it shows Battery and Temperature, but only in Phoscon, not in deConz.

What makes it very strange is, that in the Phoscon App, the device recognizes my motion.
If I go in range of the sensor, Phoscon shows me the Refreshed time "some seconds ago" but Motion state still is "none".

[image]

bergiusj, 2020-12-12:
I'm facing same sort of problem. One of my Aqara motion sensors are reporting lux correctly, but motion detection is very buggy. Sometimes it detects, and sometimes it's totally dead. If I inspect the sensor i Phoscon I also have Version: unkown which is not the case on the sensors that's working correctly.

SwoopX, 2020-12-14:
@Dominik-1980 which of the sensors in your screenshot with the expanded clusters is behaving stragely? I'd assume the left one?

Dominik-1980, 2020-12-14:
Yes. The left one. I think the problem was the repeater (plug) the sensor was connected to.
I deleted the sensor and paired it straight to the ConBee 2 Stick and it worked. 
The missing fields in deCONZ showed up after one or two hours...

Mimiix, 2020-12-15:
Case closed?

closed 2020-12-15 by Dominik-1980

Mimiix, 2020-12-16:
Case closed!

SwoopX, 2020-12-16:
Just for reference, this might be related
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3528#issuecomment-743367042